Azure/iotedge-lorawan-starterkit #21: Question:  How to resolve deployment issues from VS Code?

gerfen, 2018-10-26:
When I attempt to `Build and Push IoT Edge Solution`  I get the following error:

_Sending_ build context to Docker daemon 2.523 MB
_Step_ 1/13 : FROM microsoft/dotnet:2.0-sdk AS build-env
_Error_ parsing reference: "microsoft/dotnet:2.0-sdk AS build-env" is not a valid repository/tag: invalid _reference_ format

I've confirmed that my Docker installation matches the server. I'm running 17.03.1-ce-rc1-win3.

As far I can tell, I've set up VS Code with all of the tooling to publish IotEdge modules.  Any suggestions to resolve this would be greatly appreciated.

ronniesa, 2018-10-26:
I'm using Version 18.03.0-ce-win59 (16762)

gerfen, 2018-10-26:
That did the trick.  Thanks!  

I think updating the iotedge-lorawan-starterkit/LoRaEngine/readme.md with the required Docker version would be helpful.  I can submit a PR if you like.

gerfen, 2018-10-26:
Well I jumped the gun.  My code now builds but now I'm getting an "unauthorized: authentication required" error.  I've authenticated to Azure via the Azure: Sign In command.

_Sending build context to Docker daemon  3.508MB
Step 1/13 : FROM microsoft/dotnet:2.0-sdk AS build-env
 ---> 9e06837225a4
Step 2/13 : WORKDIR /app
 ---> Using cache
 ---> 122d1c1544ed
Step 3/13 : COPY ./Logger ./Logger
 ---> Using cache
 ---> 90947ebb74b9
Step 4/13 : COPY ./LoRaWan.NetworkServer ./LoRaWan.NetworkServer
 ---> Using cache
 ---> 761fa7299018
Step 5/13 : COPY ./LoRaWanNetworkSrvModule ./LoRaWanNetworkSrvModule
 ---> Using cache
 ---> 81e3101c45bb
Step 6/13 : COPY ./LoraTools ./LoraTools
 ---> Using cache
 ---> 66b01d6279e6
Step 7/13 : WORKDIR ./LoRaWanNetworkSrvModule
 ---> Using cache
 ---> 62cf0b0133e4
Step 8/13 : RUN dotnet restore
 ---> Using cache
 ---> 9598bd800e37
Step 9/13 : RUN dotnet publish -c Release -o out
 ---> Using cache
 ---> 1a36a5053e00
Step 10/13 : FROM microsoft/dotnet:2.0-runtime-stretch-arm32v7
 ---> 87595eb7f1f4
Step 11/13 : WORKDIR /app
 ---> Using cache
 ---> d10c668177db
Step 12/13 : COPY --from=build-env /app/LoRaWanNetworkSrvModule/out/* ./
 ---> Using cache
 ---> 25626a7934dc
Step 13/13 : ENTRYPOINT ["dotnet", "LoRaWanNetworkSrvModule.dll"]
 ---> Using cache
 ---> 821c4e86976c
Successfully built 821c4e86976c
Successfully tagged evolution.azurecr.io/lorawannetworksrvmodule:0.2.0-preview-arm32v7
SECURITY WARNING: You are building a Docker image from Windows against a non-Windows Docker host. All files and directories added to build context will have '-rwxr-xr-x' permissions. It is recommended to double check and reset permissions for sensitive files and directories.
The push refers to repository [xxxxxxxxxxxxx/lorawannetworksrvmodule]
32a9761ed370: Preparing
d3376d928e07: Preparing
420a4cbda8df: Preparing
53c3793bdb6b: Preparing
002111fc932d: Preparing
2e9c4696ffa4: Waiting
unauthorized: authentication required_

gerfen, 2018-10-26:
Thanks.  Executing this from a command prompt resolved the authentication issue: 
`docker login xxxxxx.azurecr.io -u xxxxxxxx-xxxx-xxxx-xxxx-xxxxxxxxxxxx -p xxxxxx`

closed 2018-10-26 by gerfen