Azure/iotedge-lorawan-starterkit #60: Deploying LoraKeysManagerFacade results in no functions

thomasvoorend, 2018-12-12:
After creating and deploying the function app with the Functions plugin in VS Code, there are no functions visible in the Azure Portal.

In the `.vscode/settings.json` I already adjusted the deploy path to: `LoraKeysManagerFacade/bin/Release/netstandard2.0/publish`

Does anyone know how to resolve this issue?

[image]

fbeltrao, 2018-12-12:
Hi @thomasvoorend,

What works for me is opening VS Code in the function folder (LoRaEngine/LoraKeysManagerFacade) and then publish to Azure.

thomasvoorend, 2018-12-12:
Hi @fbeltrao 
Thank you! That did the trick for me also!

closed 2018-12-12 by thomasvoorend