Azure/iotedge-lorawan-starterkit #74: Join issues

zhaozhenhong5, 2018-12-28:
06:36:13.684 47AAC86800430028: join request received
2018-12-28 06:36:13.686 47AAC86800430028: querying the registry for device key
2018-12-28 06:36:13.964 47AAC86800430028: using edgeHub local queue
2018-12-28 06:36:13.967 47AAC86800430028: using cached twins for OTAA device
2018-12-28 06:36:13.980 Error processing the message Object reference not set to an instance of an object.,    at LoRaWan.NetworkServer.MessageProcessor.ProcessJoinRequest(LoRaMessageWrapper loraMessage) in /app/LoRaWan.NetworkServer/MessageProcessor.cs:line 579
   at LoRaWan.NetworkServer.MessageProcessor.ProcessMessageAsync(Byte[] message) in /app/LoRaWan.NetworkServer/MessageProcessor.cs:line 55
   at LoRaWan.NetworkServer.UdpServer.<>c__DisplayClass10_1.<<RunUdpListener>b__0>d.MoveNext() in /app/LoRaWan.NetworkServer/UdpServer.cs:line 105

#############################
The isssues above from the  "iotedge logs LoRaWanNetworkSrvModule" when my device started to join,but join fail. Do somebody know why ?

ronniesa, 2018-12-28:
Which region are you EU or US?

zhaozhenhong5, 2018-12-29:
In fact, I am using the CN470 config, see the link: https://github.com/TheThingsNetwork/gateway-conf/blob/master/CN-global_conf.json

ronniesa, 2018-12-30:
Yes, we don’t support yet CN470. If you want to test it the only thing that should work is unconfirm upstream using ABP. Anything that requires downstream messages like join accept, confirmations and cloud to device messages is not yet supported for CN470. We would like to add support for all regions but in the current preview we support EU and US only.

zhaozhenhong5, 2018-12-31:
Thank you! And if I want to test it, how can I using ABP with your code? Will change the code somewhere?

zhaozhenhong5, 2019-01-02:
Hello! Can I and the CN470 region by myself? And how to add it?Thank you!

Mandur, 2019-01-02:
Hello Zhao,
Configuring your devices for ABP requires two things:
1. In your Arduino Code, you need to set the device for ABP (No join process, but directly upstream messages). You can check our Arduino examples for reference [ABP](https://github.com/Azure/iotedge-lorawan-starterkit/blob/master/Arduino/EU/TransmissionTestABPLoRa/TransmissionTestABPLoRa.ino) [OTAA](https://github.com/Azure/iotedge-lorawan-starterkit/blob/master/Arduino/EU/TransmissionTestOTAALoRa/TransmissionTestOTAALoRa.ino).
2. You need to configure another set of device twin on your device on Azure, as described [here](https://github.com/Azure/iotedge-lorawan-starterkit/blob/04cd56aa27494e80f06eea822adc30e530813216/Arduino/EU/TransmissionTestABPLoRa/TransmissionTestABPLoRa.ino#L13).
Let me know if this helps out

Mandur, 2019-01-02:
Regarding implementation of the CN470 region, yes of course, that'd be great!

The region class is (in this file)[https://github.com/Azure/iotedge-lorawan-starterkit/blob/master/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoraTools/Regions/Region.cs]
A factory instantiate it with the correct settings (here)[https://github.com/Azure/iotedge-lorawan-starterkit/blob/master/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoraTools/Regions/RegionFactory.cs]. You find all the required settings [here](https://lora-alliance.org/sites/default/files/2018-07/lorawan_regional_parameters_v1.0.3reva_0.pdf)
Adding a CreateCNRegion with the CN specs would do it. Let us know how It goes or if you have any further questions

zhaozhenhong5, 2019-01-03:
Yeah,thanks for your message. And I added the CN470 by myself. I saw the CN470 region is similar to US region,so I replaced it . My code here:[https://github.com/zhaozhenhong5/zhao](url).

But now the "iotedge logs LoRaWanNetworkSrvModule" still show issues:
#########################
2019-01-03 07:56:29.237 Physical dataUp {"rxpk":[{"tmst":155288956,"chan":0,"rfch":1,"freq":487.100000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":12.2,"rssi":-81,"size":23,"data":"AOIUAAAAAHq+KABDAGjIqkfPlUeqPLE="}]}
2019-01-03 07:56:29.239 47AAC86800430028: join request received
2019-01-03 07:56:29.257 47AAC86800430028: querying the registry for device key
2019-01-03 07:56:29.563 47AAC86800430028: using edgeHub local queue
2019-01-03 07:56:29.567 47AAC86800430028: using cached twins for OTAA device
2019-01-03 07:56:29.570 47AAC86800430028: saving join properties twins
2019-01-03 07:56:30.785 47AAC86800430028: could not save twins with error: Operation timeout expired.
2019-01-03 07:56:30.787 47AAC86800430028: join request could not save twins, join refused
2019-01-03 07:56:30.788 47AAC86800430028: could not save twins with error: Operation timeout expired.
2019-01-03 07:56:30.790 47AAC86800430028: join request could not save twins, join refused
#######################
Do you know the problems? Is my code added wrong somewhere?

fbeltrao, 2019-01-03:
Hi @zhaozhenhong5,

According to the logs the join request failed because it took too long (> 6 seconds).

Usually a device should wait a bit and retry to join. Do you have retries on the OTAA join?

zhaozhenhong5, 2019-01-03:
Yeah, my device would wait a bit and retry to join,you can see my device log:
######################
deveui:47aac86800430028
appeui:be7a0000000014e2
appkey:8afe71a145b253e49c3031ad068277a1
ch0,487100000,50,1
ch1,487300000,50,1
ch2,487500000,50,1
ch3,487700000,50,1
ch4,486300000,50,1
ch5,486500000,50,1
ch6,486700000,50,1
ch7,486900000,50,1
ch8,487100000,50,0
ch9,487100000,50,0
ch10,487100000,50,0
ch11,487100000,50,0
ch12,487100000,50,0
ch13,487100000,50,0
ch14,487100000,50,0
ch15,487100000,50,0
rx start=502300000

start join...
TX:freq=487500000,SF=7,PW=20,BW=0
RX:freq=502700000,SF=7,BW=0,RXM=0
RX:freq=505300000,SF=12,BW=0,RXM=0
TX:freq=486900000,SF=9,PW=20,BW=0
RX:freq=503700000,SF=9,BW=0,RXM=0
RX:freq=505300000,SF=12,BW=0,RXM=0
TX:freq=487100000,SF=10,PW=20,BW=0
RX:freq=502300000,SF=10,BW=0,RXM=0
RX:freq=505300000,SF=12,BW=0,RXM=0
TX:freq=487100000,SF=12,PW=20,BW=0
RX:freq=502300000,SF=12,BW=0,RXM=0
RX:freq=505300000,SF=12,BW=0,RXM=0
join failed=4

fbeltrao, 2019-01-03:
Do you see in IoT Edge module log multiple `{timestamp} 47AAC86800430028: join request received`?

In the sample you pasted I have seen only one. Each attempt to join should log it.

zhaozhenhong5, 2019-01-03:
Is this?:
###############
{
  "deviceId": "47AAC86800430028",
  "etag": "AAAAAAAAAAI=",
  "deviceEtag": "MTAyMzc4NzAxNQ==",
  "status": "enabled",
  "statusUpdateTime": "0001-01-01T00:00:00",
  "connectionState": "Disconnected",
  "lastActivityTime": "2019-01-03T08:56:15.1736818",
  "cloudToDeviceMessageCount": 0,
  "authenticationType": "sas",
  "x509Thumbprint": {
    "primaryThumbprint": null,
    "secondaryThumbprint": null
  },
  "version": 32,
  "properties": {
    "desired": {
      "AppEUI": "BE7A0000000014E2",
      "AppKey": "8AFE71A145B253E49C3031AD068277A1",
      "GatewayID": "",
      "SensorDecoder": "DecoderValueSensor",
      "$metadata": {
        "$lastUpdated": "2019-01-03T01:55:16.9398527Z",
        "$lastUpdatedVersion": 2,
        "AppEUI": {
          "$lastUpdated": "2019-01-03T01:55:16.9398527Z",
          "$lastUpdatedVersion": 2
        },
        "AppKey": {
          "$lastUpdated": "2019-01-03T01:55:16.9398527Z",
          "$lastUpdatedVersion": 2
        },
        "GatewayID": {
          "$lastUpdated": "2019-01-03T01:55:16.9398527Z",
          "$lastUpdatedVersion": 2
        },
        "SensorDecoder": {
          "$lastUpdated": "2019-01-03T01:55:16.9398527Z",
          "$lastUpdatedVersion": 2
        }
      },
      "$version": 2
    },
    "reported": {
      "NwkSKey": "D5B85D30D13E08255A23C311AD951F41",
      "AppSKey": "9A67DBD708F0271E68B3F38320FD99EA",
      "DevEUI": "47AAC86800430028",
      "DevAddr": "02F238EF",
      "NetId": "000001",
      "FCntUp": 0,
      "FCntDown": 0,
      "DevNonce": "1ADA",
      "$metadata": {
        "$lastUpdated": "2019-01-03T08:56:18.9863279Z",
        "NwkSKey": {
          "$lastUpdated": "2019-01-03T08:56:18.9863279Z"
        },
        "AppSKey": {
          "$lastUpdated": "2019-01-03T08:56:18.9863279Z"
        },
        "DevEUI": {
          "$lastUpdated": "2019-01-03T08:56:18.9863279Z"
        },
        "DevAddr": {
          "$lastUpdated": "2019-01-03T08:56:18.9863279Z"
        },
        "NetId": {
          "$lastUpdated": "2019-01-03T08:56:18.9863279Z"
        },
        "FCntUp": {
          "$lastUpdated": "2019-01-03T08:56:18.9863279Z"
        },
        "FCntDown": {
          "$lastUpdated": "2019-01-03T08:56:18.9863279Z"
        },
        "DevNonce": {
          "$lastUpdated": "2019-01-03T08:56:18.9863279Z"
        }
      },
      "$version": 30
    }
  },
  "capabilities": {
    "iotEdge": false
  }
}

fbeltrao, 2019-01-03:
The device looks good.
The reported properties also reflect a join at 2019-01-03T08:56:18.9863279Z.

Do you see in the IoT Edge module log multiple entries like "47AAC86800430028: join request received"

zhaozhenhong5, 2019-01-04:
Yes,you can see the messages I sent above.

zhaozhenhong5, 2019-01-04:
I add the whole join fail log here:[https://github.com/zhaozhenhong5/zhao](url), including the "LoRaWanNetworkSrvModule-join fail log.txt"、“LoRaWanPktFwdModule-log.txt”、“my device-join fail log.txt”, and the CN470 region I added replacing the US region.
Can you find out the problem?

fbeltrao, 2019-01-04:
Thanks for the logs. 

I see your device is retrying OTAA join a second after it fails (every 7 seconds). Could you please try to add a 20 second delay between attempts and see if it works (send, fails after 6s, wait another 20s and try again)?

If that does not work we can try a remote session so I can have a better look at the problem. If you are interested please send an email to frbeltra at microsoft.com so I can schedule a Skype call with you.

ronniesa, 2019-01-08:
You should send upstream msg as unconfirmed, right now you sending a confirm message so you are requesting the network server to send back a confirmation msg that fails because the region is not yet supported so we are not able to create the downstream msg. Depending on the device you need to change the code or set a flag to send upstream unconfirmed.

fbeltrao, 2019-01-11:
Closing this as the problem happens because we still do not support CN region.

closed 2019-01-11 by fbeltrao