Azure/iotedge-lorawan-starterkit #77: Receive messages delay issue

zhaozhenhong5, 2019-01-10:
Now I am testing the engine with devices only on ABP mode and have not downstream messages. I am in china and using 470M-510M band. And I found the upstream messages are delayed more came back to my monitor. The detail as below:
1、I use the VScode for device messages monitoring.
2、My devices send messages every 10 second.(three devices)
3、For example, after sending for a few times.Now I am sending the message with cnt number 600,but the monitor message is still show the message with cnt number 520, there are 80 messages delayed.
4、When I stop sending messages, the delayed messages will come back to the monitor for a few times. 
5、I saw the monitor about every 20 second received one message,but I sended every 10 second.

I used the TTN lorawan server before, it can work well on every 1 second messages sending.
So,dose somebody know what is the problem?

fbeltrao, 2019-01-10:
Hi,

Could you please provide your LoRaWanNetworkSrvModule logs?
Which region is your IoT Hub? What is your latency to Azure Cloud?

zhaozhenhong5, 2019-01-10:
Now I found the engine can work well with one device sending messages every 1 second. But 2 or more would show that delay issue.

My IoT Hub region is East US.
OK，you can see my LoRaWanNetworkSrvModule test logs [here:](https://github.com/zhaozhenhong5/zhao/blob/master/LoRaWanNetworkSrvModule%20_2019-01-10_17-05-26%20-%20%E5%89%AF%E6%9C%AC.log)
I need the Azure Cloud latency is as less as possible. In fact,I need about 500 devices work sending messages every 5 second.

fbeltrao, 2019-01-10:
Hi,

Thanks for the information and logs.

We tested the solution with multiple devices and haven't experience the problem you have. In order to help us troubleshoot the problem could you please make a few changes and try again with multiple devices?

**Changes**

1. In your leaf devices please remove the property "SensorDecoder". Your payload seems to be in binary format causing problems in the value decoder "Invalid JSON returned from 'DecoderValueSensor'"

2. Disable for now cloud to device message by editing the class IoTHubConnector replacing `ReceiveAsync` with (we have a pending task to make it easier in the future):
```c#
public Task<Message> ReceiveAsync(TimeSpan timeout) => Task.FromResult((Message)null);
```
3. Build, push and deploy the solution

4. Test with 3 devices are you had before

5. Publish the logs and results

Additional questions:
- You are monitoring a single device using VsCode, aren't you? In that case you would only see the messages of a single one, not from the remaining 2
- Is your intention to connect all 500 devices to a single gateway?

If you wish we can have a call to discuss your scenario and give your more information about the project.

zhaozhenhong5, 2019-01-10:
Yes,I am monitoring a single device using VsCode. I can only see the messages of a single one.
If single gateway can connect 500 devices,that will be good,if not ,that will be two or more.

Thank you.Maybe I would do this job tomorrow because is very late in China now.And my oral English is not so good, so I will call help with my colleague from the ministry of foreign trade if we talk with a call.

Maybe we should add the skype first? This is my skype: tdrj2@todaair.cn. I just installed it,can you add me? Or I can add yours.

zhaozhenhong5, 2019-01-11:
I have changed the code just like this:`        public async Task<Message> ReceiveAsync(TimeSpan timeout)
        {
            return await Task.FromResult((Message)null);
/*             try
            {
                CreateDeviceClient();

                deviceClient.OperationTimeoutInMilliseconds = 1500;

                setRetry(true);

                Logger.Log(DevEUI, $"checking c2d message", Logger.LoggingLevel.Full);

                Message msg = await deviceClient.ReceiveAsync(timeout);

                Logger.Log(DevEUI, $"done checking c2d message", Logger.LoggingLevel.Full);

                setRetry(false);

                return msg;

            }
            catch (Exception ex)
            {
                Logger.Log(DevEUI, $"Could not retrieve c2d message with error: {ex.Message}", Logger.LoggingLevel.Error);
                return null;
            }
   */     }`
But the delay problem can't solved.You can see the log [here](https://github.com/zhaozhenhong5/zhao/blob/master/2019-01-11_11-25-55.log)

I think we should make a call as soon as you online.I can get your email and add your skype(email adress?).

fbeltrao, 2019-01-11:
My email is frbeltra at microsoft dot com

zhaozhenhong5, 2019-01-11:
I have send email to you.Can you received? And added your skype by the email(frbeltra@microsoft.com)

fbeltrao, 2019-01-11:
After our call, switching to IoT Hub direct connection on the Network Server removed the delay.

Follow up: 
 - provide guidelines on how to configure direct connection to IoT Hub
 - implement deactivation flag for C2D message on device level
 - investigate delay in IoT Edge device telemetry

zhaozhenhong5, 2019-01-12:
Yes,thank you. The delay problem is solved perfectly.

closed 2019-01-12 by zhaozhenhong5