Azure/iotedge-lorawan-starterkit #84: Hex messages receive issue

zhaozhenhong5, 2019-01-17:
Can the engine receive the hex type lora payload? I tested it, but seems wrong.

fbeltrao, 2019-01-17:
What is your input and actual/expected metric in IoT Hub?
The built-in decoder (DecoderValueSensor) works if the payload is a number. You might need to extend it by adding your own payload decoder.

The documentation is here: https://github.com/Azure/iotedge-lorawan-starterkit/tree/master/Samples/DecoderSample

zhaozhenhong5, 2019-01-17:
Yes, I just saw this link. I have a question: How can I push DecoderSample to Azure Container?

fbeltrao, 2019-01-17:
If you use the docker extensions in VsCode you can:

1. Right click on the dockerfile arm32v7 (for PI) and click Build Image. Enter the fullname of your Azure container instance: mycontainerinstance.azurecr.io/decodersample:1.0-armv32v7 (for example)

2. In VsCode you can right click on the built image and select "Push Image". The same can be done in command line by executing `docker push  mycontainerinstance.azurecr.io/decodersample:1.0-armv32v7`

zhaozhenhong5, 2019-01-18:
Thank you. I had added the DecoderSample ,but it can only decod when the end-lora-device sending string  not the hex. Can change the decoder to decode hex?

zhaozhenhong5, 2019-01-18:
Yes,thank you. The problems have solved.

closed 2019-01-18 by zhaozhenhong5