Azure/iotedge-lorawan-starterkit #87: How to add two or more edge gateway devices?

zhaozhenhong5, 2019-01-19:
So I want to use two or more edge gateway devices for my lorawan network, how to add them? Should set something different for the Azure?Thank you.

fbeltrao, 2019-01-19:
Currently multi gateway has a few limitations that you should be aware:
https://github.com/Azure/iotedge-lorawan-starterkit#current-limitations

To enable multi gateway, create a new iotedge device and deploy the starter-kit there.

Next you need to choose if leaf devices will be assigned to a single gateway or not. This can be decided on leaf devices level allowing you to have a mix of assigned and not-assigned. To assign a leaf device to a gateway set the "GatewayID" desired property in leaf device twin. The absence of a property value means the leaf device is not assigned to a gateway.

How to decide?
If a device will roam and connect to multiple gateways you should not assign a gateway to it. If they will always stay in same location where the same gateway is you can assign a gateway.

zhaozhenhong5, 2019-01-19:
I created a new iotedge device,and below is the device twin. So what should I change?
###########
{
  "deviceId": "gateway2",
  "etag": "AAAAAAAAAAE=",
  "deviceEtag": "MTA1MjcyMzY1OA==",
  "status": "enabled",
  "statusUpdateTime": "0001-01-01T00:00:00",
  "connectionState": "Disconnected",
  "lastActivityTime": "0001-01-01T00:00:00",
  "cloudToDeviceMessageCount": 0,
  "authenticationType": "sas",
  "x509Thumbprint": {
    "primaryThumbprint": null,
    "secondaryThumbprint": null
  },
  "version": 2,
  "properties": {
    "desired": {
      "$metadata": {
        "$lastUpdated": "2019-01-19T09:15:56.3578265Z"
      },
      "$version": 1
    },
    "reported": {
      "$metadata": {
        "$lastUpdated": "2019-01-19T09:15:56.3578265Z"
      },
      "$version": 1
    }
  },
  "capabilities": {
    "iotEdge": true
  },
  "deviceScope": "ms-azure-iot-edge://zhao-636834861563578265"
}

fbeltrao, 2019-01-19:
The iotedge device twin has no changes.

The iotedge network server module twin should have same properties as in the first iotedge (2 desired properties defining the function). Something like this:

```json
"desired": {
      "FacadeServerUrl": "https://<your-function>.azurewebsites.net/api/",
      "FacadeAuthCode": "<your-function-auth-code>",
```

Don't forget to install all modules in new iotedge device too (pktforwarder, netwrksrv, decoder...)

zhaozhenhong5, 2019-01-22:
Above you said "To assign a leaf device to a gateway set the "GatewayID" desired property in leaf device twin".But after you said "The iotedge device twin has no changes." 
I don't get it .

fbeltrao, 2019-01-22:
There two types of device in IoT Hub
- IoT Devices: those are the leaf devices, in this case the ones using LoRa to communicate with the packet forwarder/network server
- IoT Edge: those are the ones running the IoT Edge runtime, and in this particular case the LoRa Network Server and packet forwarder. Let's call it "gateway".

In the IoT Device twin you can define a fix gateway that will be processing the device messages. In that case other gateways will ignore messages from it. Not assigning a gateway will cause all gateways to process the message of the device.

Hope it is clearer now.

zhaozhenhong5, 2019-01-23:
I think it did have some problem with using more than one gateway. This is the issue log occur in one of my gateway when I am using two.[issue](https://github.com/zhaozhenhong5/zhao/blob/master/two%20gateway%20issue%20log.txt)

When I am using multi gateway, sometimes will show that issue in one gateway, sometimes will not.

fbeltrao, 2019-01-23:
Looking at the logs it seems as the leaf device twin does not have a value for desired.AppSKey. Can you please verify that it has the require desired properties for ABP (AppSKey, NwkSKey, DevAddr)?
Feel free to send me the twin by Email.

A few questions about your test environment:
- Are you using the same IoT Hub for all devices (IoT Edge, IoT Device)?
- Do you have 2 distinct IoT Edge devices for each LoRa gateway?
- What did you decide to use multi gateways? What is the problem you are trying to solve?

zhaozhenhong5, 2019-01-24:
OK,I have sent to you.

fbeltrao, 2019-03-30:
I am closing this as release 1.0.0 includes a more robust message processing, handling upstreams bursts better than previously.

closed 2019-03-30 by fbeltrao