Azure/iotedge-lorawan-starterkit #96: Question:  OS for AAEON AIOT-ILRA01 LoRa® Certified Intel® Based Gateway and Network Server

gerfen, 2019-01-24:
Hi,

I just received an AAEON AIOT-ILRA01 gateway configured for US 915 Mhz.  I'm in the process of configuring the gateway and I was surprised to see that I have a choice to boot into Windows 8.1 or Android.  I expected to see some flavor of Linux.  Could you provide an overview on how your team configured 868 Mhz unit?

Thanks!

--Michael

ronniesa, 2019-01-25:
Our test device came without OS, we installed Ubuntu 16.04. There is a guide at the bottom of the page here that can help you: https://www.aaeon-shop.com/internet-of-things/52-indoor-lora-alliance-certified-gateway-and-network-server.html and in particular: https://wiki.up-community.org/Ubuntu I think they will offer in future preconfigured gateway. For the pktfwr you will need to check what spidev it needs and what reset pin because I think the 915 module is different that the 868 that we have. In case that you get stuck on the pktfwr let us know we can ask AAEON for the correct settings.

marcoba81, 2019-01-25:
dear all,
this is Marco Barbato from AAEON Europe.
The AIOT-ILRA01 is x86 based gateway, so it supports several OS.
Regarding Linux we do validated it with Ubuntu and Ubilinux (Debian).
The unit you have is a prototype with 915MHz radio, while standard version is based on 868MHz and on a different radio module.
However our SW team did validate it.

if you start from scratch, please follow the guides mentioned by @ronniesa that will explain how to install Ubuntu, while for Ubilinux read here https://wiki.up-community.org/Ubilinux
After to have installed the OS, there are additional steps to install and configure the Semtech packet forwarder and for this you can read the user manual, from page 20 (link here https://www.aaeon-shop.com/index.php?controller=attachment&id_attachment=72)
The instructions about spidev and reset IRQ are the same, while if you want to test the radio using the instructions reported there, you have to change the frequency plan.

let me know if you need further assistance

gerfen, 2019-01-25:
@ronniesa  and @marcoba81  Thanks for the great information!  I'll let you know if I have further questions.

ronniesa, 2019-01-27:
For our solution no need to install the pktfwrd on the machine just build, push and deploy the amd64 version of the pktfwrd module in our solution.

gerfen, 2019-01-27:
Thanks again.  Closing.

closed 2019-01-27 by gerfen

marcoba81, 2019-01-27:
> Ah. Thanks for the clarification. That bit of knowledge will come in handy
> when I set up the next gateway. :)

Sorry Michael,
I missed to mention that one I suggested was the instruction path for the
basic general configuration stand alone.

With this kit, once in its final version, the idea is to grab a disk image
and offer to customer few steps configuration (still to be discussed), as
we usually do

Regards