Azure/iotedge-lorawan-starterkit #100: Join failures

gerfen, 2019-01-28:
Using the code form the current master branch (0.3.0)  I am receiving `Error processing the message The given key '6' was not present in the dictionary. ` for devices from one particular device manufacturer.  I asked the manufacturer to set up the devices with this frequency plan[1].  I'm trying to determine if there is something incorrectly configured on the devices or if there is a bug in the gateway code?  Here is a sampling of the logs[2].

Any insight as to where the issue lies would be much appreciated.

Thanks!

--Michael

[1] US frequency plan
````
 {
    "SX1301_conf": {
             "lorawan_public": true,
        "clksrc": 1, /* radio_1 provides clock to concentrator */
        "antenna_gain": 0, /* antenna gain, in dBi */
        "radio_0": {
            "enable": true,
            "type": "SX1257",
            "freq": 902700000,
            "rssi_offset": -166.0,
            "tx_enable": true,
            "tx_freq_min": 902000000,
            "tx_freq_max": 928000000
        },
        "radio_1": {
            "enable": true,
            "type": "SX1257",
            "freq": 903400000,
            "rssi_offset": -166.0,
            "tx_enable": false
        },
        "chan_multiSF_0": {
            /* Lora MAC channel, 125kHz, all SF, 902.3 MHz */
            "enable": true,
            "radio": 0,
            "if": -400000
        },
        "chan_multiSF_1": {
            /* Lora MAC channel, 125kHz, all SF, 902.5 MHz */
            "enable": true,
            "radio": 0,
            "if": -200000
        },
        "chan_multiSF_2": {
            /* Lora MAC channel, 125kHz, all SF, 902.7 MHz */
            "enable": true,
            "radio": 0,
            "if": 0
        },
        "chan_multiSF_3": {
            /* Lora MAC channel, 125kHz, all SF, 902.9 MHz */
            "enable": true,
            "radio": 0,
            "if": 200000
        },
        "chan_multiSF_4": {
            /* Lora MAC channel, 125kHz, all SF, 903.1 MHz */
            "enable": true,
            "radio": 1,
            "if": -300000
        },
        "chan_multiSF_5": {
            /* Lora MAC channel, 125kHz, all SF, 903.3 MHz */
            "enable": true,
            "radio": 1,
            "if": -100000
        },
        "chan_multiSF_6": {
            /* Lora MAC channel, 125kHz, all SF, 903.5 MHz */
            "enable": true,
[... 171 lines cut ...]
2019-01-28 20:48:41.379 Physical dataUp {"rxpk":[{"tmst":3877001644,"chan":8,"rfch":0,"freq":903.000000,"stat":1,"modu":"LORA","datr":"SF8BW500","codr":"4/5","lsnr":10.2,"rssi":-34,"size":23,"data":"AAMAABECxSYsQAMGEQJ3SgCJRmabVd8="}]}
2019-01-28 20:48:41.379 004A770211060340: join request received
2019-01-28 20:48:41.380 004A770211060340: querying the registry for device key
2019-01-28 20:48:41.441 004A770211060340: using edgeHub local queue
2019-01-28 20:48:41.441 004A770211060340: using cached twins for OTAA device
2019-01-28 20:48:41.445 004A770211060340: saving join properties twins
2019-01-28 20:48:41.701 004A770211060340: done saving join properties twins
2019-01-28 20:48:41.706 Error processing the message The given key '6' was not present in the dictionary.,    at System.Collections.Generic.Dictionary`2.get_Item(TKey key)
   at LoRaTools.Regions.Region.GetDownstreamDR(Rxpk upstreamChannel) in /app/LoraTools/Regions/Region.cs:line 137
   at LoRaWan.NetworkServer.MessageProcessor.ProcessJoinRequest(LoRaMessageWrapper loraMessage) in /app/LoRaWan.NetworkServer/MessageProcessor.cs:line 613
   at LoRaWan.NetworkServer.MessageProcessor.ProcessMessageAsync(Byte[] message) in /app/LoRaWan.NetworkServer/MessageProcessor.cs:line 55
   at LoRaWan.NetworkServer.UdpServer.<>c__DisplayClass10_1.<<RunUdpListener>b__0>d.MoveNext() in /app/LoRaWan.NetworkServer/UdpServer.cs:line 105
2019-01-28 20:48:45.194 Statistic: {"stat":{"time":"2019-01-28 20:48:45 GMT","rxnb":2,"rxok":2,"rxfw":2,"ackr":100.0,"dwnb":1,"txnb":1}}
2019-01-28 20:48:49.525 Physical dataUp {"rxpk":[{"tmst":3885147356,"chan":5,"rfch":1,"freq":903.300000,"stat":1,"modu":"LORA","datr":"SF9BW125","codr":"4/5","lsnr":12.5,"rssi":-34,"size":23,"data":"AAMAABECxSYsQAMGEQJ3SgB7totqHk0="}]}
2019-01-28 20:48:49.526 004A770211060340: join request received
2019-01-28 20:48:49.526 004A770211060340: querying the registry for device key
2019-01-28 20:48:49.579 004A770211060340: using edgeHub local queue
2019-01-28 20:48:49.579 004A770211060340: using cached twins for OTAA device
2019-01-28 20:48:49.580 004A770211060340: saving join properties twins
2019-01-28 20:48:49.851 004A770211060340: done saving join properties twins
2019-01-28 20:48:49.852 9F0E6502: JoinAccept {"txpk":{"imme":false,"data":"IJlFzfXgCeAEZ9kN2N7rtuQ=","tmst":3890147356,"size":17,"freq":926.3,"rfch":0,"modu":"LORA","datr":"SF11BW500","codr":"4/5","powe":14,"ipol":true}}
2019-01-28 20:48:49.853 004A770211060340: join accept sent with ID 28CD
2019-01-28 20:48:49.853 004A770211060340: processing time: 00:00:00.3283120
2019-01-28 20:48:49.855 Packet with id 28CD successfully transmitted by the aggregator
````

ronniesa, 2019-01-29:
It seems a bug from our side on how we set the downstream data rate but we need a bit of time to investigate on the issue. 
Just to be sure you did not change the global_config from your working solution?

We keep you up to date.

gerfen, 2019-01-29:
@ronniesa I just confirmed that the global_conf.us..config file is the same as that in the head.  Please let me know if I can provide anymore information.

[image]

ronniesa, 2019-01-29:
Yep, is in our part the bug. We will probably have a fix at the beginning of next week

gerfen, 2019-01-29:
I received another set of devices from a different manufacturer yesterday and just had a chance to set them up this morning.  They fail to join the network as well but with a different error message[1].

I've doubled checked the AppKey and the AppEUI to confirm they are correct.  Is this related to the same bug?

Thanks.

[1] Logs
````
2019-01-29 16:59:36.959 Physical dataUp {"rxpk":[{"tmst":3518102372,"chan":5,"rfch":1,"freq":903.300000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":12.5,"rssi":-25,"size":23,"data":"AAAAAAAQehMAMygAABB6EwD/LmeVGGw="}]}
2019-01-29 16:59:36.959 00137A1000002833: join request received
2019-01-29 16:59:36.960 00137A1000002833: querying the registry for device key
2019-01-29 16:59:37.014 00137A1000002833: using edgeHub local queue
2019-01-29 16:59:37.015 00137A1000002833: using cached twins for OTAA device
2019-01-29 16:59:37.015 00137A1000002833: AppEUI for OTAA does not match for device
2019-01-29 16:59:37.016 00137A1000002833: join request refused
2019-01-29 16:59:37.016 00137A1000002833: processing time: 00:00:00.0572299
2019-01-29 16:59:46.465 Statistic: {"stat":{"time":"2019-01-29 16:59:46 GMT","rxnb":3,"rxok":2,"rxfw":2,"ackr":100.0,"dwnb":0,"txnb":0}}
2019-01-29 17:00:16.461 Statistic: {"stat":{"time":"2019-01-29 17:00:16 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}
````

ronniesa, 2019-01-29:
Some devices like the ones that use LMIC lib expect the key entered reversed (little endian) this could be the issue. Are the devices the same or just same manufacturer? You can try to reverse the AppEUI on the device and see if the problem persist.

gerfen, 2019-01-29:
They are the same device.  I tried to clear the cache on the gateway but that didn't seem to have any effect.

gerfen, 2019-01-30:
I tried to reverse the AppEUI for my failing device but it still fails[1]
.  FWIW, I have been chatting with @Mandur and we have introduced a change to the Region class which allows more sensors  from one manufactuere to successfully join but not the other.  The changes to REgion.cs that I have deployed are here:  https://github.com/Azure/iotedge-lorawan-starterkit/tree/mik-dev-0.3.0-preview-us-dr-fix.  

FWIW, The change to Region.cs has allowed one of my ThingsNode sensors which previously required me to change the entire frequency plan for the gateway to suddenly start connecting albeit with a symmetrical AppKey and AppEUI.  

Let me know if there is some other experiment you would like me to try.

````
2019-01-30 21:55:59.246 Facade function url: https://evolutioniothubfunction.azurewebsites.net/api/
2019-01-30 21:55:59.556 LoRaWAN server started on port 1680
2019-01-30 21:56:21.500 Statistic: {"stat":{"time":"2019-01-30 21:56:21 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}
2019-01-30 21:56:29.983 Physical dataUp {"rxpk":[{"tmst":2251176527,"chan":8,"rfch":0,"freq":903.000000,"stat":1,"modu":"LORA","datr":"SF8BW500","codr":"4/5","lsnr":10.0,"rssi":-34,"size":23,"data":"AAMAABECxSYsQAMGEQJ3SgD4NJtaemI="}]}
2019-01-30 21:56:30.124 004A770211060340: join request received
2019-01-30 21:56:30.164 004A770211060340: querying the registry for device key
2019-01-30 21:56:30.994 004A770211060340: using edgeHub local queue
2019-01-30 21:56:31.043 004A770211060340: getting twins for OTAA for device
2019-01-30 21:56:31.048 004A770211060340: getting twins
2019-01-30 21:56:38.141 Physical dataUp {"rxpk":[{"tmst":2259333596,"chan":6,"rfch":1,"freq":903.500000,"stat":1,"modu":"LORA","datr":"SF9BW125","codr":"4/5","lsnr":12.8,"rssi":-35,"size":23,"data":"AAMAABECxSYsQAMGEQJ3SgD/ed0xw5I="}]}
2019-01-30 21:56:38.141 004A770211060340: join request received
2019-01-30 21:56:38.141 004A770211060340: querying the registry for device key
2019-01-30 21:56:38.240 004A770211060340: using edgeHub local queue
2019-01-30 21:56:38.242 004A770211060340: getting twins for OTAA for device
2019-01-30 21:56:38.243 004A770211060340: getting twins
2019-01-30 21:56:45.935 Physical dataUp {"rxpk":[{"tmst":2267134403,"chan":8,"rfch":0,"freq":903.000000,"stat":1,"modu":"LORA","datr":"SF8BW500","codr":"4/5","lsnr":9.5,"rssi":-34,"size":23,"data":"AAMAABECxSYsQAMGEQJ3SgCiGiICwIA="}]}
2019-01-30 21:56:45.935 004A770211060340: join request received
2019-01-30 21:56:45.935 004A770211060340: querying the registry for device key
2019-01-30 21:56:46.052 004A770211060340: using edgeHub local queue
2019-01-30 21:56:46.053 004A770211060340: getting twins for OTAA for device
2019-01-30 21:56:46.054 004A770211060340: getting twins
2019-01-30 21:56:51.465 Statistic: {"stat":{"time":"2019-01-30 21:56:51 GMT","rxnb":3,"rxok":3,"rxfw":3,"ackr":100.0,"dwnb":0,"txnb":0}}
2019-01-30 21:56:54.099 Physical dataUp {"rxpk":[{"tmst":2275290980,"chan":5,"rfch":1,"freq":903.300000,"stat":1,"modu":"LORA","datr":"SF9BW125","codr":"4/5","lsnr":12.8,"rssi":-36,"size":23,"data":"AAMAABECxSYsQAMGEQJ3SgDkrafgABg="}]}
2019-01-30 21:56:54.099 004A770211060340: join request received
2019-01-30 21:56:54.099 004A770211060340: querying the registry for device key
2019-01-30 21:56:54.162 004A770211060340: using edgeHub local queue
2019-01-30 21:56:54.163 004A770211060340: getting twins for OTAA for device
2019-01-30 21:56:54.163 004A770211060340: getting twins
2019-01-30 21:57:01.891 Physical dataUp {"rxpk":[{"tmst":2283091295,"chan":8,"rfch":0,"freq":903.000000,"stat":1,"modu":"LORA","datr":"SF8BW500","codr":"4/5","lsnr":10.5,"rssi":-36,"size":23,"data":"AAMAABECxSYsQAMGEQJ3SgDxW4PDWG0="}]}
2019-01-30 21:57:01.891 004A770211060340: join request received
2019-01-30 21:57:01.891 004A770211060340: querying the registry for device key
2019-01-30 21:57:02.017 004A770211060340: using edgeHub local queue
2019-01-30 21:57:02.018 004A770211060340: getting twins for OTAA for device
2019-01-30 21:57:02.018 004A770211060340: getting twins
2019-01-30 21:57:11.346 004A770211060340: done getting twins
2019-01-30 21:57:11.351 004A770211060340: done getting twins
2019-01-30 21:57:11.352 004A770211060340: done getting twins
2019-01-30 21:57:11.359 004A770211060340: done getting twins
2019-01-30 21:57:11.446 004A770211060340: done getting twins for OTAA device
2019-01-30 21:57:11.448 004A770211060340: done getting twins for OTAA device
2019-01-30 21:57:11.449 004A770211060340: done getting twins for OTAA device
2019-01-30 21:57:11.450 004A770211060340: done getting twins for OTAA device
2019-01-30 21:57:11.657 004A770211060340: processing of the join request took too long, sending no message
2019-01-30 21:57:11.657 004A770211060340: processing of the join request took too long, sending no message
2019-01-30 21:57:11.658 004A770211060340: processing of the join request took too long, sending no message
2019-01-30 21:57:11.659 004A770211060340: processing of the join request took too long, sending no message
2019-01-30 21:57:11.664 004A770211060340: saving join properties twins
2019-01-30 21:57:11.664 004A770211060340: saving join properties twins
2019-01-30 21:57:11.665 004A770211060340: saving join properties twins
2019-01-30 21:57:11.665 004A770211060340: saving join properties twins
2019-01-30 21:57:12.260 004A770211060340: done saving join properties twins
[... cut ...]
2019-01-30 21:57:15.727 004A770211060340: could not save twins with error: Operation timeout expired.
2019-01-30 21:57:15.727 004A770211060340: join request could not save twins, join refused
2019-01-30 21:57:15.727 004A770211060340: join request could not save twins, join refused
2019-01-30 21:57:21.462 Statistic: {"stat":{"time":"2019-01-30 21:57:21 GMT","rxnb":3,"rxok":2,"rxfw":2,"ackr":100.0,"dwnb":1,"txnb":0}}
2019-01-30 21:57:51.466 Statistic: {"stat":{"time":"2019-01-30 21:57:51 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}
````

ronniesa, 2019-01-30:
I belive AppEUI and DevEUI need to be reversed but not AppKey for devices that are little endians

Mandur, 2019-01-31:
This was due to a wrong implementation in data rate mapping in US region. I will investigate to see if there are other glitches next week. 
@gerfen thank you for your continuous support in the project! Let us know if this issue still occur in some cases or if there are other problems rising in the US region.

gerfen, 2019-02-02:
I received the following information from the manufactuer:  "The device is big endian,and based on loramac-node,not lmic".  So I think there's still some work to do with mapping the frequencies correctly for the US frequency plan.  I'm willing to help out however I can to get this resolved.

Mandur, 2019-02-05:
@gerfen if you could try the new branch and see if the issue occurs again, that would be great!

Mandur, 2019-02-07:
After the remote session we were able to join the additional device. It was as expected probably an endianess problem. 
@gerfen please let us know if you encounter any other issues in the US frequencies & thank you for your help

closed 2019-02-08 by Mandur