Azure/iotedge-lorawan-starterkit #108: Question:  Cloud to Device messages

gerfen, 2019-02-02:
I would like to adjust the reporting frequency from a couple of the devices I have joined to my gateway.  The documentation I have indicates that I should send "set period 60 seconds + save" to set the reporting frequency to 60 seconds.   It's not clear to me how send this command through the Azure IoT Hub, realizing that I may need to send multiple commands.

Any insight would be appreciated.  Note:  I have the ability to do this via a USB to  serial adapter but would like to understand how to achieve the same result through C2D messaging.

Thanks!

Mandur, 2019-02-05:
@gerfen I am not sure I fully understand your question, let's discuss this issue over the phone thursday

Mandur, 2019-02-07:
More tests will be conducted by @gerfen with C2D messaging that might lead to another session. I let the issue open in the meanwhile

gerfen, 2019-02-08:
I was doing to things incorrectly.  One I was using "Direct Message" instead of "Message to device".  I also misread the documentation for the device I am trying to configure.  The device is expecting a message in the format of |Command| Parameter| CRC |.  With those issues resolved I ended up sending the command "0002003CF5A1" (without the quotes).  Unfortunately this did not work as I received the following in the gateway logs:

```` 
2019-02-08 00:48:47.727 0004A30B0021E4E2: valid frame counter, msg: 226 server: 225
2019-02-08 00:48:47.728 0004A30B0021E4E2: decoding with: http://decodersample/api/DecoderDecentlabWaterLevelSensor port: 1
2019-02-08 00:48:47.736 0004A30B0021E4E2: sending message {"time":null,"tmms":0,"tmst":3085307900,"freq":902.5,"chan":1,"rfch":0,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","rssi":-30,"lsnr":9.2,"size":24,"data":{"deviceId":"2570","battery":3073,"pressure":-0.001251220703125,"temperature":15.78125},"port":1,"fcnt":226,"eui":"0004A30B0021E4E2","gatewayid":"baileynursery-production-gateway","edgets":1549586927726} to hub
2019-02-08 00:48:47.760 0004A30B0021E4E2: sent message {"time":null,"tmms":0,"tmst":3085307900,"freq":902.5,"chan":1,"rfch":0,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","rssi":-30,"lsnr":9.2,"size":24,"data":{"deviceId":"2570","battery":3073,"pressure":-0.001251220703125,"temperature":15.78125},"port":1,"fcnt":226,"eui":"0004A30B0021E4E2","gatewayid":"baileynursery-production-gateway","edgets":1549586927726} to hub
2019-02-08 00:48:47.760 0004A30B0021E4E2: message '{"deviceId":"2570","battery":3073,"pressure":-0.001251220703125,"temperature":15.78125}' sent to hub
2019-02-08 00:48:47.760 0004A30B0021E4E2: checking c2d message
2019-02-08 00:48:47.761 0004A30B0021E4E2: done checking c2d message
2019-02-08 00:48:47.761 0004A30B0021E4E2: checking c2d message
2019-02-08 00:48:47.765 0004A30B0021E4E2: done checking c2d message
2019-02-08 00:48:47.765 0004A30B0021E4E2: abbandoning c2d message
2019-02-08 00:49:05.447 Statistic: {"stat":{"time":"2019-02-08 00:49:05 GMT","rxnb":2,"rxok":1,"rxfw":1,"ackr":100.0,"dwnb":0,"txnb":0}}
2019-02-08 00:49:17.772 0004A30B0021E4E2: could not abbandon c2d with error: Operation timeout expired.
2019-02-08 00:49:17.773 0004A30B0021E4E2: Additional C2D messages waiting, setting FPending to 1
2019-02-08 00:49:17.773 0004A30B0021E4E2: C2D message: 0002003CF5A1
2019-02-08 00:49:17.773 0004A30B0021E4E2: abbandoning c2d message
2019-02-08 00:49:17.799 0004A30B0021E4E2: could not abbandon c2d with error: 'receiver181' is closed
2019-02-08 00:49:17.799 0004A30B0021E4E2: too late for down message, sending only ACK to gateway
2019-02-08 00:49:17.799 0004A30B0021E4E2: processing time: 00:00:30.0734066

````
````
2019-02-08 00:58:54.099 0004A30B0021E4E2: using edgeHub local queue
2019-02-08 00:58:54.100 0004A30B0021E4E2: getting twins
2019-02-08 00:58:54.698 0004A30B0021E4E2: done getting twins
2019-02-08 00:58:54.699 0004A30B0021E4E2: done getting twins
2019-02-08 00:58:54.699 0004A30B0021E4E2: processing message, device not in cache
2019-02-08 00:58:54.700 0004A30B0021E4E2: valid frame counter, msg: 227 server: 220
2019-02-08 00:58:54.701 0004A30B0021E4E2: decoding with: http://decodersample/api/DecoderDecentlabWaterLevelSensor port: 1
2019-02-08 00:58:54.743 0004A30B0021E4E2: sending message {"time":null,"tmms":0,"tmst":345493804,"freq":902.9,"chan":3,"rfch":0,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","rssi":-47,"lsnr":8.2,"size":24,"data":{"deviceId":"2570","battery":3073,"pressure":-0.001220703125,"temperature":15.85},"port":1,"fcnt":227,"eui":"0004A30B0021E4E2","gatewayid":"baileynursery-production-gateway","edgets":1549587533956} to hub
2019-02-08 00:58:54.771 0004A30B0021E4E2: sent message {"time":null,"tmms":0,"tmst":345493804,"freq":902.9,"chan":3,"rfch":0,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","rssi":-47,"lsnr":8.2,"size":24,"data":{"deviceId":"2570","battery":3073,"pressure":-0.001220703125,"temperature":15.85},"port":1,"fcnt":227,"eui":"0004A30B0021E4E2","gatewayid":"baileynursery-production-gateway","edgets":1549587533956} to hub
2019-02-08 00:58:54.771 0004A30B0021E4E2: message '{"deviceId":"2570","battery":3073,"pressure":-0.001220703125,"temperature":15.85}' sent to hub
2019-02-08 00:58:54.772 0004A30B0021E4E2: checking c2d message
2019-02-08 00:58:55.035 0004A30B0021E4E2: done checking c2d message
2019-02-08 00:58:55.044 0004A30B0021E4E2: checking c2d message
2019-02-08 00:58:55.049 0004A30B0021E4E2: done checking c2d message
2019-02-08 00:58:55.053 0004A30B0021E4E2: abbandoning c2d message
2019-02-08 00:58:55.412 0004A30B0021E4E2: done abbandoning c2d message
2019-02-08 00:58:55.412 0004A30B0021E4E2: Additional C2D messages waiting, setting FPending to 1
2019-02-08 00:58:55.415 0004A30B0021E4E2: C2D message: 0002003CF5A1
2019-02-08 00:58:55.415 0004A30B0021E4E2: down frame counter: 1
2019-02-08 00:58:55.433 0004A30B0021E4E2: using standard second receive windows
2019-02-08 00:58:55.433 0004A30B0021E4E2: Cloud to device message with a Fport of 1
2019-02-08 00:58:55.434 0004A30B0021E4E2: Sending a downstream message with ID AFC9
2019-02-08 00:58:55.469 026D7217: UnconfirmedDataDown {"txpk":{"imme":false,"data":"YBdybQIQAQABS/RugLcz9flu1ODHyG0FfA==","tmst":347493804,"size":25,"freq":923.3,"rfch":0,"modu":"LORA","datr":"SF12BW500","codr":"4/5","powe":14,"ipol":true}}
2019-02-08 00:58:55.474 0004A30B0021E4E2: completing c2d message
[... cut ...]
2019-02-08 01:09:22.497 0004A30B0021E4E2: Additional C2D messages waiting, setting FPending to 1
2019-02-08 01:09:22.498 0004A30B0021E4E2: C2D message: 0002003CF5A1
2019-02-08 01:09:22.499 0004A30B0021E4E2: abbandoning c2d message
2019-02-08 01:09:24.610 0004A30B0021E4E2: could not abbandon c2d with error: Exception of type 'Microsoft.Azure.Devices.Client.Exceptions.DeviceMessageLockLostException' was thrown.
2019-02-08 01:09:24.610 0004A30B0021E4E2: too late for down message, sending only ACK to gateway

````
I'll try to gather more data as time permits.

ronniesa, 2019-02-08:
Let us check, I think we may have a problem in how we format the msg in case you send down HEX. Let you know

Mandur, 2019-02-08:
@ronniesa is correct. We identified this as a bug. We will connect back with you to test it out.

gerfen, 2019-02-08:
@mandur and @ronniesa Thanks for the update and bug confirmation.

gerfen, 2019-02-13:
Working with @Mandur I was able to send messages to one class of devices.  We were able to do this by replacing line 446 of MessageProcessor.cs
````
 frmPayload = cloudToDeviceMessage?.GetBytes();
````
with this
````
 if (cloudToDeviceMessage.Properties.TryGetValueCaseInsensitive("hex", out var hex))
 {
      frmPayload = ConversionHelper.StringToByteArray(System.Text.Encoding.UTF8.GetString(cloudToDeviceMessage?.GetBytes()));
 }
 else
 {
       frmPayload = cloudToDeviceMessage?.GetBytes();
 }
````

This requires sending a property named 'hex' set to 1, e.g.
[image]

I am in the process of putting together a PR but it will be some time next week before I can submit it as I want to do more testing.

Mandur, 2019-02-16:
Thanks for the suggestion and help @gerfen. I know @fbeltrao is working on a C2D message refactoring. It might be good to sync to avoid duplicate or conflicting work.

fbeltrao, 2019-02-18:
I am working on cloud to device message changes. The idea is to use JSON to represent the message to send downstream instead of properties. You will also have the option to send text or raw payload (in base64 format).

The idea is something like this, you either send `payload` or `rawPayload`:
```json
{
  "fport": integer,
  "rawPayload": "base64_bytes",
  "payload": "utf8 payload",
  "confirmed": boolean,
}
```

gerfen, 2019-02-19:
@fbeltrao  This looks great.  I have some sensors which I have not been able to configure with the current 0.4.0-preview code on the master branch.  I can share the documentation with you if that would be helpful.

fbeltrao, 2019-02-20:
@gerfen as mentioned before the C2D currently available does not support binary (hex) as downstream. Once the changes I am working on becomes available on dev branch I'd appreciate if you had the time to give it a go and let us know if it works for you.

gerfen, 2019-02-21:
@fbeltrao I'm more than happy to help test the new C2D functionality.  Just let me know when it's ready.  :)

fbeltrao, 2019-03-05:
@gerfen we have added to the dev branch the bits to of the new cloud to device messaging. More information can be found here: 
https://github.com/Azure/iotedge-lorawan-starterkit/blob/dev/Docs/quickstart.md#cloud-to-device-message

It would be great if you could give it a try and share your experience.

gerfen, 2019-03-05:
@fbeltrao I'll test out the new functionality just as soon as I can unfortunately it's going to be harder than I anticipated as I have had a few set backs with procuring new hardware.  I just found out that the sensors I ordered will not arrive until April - they were supposed to arrive last week :(.  I'm a little reluctant to try out the new code on the sensors I have already deployed to production.  I may be able to a temporary deployment over the weekend, test and then roll back.  How stable it the code in the dev branch in relationship to what's currently in main?

fbeltrao, 2019-03-06:
@gerfen First, I don't want to give you too much work, causing you to temper with your production environment with a less stable version of the kit. 

Even though changes to dev are only accepted if they pass our CI tests (which include tests against actual hardware), it does have a higher chance of having bugs. When we make a release, updating master, it means we have been running ourselves for a while, thus reducing the chance of something have slipped.  So my recommendation would be to use dev if you need a feature that is available there and have additional time to react in case something is not working properly.

Having said that, the changes we are introducing in cloud to device messages are not backward compatible. The way you send messages with the current master branch won't be supported in next release.

fbeltrao, 2019-03-30:
I am closing as the new changes to cloud to device message have been released.

closed 2019-03-30 by fbeltrao

gerfen, 2019-07-17:
@fbeltrao Just wanted to let you know that I was finally able to test sending  C2D hex messages and I had no issues.