Azure/iotedge-lorawan-starterkit #129: Error building image for decoder samples on master branch

gerfen, 2019-02-13:
I've added code to support decoding for more device to the Samples project -> LoraDecoder.cs.  I can build just fine but when I go to build the image I'm getting this error[1].  I get the same error for amd64 and arm32v7 images.  FWIW, I'm developing on Windows 10.  This appears to be a regression from the 0.3.0 release.  Please advise.

Thanks!

[1] Build image error
````
PS D:\dev\GitHub\iotedge-lorawan-starter\iotedge-lorawan-starterkit> docker build --rm -f "Samples\DecoderSample\Dockerfile.arm32v7" -t <my repository>.azurecr.io/decodersample:1.1.8.arm32v7 Samples\DecoderSample
Sending build context to Docker daemon  33.79kB
Step 1/9 : FROM microsoft/dotnet:2.1-sdk AS build-env
 ---> 6e1c48456450
Step 2/9 : WORKDIR /build/Samples/DecoderSample/
 ---> Using cache
 ---> e496422db033
Step 3/9 : COPY ./Samples/DecoderSample ./
COPY failed: stat /var/lib/docker/tmp/docker-builder882806683/Samples/DecoderSample: no such file or directory
PS D:\dev\GitHub\iotedge-lorawan-starter\iotedge-lorawan-starterkit>
````

ronniesa, 2019-02-13:
Open **only** the decodersample folder in VS code : iotedge-lorawan-starterkit\Samples\DecoderSample and build from there. I just did it and it works. We threat now the decoder as an external project so it doesn't need to adhere to our stylecop rules.

gerfen, 2019-02-13:
Hmm.  I'm still getting the same error.  
[image]

I think I need to go back and read the instructions again.  Unfortunately I won't have any more time to do that today.

ronniesa, 2019-02-13:
You should build with right clicking on module.json "Build IoT Edge Module...." and then pick the platform, Building from dockerfile fails for me too

gerfen, 2019-02-13:
Yep. The instructions changed a bit since the last release - shame on me for not reading them fully. I was able to build and push "Build and Push IoT Edge Module Image".  Thanks for the help.  Sorry for the fire drill!

closed 2019-02-13 by gerfen