Azure/iotedge-lorawan-starterkit #137: Newbie Questions

Ilheu, 2019-02-21:
Hi, I'm new to Azure Iot and I would like to know the answer to some questions, if possible:
1) I'm trying to run your Lorawan modules in a Iot Edge device with windows containers, is it possible ? I builded to windows-amd64, and changed the start_pktfwd.sh to CRLF ... The modules don't even appear in my Container Registry.
2) You use Redis Cache in your architecture, is it needed or just advised ? Is there any alternative way to store the keys ?
3) Can I use an Iot edge device just to send Lorawan Packets to The Things Network Server outside Iot Edge  (Maybe like your LoRaWanPktFwdModule module)  ? If so, how can I implement it ?

ronniesa, 2019-02-22:
1, Currently only Linux is supported because Windows support for IoT Edge is still in preview: https://docs.microsoft.com/en-us/azure/iot-edge/support For the moment we don't have the intention to test on Windows the LoRa Edge module. This could change in the future based on user requests. Is Linux not an option for your gateway? 
With the new support on Windows for Container to access SPI is in theory possible to run our solution on Windows. The challenging part is to run the packet forwarder in a container on windows that requires some extensive testing. Currently almost all the commercial gateway that have LoRa concentrator and can run IoT Edge are Linux based this is why we focus first on Linux.

2. The key role of Redis in our solution is to synchronize multi gateway, allow deduplication and optimize performance. Due to the low cost and low maintenance of Azure Redis we relay on it and it is required. On a single gateway usage you could in theory skip Redis but we currently do not allow this. What is your concern with Redis?

3. Not sure I understand this point but I try to answer. 
Our solution allow you to directly process lora packets at the Edge and send the data ready to use to IoT Hub, it does not require any additional software layer or third party component. This allow also to do edge processing and in future also work partially offline.
Regarding TTN, as we do in our solution it is possible to change the configuration of the semtech packet forwarder to talk to an external network server like Loriot, TTN etc... you do this by changing the local_config on the container. You could also look on how we build the container and do the same with the TTN packet forwarder. Not sure if it make sense to run IoT Edge just to run the packet forwarder in this case, the only benefits that I see that you can update and manage the packet forwarder centrally.

Ilheu, 2019-02-22:
Thank you so much for your quick reply @ronniesa ! I don't know if creating issues here is the best way to ask this kind of questions, is there any way to ask you some future thoughts ?

> Is Linux not an option for your gateway?

It could be, my device is running windows, and I was trying to build your modules in windows-amd64 because I read that  "Linux Containers on Windows should only be used when using Windows machine as dev box for a solution that will be run on Linux.". 
I will try to test it in raspberry pi.

> What is your concern with Redis?

Just curiosity about the advantages of using Redis Cache in this architecture.

closed 2019-02-28 by ronniesa