Azure/iotedge-lorawan-starterkit #176: Newbie questions

Gijsen, 2019-04-04:
Hi I'm new to the IoT world and Azure and would like to create a gateway on a Raspberry PI with a Lora hat.

I don't really understand what I should install on the Pi, should I need to create a gateway first and then install the edge or is everything needed in this repo?

Is there any configuration to make this work with a specific Lora hat? I'm doing a proof of concept so I'm making a simple single channel gateway with the Dragino Lora hat.

TechPreacher, 2019-04-04:
Hi @gijsen! Great that you want to use our solution.
The best way to get started is the quick start guide.
https://github.com/Azure/iotedge-lorawan-starterkit/blob/master/Docs/quickstart.md
Follow it step by step to:
- set up the Azure portion using the template (blue button). *Your reset pin is 3* if you use a Dragino LoRa hat.
- install IoT Edge on your PI and add the connection string to your IoT Hub created above
- enable SPI on your PI
- check if it gets the containers for edge and our solution and start connecting LoRa devices to it.

As described here: https://wiki.dragino.com/index.php?title=LoRa_Questions#Can_Dragino_devices_communicate_with_LoRa_Devices_from_other_vendor.3F the Dragino is a single channel device and not fully up to all that LoRa is capable of. You should use ABP LoRa devices, not OTAA and won't be able to do cloud-to-device messaging or receive confirmations for messages.
We have not tested the Dragino hat so you will have to try if it works.

Gijsen, 2019-04-04:
Hi @TechPreacher! Thanks for the great response.

I will definitely try this and will let it know when this works on the Dragino hat.

I know it will be single channel, it is not used in production but just as a POC. 

Very much thanks

ronniesa, 2019-04-04:
@Gijsen if you want a very cost effective true 8 channel I recommend the RAK2245 Pi HAT around 140 usd. Using a single channel as a gateway is not supported nor tested so it probably would not work well. As @TechPreacher mentioned only ABP unconfirmed may work but not even sure.

Gijsen, 2019-04-30:
> @Gijsen if you want a very cost effective true 8 channel I recommend the RAK2245 Pi HAT around 140 usd. Using a single channel as a gateway is not supported nor tested so it probably would not work well. As @TechPreacher mentioned only ABP unconfirmed may work but not even sure.

I tried with RAK2245 but I keep getting the error [main] failed to start the concentrator and I have no idee why

ronniesa, 2019-04-30:
did you enble spi on the Pi?

Gijsen, 2019-04-30:
yes I'm using a Raspberry Pi 3

ronniesa, 2019-04-30:
I mean the SPI interface on the Pi needs to be enabled. Is off by default

Gijsen, 2019-04-30:
Yes I enabled it with raspi-config

Gijsen, 2019-04-30:
Is there a way to get more info about why the concentrator failed to start?

ronniesa, 2019-04-30:
Next up did you configure the right reset pin?

Gijsen, 2019-04-30:
I used pin 17 for reset

Gijsen, 2019-04-30:
@ronniesa do you have any idea how I could debug the issue or what it could be?

Mandur, 2019-04-30:
It's hard to troubleshoot on github, would you be open to a remote session to narrow down on the issue?

Gijsen, 2019-04-30:
> It's hard to troubleshoot on github, would you be open to a remote session to narrow down on the issue?

Oke I ccontacted you

Mandur, 2019-06-03:
The problem was solved over voice conversation.

closed 2019-06-03 by Mandur