Azure/iotedge-lorawan-starterkit #179: DeviceEUI captialization bug?

gerfen, 2019-04-06:
I'm not sure if this is a bug or the expected behavior.  I just received some sensors form a new vendor.  When I scan their QRCode, the DeviceEUI is presented like:  ```39:34:35:37:7e:37:7e:15```.  Before registering the device with Azure IoT Hub I removed the colons so that the DeviceEUI looked like this:  ```393435377e377e15```.  However the gateway refused to join the device until I deleted the original registration and changed the DeviceEUI to look like this: ```393435377E377E15```.

## Expected Behavior

I expect that the gateway should be able to normalize the DeviceEUI so that the join works regardless of the capitalization, i.e. either  ```393435377e377e15``` or ```393435377E377E15``` should work when a join request is made.

## Current Behavior

The join request is refused.
## Steps to Reproduce

1. Take any device which has letters in it's DeviceEUI and make one or more of the letters lowercase.  Register the device with Azure IoT Hub with the lowercased DeviceEUI
2. Turn on the device and allow it to make a join request

## Context (Environment)

### Device (Host) Operating System

Ubuntu 16.04
### Architecture

arm32v7 and amd64

### Container Operating System

### LoRaWAN Module Version

#### Docker

Version 18.03.0-ce-win59 (16762)
## Logs

## Additional Information

Unfortunately I don't have any logs to share at the moment as I got the devices to join after correcting the DeviceEUI to have capital letters.  I can produce the logs if necessary.

fbeltrao, 2019-04-06:
Hi @gerfen,

Thanks for posting the issue.

This is an expected behaviour. Currently the DeviceEUI must be capitalised.
Main reason is the fact that IoT Hub is case sensitive regarding device identifier. For example: querying for deviceId == 'deviceA' will not find a device with identifier 'DeviceA'.

We could make it work with the value you provided, but we believe it is easier if you always capitalise them. This way, if a device is not able to join you can quickly check if it exists in IoT Hub. Checking if it exists with all upper/lower case combinations in a 16 character long string can be a nightmare if you have 1000+ devices.

Additionally, I would recommend you to take a look at the cli tool for managing LoRa devices we released. It's goal is to make this annoyances disappear when managing LoRa devices :)

Check out here:
https://github.com/Azure/iotedge-lorawan-starterkit/tree/master/Tools/Cli-LoRa-Device-Provisioning

gerfen, 2019-04-11:
Thanks for explanation.  I'm closing the issue.

closed 2019-04-11 by gerfen