Azure/iotedge-lorawan-starterkit #184: An AMQP error occurred (condition='amqp:internal-error') during join requests

gerfen, 2019-04-11:
I'm receiving "An AMQP error occurred (condition='amqp:internal-error')"  for some of my devices when they are trying to join my gateway.  I started to get this error after join 100 or so devices.  Is this some sort of limitation of the iotedge-lorawan-starterkit code or Azure IoT?  Or is is a bug? 
## Expected Behavior

## Current Behavior

## Steps to Reproduce

1.
2.
3.
4.

## Context (Environment)

### Device (Host) Operating System

Ubuntu 16.04
### Architecture

arm32
### Container Operating System

### LoRaWAN Module Version

#### Docker

## Logs

2019-04-11 20:06:23.100 Physical dataUp {"rxpk":[{"tmst":1941663867,"chan":8,"rfch":0,"freq":903.000000,"stat":1,"modu":"LORA","datr":"SF8BW500","codr":"4/5","lsnr":-3.5,"rssi":-110,"size":23,"data":"AAAAAAAQehMAXUIAABB6EwBDikN1caQ="}]}
2019-04-11 20:06:23.101 00137A100000425D: join request received
2019-04-11 20:06:23.101 00137A100000425D: querying the registry for OTAA device
2019-04-11 20:06:23.157 00137A100000425D: using edgeHub local queue
2019-04-11 20:06:23.158 00137A100000425D: getting device twin
2019-04-11 20:06:23.222 00137A100000425D: could not retrieve device twin with error: An AMQP error occurred (condition='amqp:internal-error').
2019-04-11 20:06:23.222 00137A100000425D: join refused: error initializing OTAA device, getting twin failed
2019-04-11 20:06:23.222 00137A100000425D: processing time: 00:00:00.1220033
2019-04-11 20:06:23.222 00137A100000425D: join refused: unknown device
2019-04-11 20:06:26.317 Statistic: {"stat":{"time":"2019-04-11 20:06:26 GMT","rxnb":9,"rxok":8,"rxfw":8,"ackr":100.0,"dwnb":0,"txnb":0}}

## Additional Information

ronniesa, 2019-04-12:
This is expected because by default IoT Edge has a maximum of 100 devices concurrently connected. You can increase this settings on the edgeHub module with the env variable:  "MaxConnectedClients". 

Depending on the type of the lora device you can optimize for latency and downstream msgs (keeping the connection open for a long time, our default is 1 day) or for sensors density where after every lora message we close the connection to iot hub (this is good for something like parking sensors that are mostly, if not only upstream, and are not high frequency in sending msgs). You control this per device in specifying "KeepAliveTimeout" desired twin: https://github.com/Azure/iotedge-lorawan-starterkit/blob/master/Docs/quickstart.md#optional-device-properties 

You could expect on an Intel gateway to have 500 or even up to 990 active connections at the same time. On a Pi we tested with 250 concurrent.

gerfen, 2019-04-12:
@ronniesa Thanks for the info.  I'm running the AAEON AIOT-IP68 gateway so I think I should easily be able to support 500 devices.  Just want to confirm that I set the "MaxConnectedClients" variable in deviceConfiguration.json?

````
"env": {
    "OptimizeForPerformance": {
          "value": "false"
    },
    "MaxConnectedClients": {
	  "value": "500"
    },
    "mqttSettings__enabled": {
           "value": "false"
    },
    "httpSettings__enabled": {
            "value": "false"
    },
    "TwinManagerVersion": {
            "value": "v2"
    }
},
````
I've deploy 110 temperature and humidity sensors which are reporting every 5 minutes.  If I understand correctly  I think I should stay with the default KeepAliveTimeout for the device twins for these devices?

ronniesa, 2019-04-12:
Correct. 

If the devices are upstream only you could further optimize by setting the device twin “Downlink” to false.

gerfen, 2019-04-12:
Would setting "Downlink" to false prevent sending C2D messages for setting things like reporting interval?

--Michael

fbeltrao, 2019-04-12:
Yes, setting `Downlink=false` will prevent the device from receiving cloud to device messages.

gerfen, 2019-04-12:
@fbeltrao Thanks.  I'll close this issue after I've confirmed that I can connect more than 100 devices sometime later this afternoon my time.

gerfen, 2019-04-15:
@ronniesa  and @fbeltrao  I've confirmed that I can register more than 100 devices with my gateway.  Thanks for the assistance!

Closing the issue.

closed 2019-04-15 by gerfen