Azure/iotedge-lorawan-starterkit #194: Question:  Response message from Class C direct message?

gerfen, 2019-06-18:
My class C devices have the ability to send a response message after receiving a direct message.  For example:

[image]

 Currently the returned payload from my successful calls are null.  

```
{"status":200,"payload":null}
```

Is this scenario supported in the currently release?

ronniesa, 2019-06-19:
The Class C device will either send an ACK or upstream message so you will see it as device message and not as return of the direct method call "{"status":200,"payload":null}". The reason is that the device can take the time that it needs to process and the ack back or send a upstream message with the payload. You can specify a messagId in the payload that you can use as correlation: https://github.com/Azure/iotedge-lorawan-starterkit/blob/master/Docs/Pictures/cloudtodevice-classc.png

gerfen, 2019-06-21:
@ronniesa - that makes sense.  I'm seeing intermittent upstream messages from the class C device as shown here[1].  I'm not sure if this is due to a further frequency issue related to US frequencies or if there is an issue with the device? I'm currently talking with the device manufacturer to understand the behavior.  I'd like to leave this issue open for a few more days until I hear back from the manufacturer.  Is that OK?

```
[1] Gateway logs
2019-06-20 17:12:43.601 Statistic: {"stat":{"time":"2019-06-20 17:12:43 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}
2019-06-20 17:12:55.375 1152612292133022: received cloud to device message from direct method: {"devEUI":"1152612292133022","fport":"55","confirmed":true,"payload":"digitalOutput!1","MessageId":"53be5fb4c6bc48ef9031032a26db86d1"}  Command to set DO to high
2019-06-20 17:12:55.375 1152612292133022: [class-c] down frame counter: 19
2019-06-20 17:12:55.376 1152612292133022: ConfirmedDataDown {"txpk":{"imme":true,"data":"oJ1c+gIAEwA3T5wTGAPYkvUhXt6QoT1wciBy3Q==","tmst":0,"size":28,"freq":923.3,"rfch":0,"modu":"LORA","datr":"SF10BW500","codr":"4/5","powe":14,"ipol":true}}
2019-06-20 17:12:55.376 UDP: sending message with ID A3CA, to 172.18.0.1:58495
2019-06-20 17:12:55.377 UDP: message sent with ID A3CA
2019-06-20 17:12:55.378 UDP: packet with id A3CA successfully transmitted by the aggregator
2019-06-20 17:13:13.596 Statistic: {"stat":{"time":"2019-06-20 17:13:13 GMT","rxnb":1,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":1,"txnb":1}}
2019-06-20 17:13:43.600 Statistic: {"stat":{"time":"2019-06-20 17:13:43 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}
2019-06-20 17:14:13.600 Statistic: {"stat":{"time":"2019-06-20 17:14:13 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}
2019-06-20 17:14:43.595 Statistic: {"stat":{"time":"2019-06-20 17:14:43 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}
2019-06-20 17:15:13.604 Statistic: {"stat":{"time":"2019-06-20 17:15:13 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}
2019-06-20 17:15:43.596 Statistic: {"stat":{"time":"2019-06-20 17:15:43 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}
2019-06-20 17:16:13.599 Statistic: {"stat":{"time":"2019-06-20 17:16:13 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}
2019-06-20 17:16:22.435 1152612292133022: received cloud to device message from direct method: {"devEUI":"1152612292133022","fport":"55","confirmed":true,"payload":"digitalOutput!0","MessageId":"f866120bd2924af885e4b1eb146b8b92"} 
2019-06-20 17:16:22.437 1152612292133022: [class-c] down frame counter: 20
2019-06-20 17:16:22.438 1152612292133022: ConfirmedDataDown {"txpk":{"imme":true,"data":"oJ1c+gIAFAA3Rw4vmScDw/QwpW59jF4031q27w==","tmst":0,"size":28,"freq":923.3,"rfch":0,"modu":"LORA","datr":"SF10BW500","codr":"4/5","powe":14,"ipol":true}}
2019-06-20 17:16:22.438 UDP: sending message with ID 5FF0, to 172.18.0.1:58495
2019-06-20 17:16:22.438 UDP: message sent with ID 5FF0
2019-06-20 17:16:22.441 UDP: packet with id 5FF0 successfully transmitted by the aggregator
2019-06-20 17:16:43.600 Statistic: {"stat":{"time":"2019-06-20 17:16:43 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":1,"txnb":1}}
2019-06-20 17:17:13.602 Statistic: {"stat":{"time":"2019-06-20 17:17:13 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}
2019-06-20 17:17:43.375 Physical dataUp {"rxpk":[{"tmst":304396796,"chan":2,"rfch":0,"freq":902.700000,"stat":1,"modu":"LORA","datr":"SF8BW125","codr":"4/5","lsnr":8.2,"rssi":-49,"size":96,"data":"QJ1c+gIAigA3efE0qYSlbS1ryVENDIiFTI3ff0WFpD8WWjYsYaAVZ7R9cPh6jIcAphOVmmxXalHCrV38+x60EOvW2QC0dVLlaH9Ck2TlAE8PLLqU5EbEKVHlXFmdWlT/"}]}
2019-06-20 17:17:43.376 1152612292133022: device in cache
2019-06-20 17:17:43.377 1152612292133022: converted 16bit FCnt 138 to 32bit FCnt 138
2019-06-20 17:17:43.377 1152612292133022: valid frame counter, msg: 138 server: 118
2019-06-20 17:17:43.377 1152612292133022: no decoder set in device twin. port: 55
2019-06-20 17:17:43.378 1152612292133022: sending message {"time":null,"tmms":0,"tmst":304396796,"freq":902.7,"chan":2,"rfch":0,"stat":1,"modu":"LORA","datr":"SF8BW125","codr":"4/5","rssi":-49,"lsnr":8.2,"size":96,"data":{"value":"fvRTALW/C10AAAAeAAAAAwEBAQAeAAAAAgIAAgAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAH4="},"port":55,"fcnt":138,"rawdata":"fvRTALW/C10AAAAeAAAAAwEBAQAeAAAAAgIAAgAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAH4=","eui":"1152612292133022","gatewayid":"evolution-gateway","edgets":1561051063374} to hub
2019-06-20 17:17:43.407 1152612292133022: message '{"value":"fvRTALW/C10AAAAeAAAAAwEBAQAeAAAAAgIAAgAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAH4="}' sent to hub 
2019-06-20 17:17:43.408 1152612292133022: checking cloud to device message for 00:00:00.5664509
2019-06-20 17:17:43.603 Statistic: {"stat":{"time":"2019-06-20 17:17:43 GMT","rxnb":2,"rxok":1,"rxfw":1,"ackr":100.0,"dwnb":0,"txnb":0}}
2019-06-20 17:17:43.978 1152612292133022: done checking cloud to device message, found no message
2019-06-20 17:17:43.979 1152612292133022: updating twin
2019-06-20 17:17:43.992 1152612292133022: twin updated
2019-06-20 17:17:43.992 1152612292133022: processing time: 00:00:00.6180634
2019-06-20 17:18:13.601 Statistic: {"stat":{"time":"2019-06-20 17:18:13 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}
[... cut ...]
2019-06-20 17:30:46.125 1152612292133022: message '{"value":"fvRTAMPCC10AAQAeAAAAAwEBAQAeAAAAAgIAAgAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAAAeAAAAAAAAAH4="}' sent to hub 
2019-06-20 17:30:46.125 1152612292133022: checking cloud to device message for 00:00:00.5674510
2019-06-20 17:30:46.694 1152612292133022: done checking cloud to device message, found no message
2019-06-20 17:30:46.694 1152612292133022: updating twin
2019-06-20 17:30:46.706 1152612292133022: twin updated
2019-06-20 17:30:46.706 1152612292133022: processing time: 00:00:00.6131619
```

ronniesa, 2019-06-21:
Yes you can keep it open. Upstream should not be a problem for our stack between frequencies (e.g. US vs EU). If the PktFwr is correctly configured all the freq stuff is done by the device and the pktfwrd….. in our stack we just get what the pktfwrd is sending us.... downstream is another story because we need to prepare the msg with the correct DR, freq, etc...

gerfen, 2019-06-27:
@ronniesa I'm still having issues getting consistent upstream messages from the Class C devices in response to a downstream message.  I started monitoring the packet forwarder logs and I'm seeing this...

1.  When a message is successfully sent upstream the logs look like this:
```
##### 2019-06-27 17:36:56 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 1
# CRC_OK: 100.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 1 (24 bytes)
# PUSH_DATA datagrams sent: 2 (319 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 1 (185 bytes)
# RF packets sent to concentrator: 1 (28 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:26, rej:0)
# TX rejected (collision beacon): 0.00% (req:26, rej:0)
# TX rejected (too late): 0.00% (req:26, rej:0)
# TX rejected (too early): 0.00% (req:26, rej:0)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 284111900
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# GPS sync is disabled
##### END #####
```

2.  When I do not get an upstream message I see one of the following:
```
### [UPSTREAM] ###
# RF packets received by concentrator: 2
# CRC_OK: 0.00%, CRC_FAIL: 100.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 1 (113 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 1 (185 bytes)
# RF packets sent to concentrator: 1 (28 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:20, rej:0)
# TX rejected (collision beacon): 0.00% (req:20, rej:0)
# TX rejected (too late): 0.00% (req:20, rej:0)
# TX rejected (too early): 0.00% (req:20, rej:0)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 4099071842
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# GPS sync is disabled
##### END #####
```
or
```
##### 2019-06-27 17:31:26 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 2
# CRC_OK: 50.00%, CRC_FAIL: 50.00%, NO_CRC: 0.00%
# RF packets forwarded: 1 (28 bytes)
# PUSH_DATA datagrams sent: 2 (328 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 1 (185 bytes)
# RF packets sent to concentrator: 1 (28 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:21, rej:0)
# TX rejected (collision beacon): 0.00% (req:21, rej:0)
# TX rejected (too late): 0.00% (req:21, rej:0)
# TX rejected (too early): 0.00% (req:21, rej:0)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 4279074379
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# GPS sync is disabled
##### END #####
```
or
```
##### 2019-06-27 17:38:26 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 0
# CRC_OK: 0.00%, CRC_FAIL: 0.00%, NO_CRC: 0.00%
# RF packets forwarded: 0 (0 bytes)
# PUSH_DATA datagrams sent: 1 (113 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 1 (185 bytes)
# RF packets sent to concentrator: 1 (28 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:27, rej:0)
# TX rejected (collision beacon): 0.00% (req:27, rej:0)
# TX rejected (too late): 0.00% (req:27, rej:0)
# TX rejected (too early): 0.00% (req:27, rej:0)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 404113897
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# GPS sync is disabled
##### END #####
```

Does this point to a configuration issue with the packet forwarder or perhaps a frequency mismatch between the class devices and the gateway?  I'm not sure how to interpret the logs.

ronniesa, 2019-06-28:
It seems a problem between the device and the gateway. You can see that there are CRC_FAIL and you can see here some possible causes: https://www.thethingsnetwork.org/docs/gateways/troubleshooting/semtech-udp.html#uplink-issues It could be a mismatch of settings between the concentrator and the device or you could try to move the gateway and the device further apart or in another location to see if it helps. 

If you did not change it here is the us configuration that we use for the pktwrd for US freq: https://github.com/Azure/iotedge-lorawan-starterkit/blob/master/LoRaEngine/modules/LoRaWanPktFwdModule/global_conf.us.json you can double check with your device frequencies/channels.

gerfen, 2019-06-28:
I think I'm finally getting somewhere...  I can configure the frequency channel indexes for the class C device:
[image]

By default the device is configured to support channels 0 - 71.  I've had much better luck if I configure the device to only use channels 0 - 7.  Referencing the image above could you give me guidance on which channels I should use?

gerfen, 2019-07-24:
I'm closing this as setting the device to only use 8 channels has solved my issue with these particular Class C devices.  

Thanks again for the assistance!

closed 2019-07-24 by gerfen