Azure/iotedge-lorawan-starterkit #197: Payload in HEX

lucarv, 2019-06-24:
**Is your feature request related to a problem? Please describe.**
We are working with a customer that is using two different lora gateways, one of them being this SW. The data comes encoded as HEX from the otherGW and base64 from this SW. They decode the data using a stream analytics, which needs a hex input. They have attempted to write a base64 to hex converter but it has proved hard in plain js. We were wondering if this SW must base64 encode or if it could send payload in HEX instead.

**Describe the solution you'd like**
Payload upload to iot hub as HEX

**Describe alternatives you've considered**
Tried to write a simple base64 to hex using plain javascrict as UDF in stream analytics. It can be done but maybe this is easier to do at the gateway

**Additional context**
Add any other context or screenshots about the feature request here.

ronniesa, 2019-06-25:
The easiest way would be to create a custom decoder for now. But we could have an additional default decoder that you could turn on device by device. We will discuss this feature and let you know.

ronniesa, 2019-07-24:
We have implemented the HEX decoder in our DEV branch: https://github.com/Azure/iotedge-lorawan-starterkit/pull/203 at it will be part of the next realse. Thanks for the suggestion!

ronniesa, 2019-09-30:
It is now in release 1.0.2

closed 2019-09-30 by ronniesa