Azure/iotedge-lorawan-starterkit #198: Intercepting NetworkSrvModule upstream messages

iSnackTwoPointOh, 2019-06-27:
Hi, I'm reasonably new to both LoRa and Azure and I'm currently working on a project that is looking to apply some ML to a LoRa network. In the root readme file there is a diagram that shows custom IoT Edge modules tapping into upstream messages from NetworkSrvModule. Ideally, I want to deploy a docker image on the edge device to do the ML processing. I haven't had much luck finding details on how to do this and any help would be much appreciated.

Originally we thought the module was using the routing between docker modules (ie. FROM /* INTO $upstream). However, this appears to be for logging only and the device payloads are sent using SendEventAsync() of DeviceClient. 

I guess my question is two parts:
What was the intended implementation of the architecture mentioned in the readme?
and,
Is it possible to intercept the upstream payload using docker endpoints before using SendEventAsync()?

Thanks in advance :)

ronniesa, 2019-06-27:
Hi, yes the concept it is like in the architecture diagram and follows the general composition design of IoT Edge modules. 
In the LoRa implementation we have two source of the messages: the first are the logs from the networkservermodule (you can enable or disable with the Environment Variable LOG_TO_HUB to true or false on the module settings from the portal or in the deployment template), the second are the devices LoRa messages (sent by DeviceClient.SendEventAsync).

Now this route (FROM /* INTO $upstream) means send all messages to IoT Hub so server side you can get all the logs and the device messages. 
If you want to have a module at the Edge to perform actions you probably want to just send the device messages to the module and not the logs and this is done by adding an additional route like this:
{
  "routes": {
    "alltoiothub": " FROM /* INTO $upstream",
    "devicetomodule": "FROM /messages/\* WHERE NOT IS_DEFINED($connectionModuleId) INTO BrokeredEndpoint(\"/modules/yourmodulename /inputs/yourinputname\")"
  }
}

Here you can learn more on how the composition works: https://docs.microsoft.com/en-us/azure/iot-edge/module-composition#declare-routes 

The cool part is that you can also add filters like only certain device goes to certain modules etc. Just pay attention on the spelling of the module name and input name for the module.

Let us know if you need more info

iSnackTwoPointOh, 2019-07-01:
@ronniesa Thanks so much! I've tested your suggestion and it's working :) That really cleared up the confusion I had about the routes. I think what I was doing before was looping all outputs back to the inputs and never getting anything upstream.

Mandur, 2019-07-04:
I will close the issue for now, in case you have some follow up questions please feel free to reopen.

closed 2019-07-04 by Mandur