Azure/iotedge-lorawan-starterkit #209: Class A - out of time for downstream message

gerfen, 2019-07-24:
I am successfully sending C2D messages to my Class A devices but on occasion I get the error shown in my gateway logs below.  I haven't been able to see any reproducible pattern but it seems to happen if I have not been sending messages for a long period of time.  The message always seems to succeed the next time the device sends an upstream message.  It concerns me because I will be controlling irrigation valves and then will typically be configured to send upstream messages on an infrequent basis.  This could be catastrophic on a hot day if the period between upstream messages is long. 

I'm not sure if this is an issue with the device, the gateway or something I will just need to live with?

### Device (Host) Operating System

Ubuntu 16.04
### Architecture

AMD64
### Container Operating System

### LoRaWAN Module Version
PKT_FWD_VERSION=1.0.7
NET_SRV_VERSION=1.0.7
#### Docker

 Version:           3.0.3
 API version:       1.40
 Go version:        go1.11.4
 Git commit:        48bd4c6d
 Built:             Wed Jan 23 16:17:56 2019
 OS/Arch:           linux/amd64
 Experimental:      false

## Logs

```
2019-07-24 04:04:37.262 0004A30B00240C03: done checking cloud to device message, found message id: c3a0c03d-a205-4b20-9b47-feeef02333f0
2019-07-24 04:04:37.262 0004A30B00240C03: syncing FCntDown for multigateway
2019-07-24 04:04:39.425 0004A30B00240C03: down frame counter: 43
2019-07-24 04:04:39.425 0004A30B00240C03: out of time for downstream message, will abandon cloud to device message id: c3a0c03d-a205-4b20-9b47-feeef02333f0
2019-07-24 04:04:39.426 0004A30B00240C03: abandoning cloud to device message, id: c3a0c03d-a205-4b20-9b47-feeef02333f0
2019-07-24 04:04:39.427 0004A30B00240C03: updating twin
2019-07-24 04:04:39.437 0004A30B00240C03: twin updated
2019-07-24 04:04:39.437 0004A30B00240C03: processing time: 00:00:02.2117083
2019-07-24 04:04:39.505 0004A30B00240C03: done abandoning cloud to device message, id: c3a0c03d-a205-4b20-9b47-feeef02333f0

```

ronniesa, 2019-07-24:
This is normal and it can happen the reason is that we do not receive the downstream message in time from iot hub so we need to abandon, at the next upstream message we try to deliver the message again. 

For a customer that needs to send down the current time for time sync we implemented a feature that the decoder can send the downstream message as an answer so basically you do all the logic iside the decoder so is faster and you don't relay on queues or server or network. 
For an additional customer we implemented that you can even send it to a Class C device from within the decoder. This was implemented for a usecase where a pump (ClassC) needs to be turn on based on a ClassA sensor value.

So the following patterns can be implemented:
- deviceA upstream messsage is decoded and a downstream message is sent to the same deviceA from within the decoder
- deviceA upstream messsage is decoded and a downstream message is sent to a differenc deviceC from within the decoder

We don't have yet a doc but the implementation is all there. The decoder DecodePayloadResult https://github.com/Azure/iotedge-lorawan-starterkit/blob/master/Samples/DecoderSample/Classes/DecodePayloadResult.cs has a new optional property LoRaCloudToDeviceMessage where you can assigne a LoRaCloudToDeviceMessage to it https://github.com/Azure/iotedge-lorawan-starterkit/blob/master/Samples/DecoderSample/Classes/LoRaCloudToDeviceMessage.cs if you omit the LoRaCloudToDeviceMessage.DevEUI it will send back to the same ClassA device if you specify a DevEUI it will try to send it to that device as ClassC

**Pay super attention at how long it takes for the operation to prepare the downstream message in case you call a server or do something more complex that will not be local only that is fast enough to stay below 1 second.**  It dosen't have a queue so if it cannot deliver it in the current window then the message is lost.

Let us know if you need help in trying this feature

gerfen, 2019-07-24:
@ronniesa Thanks for the info.  I will need to think this through.  I'd like to keep this issue open for a bit while I ponder the possibilities.

ronniesa, 2020-05-14:
@gerfen is this issue still active or can we close it?

gerfen, 2020-05-14:
I'm still working on how best to deal with the issue for the devices I'm working with.  I'll close for now and will reopen or add a new issue when I can get back to experimenting with the devices.

closed 2020-05-14 by gerfen