Azure/iotedge-lorawan-starterkit #212: Question: cloud to device message too large

gerfen, 2019-08-04:
I have a class C device which on occasion, gets into a state where I can no longer send it a direct message.  I get a message in my logs which state "cloud to device message too large, rejecting" as shown in the logs[1] below.  

The only way I can get the device to receive messages again is to reboot my gateway or clear it's cache and have the device rejoin.  Neither of which is very practical.  

I'm still running on the 1.0.0 release of the starter kit  but with the IoT Edge 1.0.7 release which is required to send direct messages.  Is this a known issue?  Please let me know if I can provide more information.

[1] Logs.

```
2019-08-04 02:11:40.661 1114611693080229: device in cache
2019-08-04 02:11:40.662 1114611693080229: converted 16bit FCnt 2916 to 32bit FCnt 2916
2019-08-04 02:11:40.662 1114611693080229: valid frame counter, msg: 2916 server: 2915
2019-08-04 02:11:40.663 1114611693080229: no decoder set in device twin. port: 55
2019-08-04 02:11:40.663 1114611693080229: sending message {"time":null,"tmms":0,"tmst":1668170140,"freq":903.7,"chan":7,"rfch":1,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","rssi":-50,"lsnr":11.2,"size":25,"data":{"value":"fvQMANk+Rl03AAB+"},"port":55,"fcnt":2916,"rawdata":"fvQMANk+Rl03AAB+","eui":"1114611693080229","gatewayid":"evolution-gateway","edgets":1564884700659} to hub
2019-08-04 02:11:40.689 1114611693080229: message '{"value":"fvQMANk+Rl03AAB+"}' sent to hub
2019-08-04 02:11:40.689 1114611693080229: checking cloud to device message for 00:00:00.5707346
2019-08-04 02:11:41.260 1114611693080229: done checking cloud to device message, found no message
2019-08-04 02:11:41.260 1114611693080229: processing time: 00:00:00.6007272
2019-08-04 02:11:43.974 1114611693080229: received cloud to device message from direct method: {"devEUI":"1114611693080229","fport":"55","confirmed":false,"payload":"CurtainLeftUp","MessageId":"b1f726eb6f0349308d077a8d9621bd6c"}
2019-08-04 02:11:43.975 1114611693080229: [class-c] down frame counter: 16
2019-08-04 02:11:43.975 1114611693080229: [class-c] cloud to device message too large, rejecting. Id: b1f726eb6f0349308d077a8d9621bd6c
```

Mandur, 2019-08-06:
Hello Gerfen,

Thanks for your question, I would need further information to see if it is expected or it is a bug.
* Do you know if you have ADR on on those devices?
* The message you are trying to send is CurtainLeftUp correct?

Best regards,

Mikhail

gerfen, 2019-08-12:
@Mandur Sorry for going dark.  I recently had some more time to try to get more info for this issue.  T\Yes, the message I was sending was CurtainLeftUp.  The devices were set for ADR.  I just turned that off and will let my system run overnight to see if the issue clears.

I'm seeing some other issues related to class C devices which I am collecting log files and trying to understand how to gather steps to reproduce.  These issues may also be related to ADR.  I will report back over the next couple of days.  Thanks for your patience.

gerfen, 2019-09-04:
I'm going to close this issue to open another related to my class C devices not being able to receive C2D messages after a gateway reboot.

closed 2019-09-04 by gerfen