Azure/iotedge-lorawan-starterkit #213: join refused: DevNonce already used by this device

gerfen, 2019-08-16:
I recently made a change to my device twins to specifically set the GatewayID.  After clearing the cache I started getting this message when many but not all of my devices are trying to connect to the gateway:

```
2019-08-16 21:15:18.171 1114611693080229: join request received
2019-08-16 21:15:18.172 1114611693080229: querying the registry for OTAA device
2019-08-16 21:15:18.282 1114611693080229: join refused: DevNonce already used by this device
2019-08-16 21:15:18.282 1114611693080229: processing time: 00:00:00.1123217
2019-08-16 21:15:18.282 1114611693080229: join refused: unknown device
2019-08-16 21:15:19.813 Statistic: {"stat":{"time":"2019-08-16 21:15:19 GMT","rxnb":11,"rxok":11,"rxfw":11,"ackr":100.0,"dwnb":0,"txnb":0}}
2019-08-16 21:15:22.965 Physical dataUp {"rxpk":[{"tmst":3635755724,"chan":1,"rfch":0,"freq":902.500000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":10.5,"rssi":-29,"size":23,"data":"AExrbmlMQHJVcVYGkxZhFBFed8V73ss="}]}
2019-08-16 21:15:22.966 1114611693065671: join request received
2019-08-16 21:15:22.966 1114611693065671: querying the registry for OTAA device
2019-08-16 21:15:23.101 1114611693065671: join refused: DevNonce already used by this device
2019-08-16 21:15:23.101 1114611693065671: processing time: 00:00:00.1364838
2019-08-16 21:15:23.101 1114611693065671: join refused: unknown device
2019-08-16 21:15:24.288 Physical dataUp {"rxpk":[{"tmst":3637081948,"chan":7,"rfch":1,"freq":903.700000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":13.8,"rssi":-77,"size":24,"data":"QKdIvQKAXo8G+WdQZlSpmP2wRjYnQ3LI"}]}
```

I also tried restarting my gateway but nothing seems to help.  What do I need to do to resolve this issue?

Thanks.

ronniesa, 2019-08-18:
Are you using 1.0 or 1.01? 
Can you try first to restart Redis from the Azure portal and right after the gateway and then turn on the device. If this does not work can you try deleting the device from IoT Hub, restart Redis and the Gateway. 

Please report back which action is correcting the problem it could be a bug or the device reusing the DevNonce when it should not.

The DevNonce is a random key that the device generates for the join and it must be always new. So in the code we compare it to the last successful one. It is stored as reported property in the twins.

ronniesa, 2019-09-30:
Version 1.0.2 should solve this issue about the devnonce too. After you updgrade please restart redis.

gerfen, 2020-03-25:
@ronniesa  Thanks for the support on this issue.  I meant to close it months ago.  Closing now.

closed 2020-03-25 by gerfen