Azure/iotedge-lorawan-starterkit #219: Deployment from Quick Start template appears to fail

gerfen, 2019-11-25:
Using Quick Start template taking a long time to complete.  

## Expected Behavior
All of the requisite Azure resources are successfully created.

## Current Behavior
After almost an hour, the deployment  appears to fail -- I don't see that an Azure function was created.  The Redis cache and the IoTHub were both created but no gateway has been added to the IoT hub.

## Steps to Reproduce

1. Click on the "Quick Start' button
2. Wait (almost an hour in my case)
3. At some point during the deployment, see a deployment failure (logs below)
4. Deployment seems to recover but then nothing else happens
5.  Wait some more
6.  The template appears to get stuck at "Get all Iot Hub keys" which appears to get called
 repeatedly as shown in the uploaded image.
[image]

7.  At this point I'm not sure what to do as it doesn't appear that I can stop the template.  

## Context (Environment)

### Device (Host) Operating System

### Architecture

### Container Operating System

### LoRaWAN Module Version

#### Docker

## Logs

```
{
    "authorization": {
        "action": "Microsoft.Resources/deployments/write",
        "scope": "/subscriptions/42d031ca-ae31-4c28-baff-23438604f5ae/resourceGroups/Production-Resource-Group/providers/Microsoft.Resources/deployments/Microsoft.Template"
    },
    "caller": "michael.gerfen@nurserysentry.com",
    "channels": "Operation",
    "claims": {
        "aud": "https://management.core.windows.net/",
        "iss": "https://sts.windows.net/f0dfc3bb-3339-4276-a0cc-e5d1835a5aaf/",
        "iat": "1574711601",
        "nbf": "1574711601",
        "exp": "1574715501",
        "http://schemas.microsoft.com/claims/authnclassreference": "1",
        "aio": "AUQAu/8NAAAAJy9NlFk0FDB/NsU4rsVfjWEghpYi60xQBmml042v3sYyf1OibdiPkaCxLFBtllcagnaLdO4NHDZrdsYuQbZenA==",
        "altsecid": "1:live.com:000340011F0A7D8E",
        "http://schemas.microsoft.com/claims/authnmethodsreferences": "pwd",
        "appid": "c44b4083-3bb0-49c1-b47d-974e53cbdf3c",
        "appidacr": "2",
        "http://schemas.xmlsoap.org/ws/2005/05/identity/claims/emailaddress": "michael.gerfen@nurserysentry.com",
        "http://schemas.xmlsoap.org/ws/2005/05/identity/claims/surname": "Gerfen",
        "http://schemas.xmlsoap.org/ws/2005/05/identity/claims/givenname": "Michael",
        "groups": "81b08094-7fab-449b-97e8-53abedf77ec0",
        "http://schemas.microsoft.com/identity/claims/identityprovider": "live.com",
        "ipaddr": "72.0.168.220",
        "name": "Michael Gerfen",
        "http://schemas.microsoft.com/identity/claims/objectidentifier": "0b448951-84ef-43ce-a7fe-7757b4ba4e69",
        "puid": "100320007D9BA485",
        "http://schemas.microsoft.com/identity/claims/scope": "user_impersonation",
        "http://schemas.xmlsoap.org/ws/2005/05/identity/claims/nameidentifier": "uRFezc_uiK5Z9VwT-Fvxtm6CV8eDXJO5sfDlmtBvUXs",
        "http://schemas.microsoft.com/identity/claims/tenantid": "f0dfc3bb-3339-4276-a0cc-e5d1835a5aaf",
        "http://schemas.xmlsoap.org/ws/2005/05/identity/claims/name": "live.com#michael.gerfen@nurserysentry.com",
        "uti": "Cyfb8ywJQ0KO4bpDpWoaAA",
        "ver": "1.0",
        "wids": "62e90394-69f5-4237-9190-012177145e10"
    },
    "correlationId": "d5d20b2a-1cc4-44e5-8459-65afd0e770fa",
    "description": "",
    "eventDataId": "c381cd48-ea6e-4331-b478-b4127bfa773f",
    "eventName": {
        "value": "EndRequest",
        "localizedValue": "End request"
    },
    "category": {
        "value": "Administrative",
        "localizedValue": "Administrative"
    },
    "eventTimestamp": "2019-11-25T20:26:05.6161459Z",
    "id": "/subscriptions/42d031ca-ae31-4c28-baff-23438604f5ae/resourceGroups/Production-Resource-Group/providers/Microsoft.Resources/deployments/Microsoft.Template/events/c381cd48-ea6e-4331-b478-b4127bfa773f/ticks/637103103656161459",
    "level": "Error",
    "operationId": "69d40079-f8d4-4353-b6f4-09f0d61b7b26",
    "operationName": {
        "value": "Microsoft.Resources/deployments/write",
        "localizedValue": "Create Deployment"
    },
    "resourceGroupName": "Production-Resource-Group",
    "resourceProviderName": {
        "value": "Microsoft.Resources",
        "localizedValue": "Microsoft Resources"
    },
    "resourceType": {
        "value": "Microsoft.Resources/deployments",
        "localizedValue": "Microsoft.Resources/deployments"
    },
    "resourceId": "/subscriptions/42d031ca-ae31-4c28-baff-23438604f5ae/resourceGroups/Production-Resource-Group/providers/Microsoft.Resources/deployments/Microsoft.Template",
    "status": {
        "value": "Failed",
        "localizedValue": "Failed"
    },
    "subStatus": {
        "value": "",
        "localizedValue": ""
    },
    "submissionTimestamp": "2019-11-25T20:27:32.1971747Z",
    "subscriptionId": "42d031ca-ae31-4c28-baff-23438604f5ae",
    "relatedEvents": []
}
```

## Additional Information

ronniesa, 2019-11-26:
Testing it out right now let you know

ronniesa, 2019-11-26:
In my case it worked and also quite fast ca. 12 minutes 
[image]

Can you please retry

gerfen, 2019-11-26:
Thanks Ronnie.  I’ll delete the resources that where created during the failed attempt and will try again.  Appreciate you checking the issue for me.

[image]<https://user-images.githubusercontent.com/24290518/69647368-a5f30880-1069-11ea-8523-bb720d1238e8.png>

gerfen, 2019-11-26:
After deleting the resources and running the template again, all of the resources were created.  FWIW it took about 12 minutes for this time as well.

closed 2019-11-26 by gerfen