Azure/iotedge-lorawan-starterkit #221: Device is not our device

gerfen, 2019-11-30:
I just did a clean install of version 1.0.2 to a brand new Azure account using the Quick Start template.  After updating the IoT Edge connection string on my gateway device, all of the various modules started as expected after a reboot however none of the child devices associated to the gateway are able to connect.  I get an message "device is not our device" for each device which is trying to connect.  

I'm using version 1.0.7 of azureiotedge-agent anf azureiotedge-hub.  I tried restarting the redis cache but this had no effect.  This is a production deployment so it important that I am able to resolve this quickly. 

any ideas as to what I can do to resolve this issue?

I will try to collect better logs tomorrow and will attach them to this issue ASAP.

### Device (Host) Operating System

Ubuntu 16.04
### Architecture

amd64

Mandur, 2019-12-02:
We had a troubleshooting call this morning and we have a theory on what is going wrong. @gerfen  will check tomorrow if the proposed solution work as expected.

gerfen, 2019-12-03:
I have an update from the proposed solution.  When I first imported the my devices into my new Azure tenant, I did not import the the device twin reported properties.  @Mandur suggest that I delete the devices from IoT Hub and import them again using some code that he kindly provided [ here ](https://github.com/Mandur/iothubdevicesynctool) .  I was not able to use this code as I took the opportunity to clean up the name of the LoRa gateways I have deployed.  Instead I used this code[1] which I am providing here in case someone else finds it useful.  One issue that I encountered while importing the reported properties was a rather unhelpful 400 error when trying to import an readonly property, i.e. $metadata and $version.  To get around that issue I was forced to convert the TwinCollection to Dictionary<string, object>, remove the readonly properties and then convert back to TwinCollection before calling `await deviceClient.UpdateReportedPropertiesAsync(twinCollection);`

After getting the devices imported with the device twins reported properties the devices were able to connect to the gateway, however only a handful of devices were able to send data.  Unfortunately I am not able to get remote access to the gateway without assistance from a local resource so I was not able to reboot the gateway.

This morning I was able to get some remote assistance.  Here are the logs prior to rebooting the gateway:

[2019-12-03-before-reboot.logs.txt](https://github.com/Azure/iotedge-lorawan-starterkit/files/3917975/2019-12-03-before-reboot.logs.txt)

And here are the logs after rebooting the gateway:

[2019-12-03-after-reboot.logs.txt](https://github.com/Azure/iotedge-lorawan-starterkit/files/3917965/2019-12-03-after-reboot.logs.txt)

Prior to rebooting the gateway, I had my helper cycle the power the power on one of the LoRa devices.  This device connected and started to send data.  

Looking at the logs, it appears that the MIC check is failing which is why the devices cannot send data to the IoT Hub.  It would be helpful to not have to cycle the power on the deployed devices since there are close to 100 devices deployed over a wide area.  

I'd be grateful for any ideas on how to programmatically resolve this issue.

[1] Import code
```
private readonly RegistryManager registryManagerSource_ = RegistryManager.CreateFromConnectionString("<elided>");
        private readonly RegistryManager registryManagerDestination_ = RegistryManager.CreateFromConnectionString("<elided>");

   private async Task TransferTwins(string gatewayId, string tenantId)
        {
            // Get the Gateway device
            var parentDevice = await registryManagerDestination_.GetDeviceAsync(gatewayId);

            var dbContext = ServiceProvider.GetService<GreenhouseDbContext>();
            var devices = dbContext.Devices
                .Where(device => device.TenantId == Guid.Parse(tenantId)).ToList();
            Output.WriteLine($"Found {devices.Count} devices to transfer");
            foreach (var device in devices)
            {
                try
                {
                   
                    var deviceId = device.DeviceId;
                    var originalTwin = await registryManagerSource_.GetTwinAsync(deviceId);
                    var originalDevice = await registryManagerSource_.GetDeviceAsync(deviceId);

                    Output.WriteLine($"Transferring '{deviceId}'");
                    var newDevice = new Device(deviceId)
                    {
                        Authentication = originalDevice.Authentication,
                        Scope = parentDevice.Scope,
                    };

                    newDevice = await registryManagerDestination_.AddDeviceAsync(newDevice);

                    // Update the desired properties
                    var deviceTwin = await registryManagerDestination_.GetTwinAsync(newDevice.Id);
                    deviceTwin.Properties.Desired = originalTwin.Properties.Desired;
                    deviceTwin.Properties.Desired["GatewayID"] = parentDevice.Id;
                    await registryManagerDestination_.UpdateTwinAsync(deviceId, deviceTwin, deviceTwin.ETag);

                    // Update reported properties
                    var primaryKey = newDevice.Authentication.SymmetricKey.PrimaryKey;
                    var deviceClient = DeviceClient.CreateFromConnectionString($"HostName=<iot hub name>.azure-devices.net;DeviceId={newDevice.Id};SharedAccessKey={primaryKey}");
                    var newTwin = await deviceClient.GetTwinAsync();

                    var dictionary = JsonConvert.DeserializeObject<Dictionary<string, object>>(originalTwin.Properties.Reported.ToJson());
                    dictionary.Remove("$metadata");
                    dictionary.Remove("$version");
                    var twinCollection = new TwinCollection(JsonConvert.SerializeObject(dictionary))
                    {
                        ["GatewayID"] = parentDevice.Id
                    };
                    await deviceClient.UpdateReportedPropertiesAsync(twinCollection);

                }
                catch (Exception ex)
                {
                    Output.WriteLine(ex.Message);
                }
            }
        }
```

gerfen, 2019-12-04:
I have an update.  The devices which are connected to the gateway, but which were not reporting data due to the MIC check failure are able to send data after having their power recycled.  I should note that this was not possible until the Redis cache and the cache on the gateway were cleared to sync the two caches.  I still find it interesting that the devices require a power cycle to force a join for the devices to be able to pass the MIC check.

Mandur, 2019-12-17:
Based on our experiments, it looks like restarting the lora edge gateway fixed the issue. We think the issue was due to a migration between multiple LoRa network were the same deviceId were provisioned. Thank you for reporting this @gerfen, please let us if any more problems rise again in this matter and feel free to reopen in case there are any more things you would want to mention

closed 2019-12-17 by Mandur