Azure/iotedge-lorawan-starterkit #224: Issues with class C device

jsaez868, 2020-02-07:
We're running into a couple of issues with our class C device.

First issue - our class C (ABP) device sends a message to the gateway once at power up and every hour thereafter. At first, we were having issues when the NetworkSrvModule attempts to identify the device, but we were able to get past it, and so eventually we got this:

Our device:
DevAddr: 12000003
DevEUI: 0000000000000007

`2020-02-07 04:04:23.718 12000003: querying the registry for device
2020-02-07 04:04:28.749 Physical dataUp {"rxpk":[{"tmst":3425803172,"time":"2020-02-07T04:04:28.747801Z","chan":0,"rfch":0,"freq":923.200000,"stat":1,"modu":"LORA","datr":"SF8BW125","codr":"4/5","lsnr":12.0,"rssi":-39,"size":104,"data":"gAMAABKAAwAJaVmujv9obNJMdTVmG3/V/5kjwKDfuyoqea0/Q1wvNMYMvV123ClrXbshfhw8M1T7zCqh73ll1ymef59gJxWoNZX86SDxYhnwTLbIZQDjc+GCMOLIbe3g25XHFmmwG1U="}]}
2020-02-07 04:04:32.085 12000003: querying the registry for devices by devAddr 12000003 found 1 devices
2020-02-07 04:04:32.123 0000000000000007: using edgeHub local queue
2020-02-07 04:04:32.428 0000000000000007: getting device twin
2020-02-07 04:04:33.637 0000000000000007: done getting device twin
2020-02-07 04:04:34.301 0000000000000007: device added to cache
2020-02-07 04:04:34.495 0000000000000007: converted 16bit FCnt 3 to 32bit FCnt 3
2020-02-07 04:04:34.549 0000000000000007: no Deduplication Strategy selected
2020-02-07 04:04:34.596 0000000000000007: FunctionBundler ADR request finished preparing.
2020-02-07 04:04:34.665 0000000000000007: FunctionBundler request: {"GatewayId":"loragw01","ClientFCntUp":3,"ClientFCntDown":0,"Rssi":-41,"AdrRequest":{"DataRate":2,"RequiredSnr":-10.0,"PerformADRCalculation":false,"FCntUp":3,"FCntDown":0,"MinTxPowerIndex":13,"GatewayId":"loragw01","ClearCache":false},"FunctionItems":13}
2020-02-07 04:04:35.879 0000000000000007: FunctionBundler result: {"DeduplicationResult":null,"AdrResult":{"NbRepetition":null,"TxPower":null,"DataRate":0,"CanConfirmToDevice":false,"FCntDown":null,"NumberOfFrames":1},"NextFCntDown":1,"PreferredGatewayResult":{"DevEUI":"0000000000000007","RequestFcntUp":3,"CurrentFcntUp":3,"PreferredGatewayID":"loragw01"}}
2020-02-07 04:04:35.901 0000000000000007: preferred gateway changed from '' to 'loragw01'
2020-02-07 04:04:35.904 0000000000000007: down frame counter: 1
2020-02-07 04:04:35.920 0000000000000007: valid frame counter, msg: 3 server: 0
2020-02-07 04:04:35.955 0000000000000007: no decoder set in device twin. port: 9
2020-02-07 04:04:36.126 0000000000000007: sending message {"time":"2020-02-07T04:04:23.547161Z","tmms":0,"tmst":3420600068,"freq":923.2,"chan":0,"rfch":0,"stat":1,"modu":"LORA","datr":"SF8BW125","codr":"4/5","rssi":-41,"lsnr":11.8,"size":104,"data":{"value":"AQNAACoSAwEAATs6BAAAAAAAAAjKAAAThAAAAAAAAAEOAAAAAAAABtwAAAAAAIgBMy4YLhgAAAABAAAAAQAAAAEAAAABAAAAAQAAAAEAAAABAAAAAQAAAAFCTQ=="},"port":9,"fcnt":3,"rawdata":"AQNAACoSAwEAATs6BAAAAAAAAAjKAAAThAAAAAAAAAEOAAAAAAAABtwAAAAAAIgBMy4YLhgAAAABAAAAAQAAAAEAAAABAAAAAQAAAAEAAAABAAAAAQAAAAFCTQ==","eui":"0000000000000007","gatewayid":"loragw01","edgets":1581048263548} to hub
2020-02-07 04:04:37.029 0000000000000007: message '{"value":"AQNAACoSAwEAATs6BAAAAAAAAAjKAAAThAAAAAAAAAEOAAAAAAAABtwAAAAAAIgBMy4YLhgAAAABAAAAAQAAAAEAAAABAAAAAQAAAAEAAAABAAAAAQAAAAFCTQ=="}' sent to hub
2020-02-07 04:04:37.039 0000000000000007: too late for down message (00:00:13.4902565)
2020-02-07 04:04:37.183 0000000000000007: updating twin
2020-02-07 04:04:37.423 0000000000000007: twin updated
2020-02-07 04:04:37.439 0000000000000007: converted 16bit FCnt 3 to 32bit FCnt 3
2020-02-07 04:04:37.439 0000000000000007: processing time: 00:00:13.8910019
2020-02-07 04:04:37.442 0000000000000007: resubmit from confirmed message detected, msg: 3 server: 3
2020-02-07 04:04:37.443 0000000000000007: no Deduplication Strategy selected
2020-02-07 04:04:37.444 0000000000000007: FunctionBundler ADR request finished preparing.
2020-02-07 04:04:37.445 0000000000000007: FunctionBundler request: {"GatewayId":"loragw01","ClientFCntUp":3,"ClientFCntDown":1,"Rssi":-39,"AdrRequest":{"DataRate":2,"RequiredSnr":-10.0,"PerformADRCalculation":false,"FCntUp":3,"FCntDown":1,"MinTxPowerIndex":13,"GatewayId":"loragw01","ClearCache":false},"FunctionItems":13}
2020-02-07 04:04:37.729 0000000000000007: FunctionBundler result: {"DeduplicationResult":null,"AdrResult":{"NbRepetition":null,"TxPower":null,"DataRate":0,"CanConfirmToDevice":false,"FCntDown":null,"NumberOfFrames":1},"NextFCntDown":2,"PreferredGatewayResult":{"DevEUI":"0000000000000007","RequestFcntUp":3,"CurrentFcntUp":3,"PreferredGatewayID":"loragw01"}}
2020-02-07 04:04:37.730 0000000000000007: down frame counter: 2
2020-02-07 04:04:37.730 0000000000000007: no decoder set in device twin. port: 9
2020-02-07 04:04:37.730 0000000000000007: too late for down message (00:00:08.9816327)
2020-02-07 04:04:37.730 0000000000000007: processing time: 00:00:08.9819513`

For the next 2 hours, it did the same entries in the log - until the 4th one when it changed to this:
`2020-02-07 05:04:23.279 0000000000000007: device in cache
2020-02-07 05:04:23.280 0000000000000007: converted 16bit FCnt 4 to 32bit FCnt 4
2020-02-07 05:04:23.280 0000000000000007: no Deduplication Strategy selected
2020-02-07 05:04:23.280 0000000000000007: FunctionBundler ADR request finished preparing.
2020-02-07 05:04:23.281 0000000000000007: FunctionBundler request: {"GatewayId":"loragw01","ClientFCntUp":4,"ClientFCntDown":2,"Rssi":-48,"AdrRequest":{"DataRate":2,"RequiredSnr":-10.0,"PerformADRCalculation":false,"FCntUp":4,"FCntDown":2,"MinTxPowerIndex":13,"GatewayId":"loragw01","ClearCache":false},"FunctionItems":13}
2020-02-07 05:04:24.537 0000000000000007: FunctionBundler result: {"DeduplicationResult":null,"AdrResult":{"NbRepetition":null,"TxPower":null,"DataRate":0,"CanConfirmToDevice":false,"FCntDown":null,"NumberOfFrames":2},"NextFCntDown":3,"PreferredGatewayResult":{"DevEUI":"0000000000000007","RequestFcntUp":4,"CurrentFcntUp":4,"PreferredGatewayID":"loragw01"}}
2020-02-07 05:04:24.537 0000000000000007: down frame counter: 3
2020-02-07 05:04:24.538 0000000000000007: valid frame counter, msg: 4 server: 3
2020-02-07 05:04:24.538 0000000000000007: no decoder set in device twin. port: 9
2020-02-07 05:04:24.539 0000000000000007: sending message {"time":"2020-02-07T05:04:23.270970Z","tmms":0,"tmst":2725358868,"freq":923.4,"chan":1,"rfch":0,"stat":1,"modu":"LORA","datr":"SF8BW125","codr":"4/5","rssi":-48,"lsnr":11.2,"size":104,"data":{"value":"AQNAACoSAwEAAjs6BAAAAAAAAAjUAAATgwAAAAAAAAEOAAAAAAAABtwAAAAAAIgBNS4YLhgAAAABAAAAAQAAAAEAAAABAAAAAQAAAAEAAAABAAAAAQAAAAHOQQ=="},"port":9,"fcnt":4,"rawdata":"AQNAACoSAwEAAjs6BAAAAAAAAAjUAAATgwAAAAAAAAEOAAAAAAAABtwAAAAAAIgBNS4YLhgAAAABAAAAAQAAAAEAAAABAAAAAQAAAAEAAAABAAAAAQAAAAHOQQ==","eui":"0000000000000007","gatewayid":"loragw01","edgets":1581051863271} to hub
2020-02-07 05:04:24.580 0000000000000007: message '{"value":"AQNAACoSAwEAAjs6BAAAAAAAAAjUAAATgwAAAAAAAAEOAAAAAAAABtwAAAAAAIgBNS4YLhgAAAABAAAAAQAAAAEAAAABAAAAAQAAAAEAAAABAAAAAQAAAAHOQQ=="}' sent to hub
2020-02-07 05:04:24.644 0000000000000007: checking cloud to device message for 00:00:00.2827146
2020-02-07 05:04:25.172 0000000000000007: done checking cloud to device message, found no message
2020-02-07 05:04:25.227 0000000000000007: processing time: 00:00:01.9553016
2020-02-07 05:04:29.245 Physical dataUp {"rxpk":[{"tmst":2731325332,"time":"2020-02-07T05:04:29.243155Z","chan":1,"rfch":0,"freq":923.400000,"stat":1,"modu":"LORA","datr":"SF8BW125","codr":"4/5","lsnr":11.8,"rssi":-51,"size":104,"data":"gAMAABKABAAJNcjOmijRpSOokCoVqheNtFqSRKV9xZh/PwE3Klmy6wyfDCkSFK2lbQ34kUXJYy/E7Ixs1mgAKpS746SFXJuh+yYPrNWWpmV74cvFRIquZkb+Jt/hkiQPeEAtBaYi7Dg="}]}
2020-02-07 05:04:29.246 0000000000000007: device in cache
2020-02-07 05:04:29.246 0000000000000007: converted 16bit FCnt 4 to 32bit FCnt 4
2020-02-07 05:04:29.246 0000000000000007: resubmit from confirmed message detected, msg: 4 server: 4
2020-02-07 05:04:29.247 0000000000000007: no Deduplication Strategy selected
2020-02-07 05:04:29.247 0000000000000007: FunctionBundler ADR request finished preparing.
2020-02-07 05:04:29.247 0000000000000007: FunctionBundler request: {"GatewayId":"loragw01","ClientFCntUp":4,"ClientFCntDown":3,"Rssi":-51,"AdrRequest":{"DataRate":2,"RequiredSnr":-10.0,"PerformADRCalculation":false,"FCntUp":4,"FCntDown":3,"MinTxPowerIndex":13,"GatewayId":"loragw01","ClearCache":false},"FunctionItems":13}
[... cut ...]
`2020-02-07 07:56:24.290 0000000000000007: [class-c] device does not have a region assigned. Ensure the device has connected at least once with the network`

The latter one is actually executed earlier in the DefaultClassCDevicesMessageSender.SendAsync method - so it seems we were getting past it at 05:34:54.901, and then later at 07:56:24.290 - we're not even getting there. So question - where do we need to configure everything that's getting validated in this DefaultClassCDevicesMessageSender.SendAsync method? There doesn't seem to be anything in the documentation.

Thank you.

jsaez868, 2020-02-11:
We've resolved it

closed 2020-02-11 by jsaez868

ronniesa, 2020-02-11:
Out of curiosity what was the solution so it may help others?

jsaez868, 2020-02-11:
Nothing too creative - just restarted everything and including clearing redis cache =)

A couple of follow up questions unrelated - if we needed to support a new region, where would be a good starting point? and what are all the different components we have to implement? Secondly, is the NetworkSrvModule compatible with the TTN packet forwarder?

ronniesa, 2020-02-13:
Glad that it worked out.

What frequencies/regions you would like to add? As you know currently we support EU and US only mostly due to testing challanges. 
We can provide some guidance on which classes you would need to modify to support other regions. 
The first step for a new frequency/region is to test as is with the right packet forwarder and config settings and use ABP, no ADR and only unconfirmed up messages, as this does not require any down stream communication it should work out of the box. Once you get this you need to implement the settings in the classes that we can guide you through to get down stream communications that will enable OTAA, downstream messages, acknowledgement, ADR etc...
If you want to go this route we can support you a bit so hopefully you can PR back the changes so we can support other regions.

NetworkSrvModule is compatible with the Semtech Packet Forwarder, as basically almost all gateways  ssupport this or have a forked one they works with our NetworkSrvModule. Never tested with the TTN one but I believe they have some particularities to communicate with their network server so I doubt it will work.

jsaez868, 2020-02-13:
We're looking to support AS923 region. We've already tested the kit and we see upstream messages getting to the hub. The issue we're running into now is with downstream messages - they're not reaching the device. Please provide guidance on what we need to implement in order to fully support this new region.

For downstream messages from an ABP class C device, how would the NetworkSrvModule resolve which region to use? Do we have to set any device twin info?

Thanks.

ronniesa, 2020-02-14:
Let's move the conversation here: https://github.com/Azure/iotedge-lorawan-starterkit/issues/225
As said we can help you a bit @Mandur in my team owns the regions parts

We infer the frequency by the upstream message. So as with others NtwSrv a Class C needs first to send an upstream messagee or a join.