Azure/iotedge-lorawan-starterkit #225: AS923 Support

Mandur, 2020-02-14:
**Is your feature request related to a problem? Please describe.**
Support of AS923 frequency range

**Describe the solution you'd like**
LoRa starter kit should support AS 923 freq range

**Describe alternatives you've considered**
N/A

**Additional context**
Requested by community

jsaez868, 2020-02-14:
OK, so we figured we need to implement the new region in the RegionManager using https://lora-alliance.org/sites/default/files/2018-04/lorawantm_regional_parameters_v1.1rb_-_final.pdf as reference, correct?

Mandur, 2020-02-16:
Hello! Thank you for the interest, yes you are Correct. I tried here below to list the changes needed to implement a new region. In case you have questions or you need additional help, I'd be happy to help you out. My skype handle is mandulevrai.

* Add a region enum in [LoraRegion](https://github.com/Azure/iotedge-lorawan-starterkit/blob/master/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoraTools/Regions/LoRaRegionType.cs)
* Add correct mapping of downstream frequency in [TryGetUpstreamChannelFrequency](https://github.com/Azure/iotedge-lorawan-starterkit/blob/4815a61cc3822ad9a24bd69518f41201b0b26799/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoraTools/Regions/Region.cs#L124) .
* most changes need to be done in [RegionManager](https://github.com/Azure/iotedge-lorawan-starterkit/blob/master/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoraTools/Regions/RegionManager.cs)
  * We are using upstream frequency of first messages to detect current region. I would suggest changing max us frequencies to 915 and setting as range of 923-925 for the as region. [TryResolveRegion](https://github.com/Azure/iotedge-lorawan-starterkit/blob/4815a61cc3822ad9a24bd69518f41201b0b26799/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoraTools/Regions/RegionManager.cs#L49)
  * Create the region similarly to [CreateEU868Region](https://github.com/Azure/iotedge-lorawan-starterkit/blob/4815a61cc3822ad9a24bd69518f41201b0b26799/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoraTools/Regions/RegionManager.cs#L89) with adequate settings from the spec. 
  * Add your new region to all the places where needed (like in [TryTranslateToRegion](https://github.com/Azure/iotedge-lorawan-starterkit/blob/4815a61cc3822ad9a24bd69518f41201b0b26799/LoRaEngine/modules/LoRaWanNetworkSrvModule/LoraTools/Regions/RegionManager.cs#L26)

jsaez868, 2020-02-17:
Hi @Mandur I've also sent you an invite to connect on Skype.

Here are my responses to your last post:

- Add a region enum in LoraRegion
**We've added new AS923 LoRaRegionType.**
- Add correct mapping of downstream frequency in TryGetUpstreamChannelFrequency .
**We can't seem to find info about this on the specs. Does AS923 behave like EU868 region (respond on same frequency)?**
- most changes need to be done in RegionManager
  - We are using upstream frequency of first messages to detect current region. I would suggest changing max us frequencies to 915 and setting as range of 923-925 for the as region. TryResolveRegion

  **Should we set it to that range even if specs states the following?** 
  > This section applies to regions where the frequencies [923…923.5MHz] are comprised...
  - Create the region similarly to CreateEU868Region with adequate settings from the spec.
**This is exactly what we did. For RX1DROffsetTable , did we do the following correctly:**

>             as923.RX1DROffsetTable = new int[1, 8]
>             {
>             { 0, 1, 2, 3, 4, 5, -1, -2 }
>             };

  - Add your new region to all the places where needed (like in TryTranslateToRegion
**We did this.**

**Based on our initial testing, C2D messages are still not making it downstream**. On the log, I see the following being inferred/resolved by the kit incorrectly:

> using device rx2: 0, datr: SF12BW125, region: AS923

Values used look incorrect - they should have been rx2dr = 2 and datr = SF10BW125.

Per specs => _The RX2 receive window uses a fixed frequency and data rate. The default parameters are 923.2 MHz / DR2 (SF10/125KHz)._ which is what we've coded for the region:

>  new Region(
>                 LoRaRegionType.AS923,
>                 0x34,
>                 ConversionHelper.StringToByteArray("C194C1"),
>                 (frequency: 923.2, datr: 2),
>                 1,
>                 2,
>                 5,
>                 6,
>                 16384,
>                 64,
>                 32,
>                 (min: 1, max: 3));

How does the kit resolve the region to use for a downstream message? Using the last upstream message from the device (our device sends some sort of status/heartbeat message every hour)?

The log entry above seems to have come from the following lines of code in Region.cs:

>                 uint rx2Dr = (uint)rx2DrFromTwins;
>                 if (this.RegionLimits.IsCurrentDownstreamDRIndexWithinAcceptableValue(rx2Dr))
>                 {
>                     datr = this.DRtoConfiguration[rx2Dr].configuration;
>                     Logger.Log(devEUI, $"using device rx2: {rx2Dr}, datr: {datr}, region: {this.LoRaRegion}", LogLevel.Debug);
>                 }
> 

RX2 value it's using seems incorrect. We've set the RX2DataRate property for the device twin to "**2**" - so it's in the desired properties but currently not in the reported properties. Our device is class C, ABP device.

These are the log entries from last hourly upstream message from the device:

```
2020-02-17 06:30:44.421 0000000000000007: device in cache
2020-02-17 06:30:44.421 0000000000000007: converted 16bit FCnt 99 to 32bit FCnt 99
2020-02-17 06:30:44.422 0000000000000007: ADR ack request received
2020-02-17 06:30:44.422 0000000000000007: calculated ADR: CanConfirmToDevice: True, TxPower: 7, DataRate: 5
2020-02-17 06:30:44.422 0000000000000007: device sent ADR ack request, computing an answer
2020-02-17 06:30:44.422 0000000000000007: down frame counter: 234
2020-02-17 06:30:44.422 0000000000000007: valid frame counter, msg: 99 server: 98
[... 9 lines cut ...]
2020-02-17 06:30:49.647 0000000000000007: converted 16bit FCnt 99 to 32bit FCnt 99
2020-02-17 06:30:49.647 0000000000000007: resubmit from confirmed message detected, msg: 99 server: 99
2020-02-17 06:30:49.647 0000000000000007: ADR ack request received
2020-02-17 06:30:49.647 0000000000000007: calculated ADR: CanConfirmToDevice: True, TxPower: 7, DataRate: 5
2020-02-17 06:30:49.648 0000000000000007: device sent ADR ack request, computing an answer
2020-02-17 06:30:49.648 0000000000000007: down frame counter: 235
2020-02-17 06:30:49.648 0000000000000007: no decoder set in device twin. port: 9
2020-02-17 06:30:49.651 0000000000000007: checking cloud to device message for 00:00:00.5969923
2020-02-17 06:30:50.249 0000000000000007: done checking cloud to device message, found no message
2020-02-17 06:30:50.255 0000000000000007: processing time: 00:00:00.6105793
2020-02-17 06:30:50.255 0000000000000007: error processing request: Index was outside the bounds of the array.
```

The last downstream message we sent C2D logged the following entries, but it doesn't look like it reached the device since we didn't see expected result:

```
2020-02-17 06:47:32.340 0000000000000007: received cloud to device message from direct method: {"devEUI":"0000000000000007","fport":8,"rawPayload":"AQbfCgBkk/c=","confirmed":true}
2020-02-17 06:47:32.341 0000000000000007: [class-c] down frame counter: 236
2020-02-17 06:47:32.343 0000000000000007: using device rx2: 0, datr: SF12BW125, region: AS923
2020-02-17 06:47:32.344 0000000000000007: cloud to device message: 0106DF0A006493F7, id: undefined, fport: 8, confirmed: True, cidType: Zero
2020-02-17 06:47:32.345 0000000000000007: sending a downstream message with ID 09C9
2020-02-17 06:47:32.346 0000000000000007: ConfirmedDataDown {"txpk":{"imme":true,"data":"oAMAABIA7AAIPgfY4Sx+K9fOY0hM","tmst":0,"size":21,"freq":923.2,"rfch":0,"modu":"LORA","datr":"SF12BW125","codr":"4/5","powe":14,"ipol":true}}
2020-02-17 06:47:32.347 UDP: sending message with ID 5769, to 172.18.0.1:47169
2020-02-17 06:47:32.349 UDP: message sent with ID 5769
```

jsaez868, 2020-02-17:
Here's the device twin info:

```
{
  "deviceId": "0000000000000007",
  "etag": "AAAAAAAAAAo=",
  "deviceEtag": "NjQ5MDEzMzYy",
  "status": "enabled",
  "statusUpdateTime": "0001-01-01T00:00:00Z",
  "connectionState": "Connected",
  "lastActivityTime": "2020-02-17T07:30:44.165017Z",
  "cloudToDeviceMessageCount": 0,
  "authenticationType": "sas",
  "x509Thumbprint": {
    "primaryThumbprint": null,
    "secondaryThumbprint": null
  },
  "version": 60,
  "properties": {
    "desired": {
      "AppSKey": "a8a6a0a7a6a9a8a8a0a1a0a8a8a8a0a0",
      "NwkSKey": "a8a6a0a7a6a9a8a8a0a1a0a8a8a8a0a0",
      "DevAddr": "12000003",
      "GatewayID": "loragw01",
      "SensorDecoder": "",
      "ClassType": "C",
      "RX1DROffset": 0,
      "FCntUpStart": 1,
      "FCntDownStart": 1,
      "FCntResetCounter": 1,
      "ABPRelaxMode": true,
      "RX2DataRate": 2,
      "$metadata": {
        "$lastUpdated": "2020-02-15T20:17:55.0966586Z",
        "$lastUpdatedVersion": 10,
        "AppSKey": {
          "$lastUpdated": "2020-02-15T20:17:55.0966586Z",
          "$lastUpdatedVersion": 10
        },
        "NwkSKey": {
          "$lastUpdated": "2020-02-15T20:17:55.0966586Z",
          "$lastUpdatedVersion": 10
        },
        "DevAddr": {
          "$lastUpdated": "2020-02-15T20:17:55.0966586Z",
          "$lastUpdatedVersion": 10
        },
        "GatewayID": {
          "$lastUpdated": "2020-02-15T20:17:55.0966586Z",
          "$lastUpdatedVersion": 10
        },
        "SensorDecoder": {
          "$lastUpdated": "2020-02-15T20:17:55.0966586Z",
          "$lastUpdatedVersion": 10
        },
        "ClassType": {
          "$lastUpdated": "2020-02-15T20:17:55.0966586Z",
          "$lastUpdatedVersion": 10
        },
        "RX1DROffset": {
          "$lastUpdated": "2020-02-15T20:17:55.0966586Z",
          "$lastUpdatedVersion": 10
        },
        "FCntUpStart": {
          "$lastUpdated": "2020-02-15T20:17:55.0966586Z",
          "$lastUpdatedVersion": 10
        },
        "FCntDownStart": {
          "$lastUpdated": "2020-02-15T20:17:55.0966586Z",
          "$lastUpdatedVersion": 10
        },
        "FCntResetCounter": {
          "$lastUpdated": "2020-02-15T20:17:55.0966586Z",
          "$lastUpdatedVersion": 10
        },
        "ABPRelaxMode": {
          "$lastUpdated": "2020-02-15T20:17:55.0966586Z",
          "$lastUpdatedVersion": 10
        },
        "RX2DataRate": {
          "$lastUpdated": "2020-02-15T20:17:55.0966586Z",
          "$lastUpdatedVersion": 10
        }
      },
      "$version": 10
    },
    "reported": {
      "PreferredGatewayID": "loragw01",
      "Region": "AS923",
      "FCntDown": 232,
      "FCntUp": 97,
      "FCntResetCounter": 1,
      "FCntUpStart": 1,
      "FCntDownStart": 1,
      "DataRate": 5,
      "NbRep": 1,
      "TxPower": 7,
      "$metadata": {
        "$lastUpdated": "2020-02-17T04:30:45.6044466Z",
        "PreferredGatewayID": {
          "$lastUpdated": "2020-02-07T04:04:38.7074891Z"
        },
        "Region": {
          "$lastUpdated": "2020-02-15T18:31:29.4094113Z"
        },
        "FCntDown": {
          "$lastUpdated": "2020-02-17T04:30:45.6044466Z"
        },
        "FCntUp": {
          "$lastUpdated": "2020-02-17T04:30:45.6044466Z"
        },
        "FCntResetCounter": {
          "$lastUpdated": "2020-02-07T05:14:39.4018464Z"
        },
        "FCntUpStart": {
          "$lastUpdated": "2020-02-07T18:04:38.1527326Z"
        },
        "FCntDownStart": {
          "$lastUpdated": "2020-02-07T18:04:38.1527326Z"
        },
        "DataRate": {
          "$lastUpdated": "2020-02-16T16:30:48.7563306Z"
        },
        "NbRep": {
          "$lastUpdated": "2020-02-15T21:30:55.1405018Z"
        },
        "TxPower": {
          "$lastUpdated": "2020-02-16T16:30:48.7563306Z"
        }
      },
      "$version": 50
    }
  },
  "capabilities": {
    "iotEdge": false
  }
}
```

Mandur, 2020-08-10:
@jsaez868 Would you still be interrested in merging the AS923 frequency plan with upstream?

closed 2021-09-29 by roel4ez