Azure/iotedge-lorawan-starterkit #228: Gateway stops sending data for devices after running for a period of time.

gerfen, 2020-04-16:
I have a gateway running on the 1.0.2 release.  It's been operational for many months - since September 2019.  Over those months I have experienced the gateway no longer sending data for devices after many weeks of operation - this usually occurs after an interruption to internet connectivity.  If the gateway is rebooted, all of the associated devices reconnect and start sending data until the internet connectivity is interrupted.   I am currently rebooting the gateway once a week which has helped but I really need to get to the bottom of the issue. 

Unfortunately I am not able to remotely connect to the gateway to collect the log files without the assistance of IT personnel who are sadly not co-located at the facility so I must schedule assistance in advance.  I know that I can have the LoRaWanNetworkSrvModule send logs to the IoT Hub infrastructure but unfortunately I have also experienced the LoRaWanNetworkSrvModule not restarting properly when I turn on logging from the IoT Hub portal.  This leaves the gateway in a broken state until I can get IT assistance.

I tried clearing the Redis cache from the Azure portal but this has no effect.  As noted toggling remote logging from the LoRaWanNetworkSrvModule was effective at resolving the issue since it caused the module to restart but as noted in recent months the module has failed to restart which leaves me in a worse situation.

I really don't want to force a gateway reboot on a daily basis as it's masking the  core issue. I'd appreciate any insight into how I might manage this situation.  Being able to force an on demand reboot from the IoT Hub portal would be helpful.

### Device (Host) Operating System

Ubuntu 16.04
### Architecture

x64

ronniesa, 2020-04-16:
One trick that you can do is to change a dummy env variable on each module this will trigger a restart of every module. Maybe is worth to try just forcing the restart of the edgeHub module that could be the cause of not correctly reconnecting. 

If you have a lot of network disconnects is also recommended to put a DNS settings in to edge: https://docs.microsoft.com/en-us/azure/iot-edge/troubleshoot#edge-agent-module-continually-reports-empty-config-file-and-no-modules-start-on-the-device

Yes, it would be nice to be able to force a reboot of edge from IoT Hub. 
You could implement a module that runs in priviledged mode and allow you to call a directmethod to trigger the reboot. Not probably a best practice but it should work.

gerfen, 2020-04-16:
@ronniesa Thanks for the suggestions.  I think I will start with the DNS settings to see if that improves the situation.

Mandur, 2020-04-20:
Hello @gerfen, I had luckily a similar requirement for our ci. Sadly my method don't work on x86, but I found a more aggressive one that can work with x86 devices :  https://github.com/Mandur/IoTEdgeDeviceRebooter.

ronniesa, 2020-05-14:
@gerfen  did the DNS entry solve the issue?

gerfen, 2020-05-14:
@ronniesa Unfortunately I did not have time to try to add the DNS entry while I was on site as I got side tracked with not be able to join my new devices to the gateway.  I have plans to return in two weeks and will attempt to add the DNS entry then.

I can close this issue and re-open i if necessary.

ronniesa, 2020-05-14:
@gerfen let's leave it open for now until you test it out so we can better track it

AxelBertilsson, 2020-06-04:
@gerfen Any progress on the issue?

gerfen, 2020-06-04:
Unfortunately no due to the current situation here in the States.  I will hopefully be back on site in two weeks.  We can close the issue and reopen it if appropriate.

Mandur, 2020-08-10:
@gerfen Were you able to test the proposed solution?

gerfen, 2020-08-10:
@mandur No I haven't been able to test the solution.  That said the system has stabilized.  The real issue was related to the max number of devices getting reset to 100 when I did an upgrade.  It took me ages to figure that out.  I'll close this issue now.  Sorry for the fire drill.

closed 2020-08-10 by gerfen