Azure/iotedge-lorawan-starterkit #229: LoRaEngin gives MIC validation error with ABP message from Simulator

pietrobr, 2020-04-19:
I want to test the Lora Engine separated by the Simulator as  described in the documentation.
I've created a simple ABP device in testconfig.json used for both Engine and Simulator. I've added the AppSKey, NetSkey, DevAddr as desired props in IoT Hub.

Now if I run from VS first the Simulator and then the Engine I seed that the device is found in IoT hub, but I get an error validating the MIC in the Engine

I didn't expect a validation error for MIC, debugging with VS it seems related to the 32SupportdFcnt that should be true to go on, but in my case it's false.

What am I missing?

ronniesa, 2020-04-19:
Let us double check as we don't use the simulator lately as the CI is using real gateway and devices and the unit tests use a bit different code.  As the 32SupportdFcnt was probably added after the simulator was completed. @Mandur do you have an idea?

One quick test is to add to your device twin: "Supports32BitFCnt": false. Try also with true. As we do caching a restart (or call the clearcache method) of the netwrk server module (ev. clear the Redis cache too) would be required.

pietrobr, 2020-04-19:
I try to add a desiresd property "Supports32BitFCnt" and set it yo true, now the interal property of the Engine is set to true, but the MIC validation fails anyway in function payload.CheckMic

ronniesa, 2020-04-19:
I would tend to believe that for the Simulator as with most of the Lora devices Supports32BitFCnt must be set to false. This is also the default.  
Are you sure that the keys are not inverted? Maybe you could try to use the same value for the ntw and the app key.
Could be that we have a bug in the Simulator

pietrobr, 2020-04-20:
I set Supports32BitFCnt  to false, that was the default value and the MIC validation fails.

Setting AppSKey, NetSkey if they are the same doesn't seem affecting the MIC validation from a quick test.

Question: having a quick look at the code of the Engine I don't see the use of the testconfig.json, where is it used? 

does the  payload.CheckMic in the Engine need the testconfig.json?

Mandur, 2020-04-20:
Hello @pietrobr, Thanks for reporting the issue. I think the problem is our documentation that is missing an element. Could you please try to add the following settings to your debug settings of the LoRaEngine : IOTEDGE_DEVICEID to "gateway". PS. You could set it to any string, as you are not using gateway mode I don't think it will matter

pietrobr, 2020-04-20:
Hi @Mandur I've tried to set the string "gateway" with and without quotes, but nothing changed.

I'm not sure if the IOTEDGE_DEVIEID is used in payload.CheckMic that it's where the MIC validation fails.

Mandur, 2020-04-20:
Hmmm, sorry you are right, when I tried to replicate your error this is the problem I had... I am unable to replicate the error you are facing :(. It would be of great help to get the settings you are using so I could reproduce

ronniesa, 2020-04-21:
As the problem was the missmatch of the ntw and app key we close the issue as resolved.

closed 2020-04-21 by ronniesa