Azure/iotedge-lorawan-starterkit #237: Class C devices are not receiving direct messages

gerfen, 2020-05-15:
I'm having an issue with some new Class C devices, I have older versions of the same device which work as expected.  I'm not sure if this related to [AppKey and AppEUI issue ](https://github.com/Azure/iotedge-lorawan-starterkit/issues/236)- lowercase vs upper case.  I have configured the new devices exactly as I configured the older version of the same device.

Here are the network server logs when a direct message is received:
```
2020-05-15 16:46:24.277 Statistic: {"stat":{"time":"2020-05-15 16:46:24 GMT","rxnb":2,"rxok":1,"rxfw":1,"ackr":100.0,"dwnb":0,"txnb":0}}
2020-05-15 16:46:37.884 24E1611693965996: received cloud to device message from direct method: {"devEUI":"24E1611693965996","fport":"55","confirmed":false,"payload":"CurtainDown","MessageId":"79d963f35f834250a180d4cd78660740"}
2020-05-15 16:46:37.884 24E1611693965996: [class-c] down frame counter: 11
2020-05-15 16:46:37.885 24E1611693965996: using device rx2: 10, datr: SF10BW500, region: US915
2020-05-15 16:46:37.886 24E1611693965996: cloud to device message: 4375727461696E446F776E, id: 79d963f35f834250a180d4cd78660740, fport: 55, confirmed: False, cidType: Zero
2020-05-15 16:46:37.886 24E1611693965996: sending a downstream message with ID D914
2020-05-15 16:46:37.886 24E1611693965996: UnconfirmedDataDown {"txpk":{"imme":true,"data":"YHzudwIACwA3pdc05aaKR8iFWkNgR4Gz","tmst":0,"size":24,"freq":923.3,"rfch":0,"modu":"LORA","datr":"SF10BW500","codr":"4/5","powe":14,"ipol":true}}
2020-05-15 16:46:37.886 UDP: sending message with ID 8F7A, to 172.18.0.1:52297
2020-05-15 16:46:37.887 UDP: message sent with ID 8F7A
2020-05-15 16:46:37.888 UDP: packet with id 8F7A successfully transmitted by the aggregator
2020-05-15 16:46:38.013 24E1611693947029: received cloud to device message from direct method: {"devEUI":"24E1611693947029","fport":"55","confirmed":false,"payload":"CurtainDown","MessageId":"80108a1a2fbd4ef3ad7f948c4e9c6d32"}
2020-05-15 16:46:38.013 24E1611693947029: [class-c] down frame counter: 11
2020-05-15 16:46:38.013 24E1611693947029: using device rx2: 10, datr: SF10BW500, region: US915
2020-05-15 16:46:38.014 24E1611693947029: cloud to device message: 4375727461696E446F776E, id: 80108a1a2fbd4ef3ad7f948c4e9c6d32, fport: 55, confirmed: False, cidType: Zero
2020-05-15 16:46:38.014 24E1611693947029: sending a downstream message with ID EEA6
2020-05-15 16:46:38.014 24E1611693947029: UnconfirmedDataDown {"txpk":{"imme":true,"data":"YHSakAMACwA3gQ/lndnHfY/OTMMd00S2","tmst":0,"size":24,"freq":923.3,"rfch":0,"modu":"LORA","datr":"SF10BW500","codr":"4/5","powe":14,"ipol":true}}
2020-05-15 16:46:38.014 UDP: sending message with ID C617, to 172.18.0.1:52297
2020-05-15 16:46:38.017 UDP: message sent with ID C617
2020-05-15 16:46:38.018 UDP: packet with id C617 successfully transmitted by the aggregator

```

And here are the logs from the packet forwarder:
```
JSON up: {"stat":{"time":"2020-05-15 16:45:54 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}
INFO: [up] PUSH_ACK received in 0 ms
INFO: [down] PULL_ACK received in 0 ms
INFO: [down] PULL_ACK received in 0 ms

INFO: Received pkt from mote: 03FDA89F (fcnt=59497)

JSON up: {"rxpk":[{"tmst":2815279628,"chan":4,"rfch":1,"freq":903.100000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":6.5,"rssi":-35,"size":18,"data":"QJ+o/QMAaegEIdyfWFbpJyKp"}]}
INFO: [up] PUSH_ACK received in 1 ms

INFO: Disabling GPS mode for concentrator's counter...
INFO: host/sx1301 time offset=(1589558364s:830447µs) - drift=-182µs
INFO: Enabling GPS mode for concentrator's counter.

##### 2020-05-15 16:46:24 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 2
# CRC_OK: 50.00%, CRC_FAIL: 50.00%, NO_CRC: 0.00%
# RF packets forwarded: 1 (18 bytes)
# PUSH_DATA datagrams sent: 2 (312 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 2 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
# TX rejected (collision packet): 0.00% (req:77, rej:0)
# TX rejected (collision beacon): 0.00% (req:77, rej:0)
# TX rejected (too late): 0.00% (req:77, rej:0)
# TX rejected (too early): 0.00% (req:77, rej:0)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 2939226827
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# GPS sync is disabled
##### END #####

```

Here is my redacted twins for one of the devices:

```
{
  "deviceId": "24E1611693965996",
  "etag": "AAAAAAAAAAg=",
  "deviceEtag": "OTQ4MjI3MjQy",
  "status": "enabled",
  "statusUpdateTime": "0001-01-01T00:00:00Z",
  "connectionState": "Connected",
  "lastActivityTime": "2020-05-15T15:40:38.4191822Z",
  "cloudToDeviceMessageCount": 0,
  "authenticationType": "sas",
  "x509Thumbprint": {
    "primaryThumbprint": null,
    "secondaryThumbprint": null
  },
  "version": 27,
  "properties": {
    "desired": {
      "AppEUI": "*****************",
      "AppKey": "**************************************",
      "SensorDecoder": "http://decoder/api/DecodeUrsaLinkUc1114",
      "GatewayID": "evolution-indoor",
      "ClassType": "C",
      "RX2DataRate": 10,
      "$metadata": {
        "$lastUpdated": "2020-05-14T17:16:02.6554813Z",
        "$lastUpdatedVersion": 8,
        "AppEUI": {
          "$lastUpdated": "2020-05-14T17:16:02.6554813Z",
          "$lastUpdatedVersion": 8
        },
        "AppKey": {
          "$lastUpdated": "2020-05-14T17:16:02.6554813Z",
          "$lastUpdatedVersion": 8
        },
        "SensorDecoder": {
          "$lastUpdated": "2020-05-14T17:16:02.6554813Z",
          "$lastUpdatedVersion": 8
        },
        "GatewayID": {
          "$lastUpdated": "2020-05-14T17:16:02.6554813Z",
          "$lastUpdatedVersion": 8
        },
        "ClassType": {
          "$lastUpdated": "2020-05-14T17:16:02.6554813Z",
          "$lastUpdatedVersion": 8
        },
        "RX2DataRate": {
          "$lastUpdated": "2020-05-14T17:16:02.6554813Z",
          "$lastUpdatedVersion": 8
        }
      },
      "$version": 8
    },
    "reported": {
      "AppSKey": "",
      "NwkSKey": "",
      "DevAddr": "0277EE7C",
      "FCntDown": 7,
      "FCntUp": 70,
      "DevEUI": "24E1611693965996",
      "NetId": "010000",
      "DevNonce": "89F3",
      "Region": "US915",
      "RX2DataRate": 10,
      "$metadata": {
        "$lastUpdated": "2020-05-15T15:10:33.937785Z",
        "AppSKey": {
          "$lastUpdated": "2020-05-14T22:38:36.476482Z"
        },
        "NwkSKey": {
          "$lastUpdated": "2020-05-14T22:38:36.476482Z"
        },
        "DevAddr": {
          "$lastUpdated": "2020-05-14T22:38:36.476482Z"
        },
        "FCntDown": {
          "$lastUpdated": "2020-05-15T15:10:33.937785Z"
        },
        "FCntUp": {
          "$lastUpdated": "2020-05-15T15:10:33.937785Z"
        },
        "DevEUI": {
          "$lastUpdated": "2020-05-14T22:38:36.476482Z"
        },
        "NetId": {
          "$lastUpdated": "2020-05-14T22:38:36.476482Z"
        },
        "DevNonce": {
          "$lastUpdated": "2020-05-14T22:38:36.476482Z"
        },
        "Region": {
          "$lastUpdated": "2020-05-14T16:53:46.3250049Z"
        },
        "RX2DataRate": {
          "$lastUpdated": "2020-05-14T22:38:36.476482Z"
        }
      },
      "$version": 19
    }
  },
  "capabilities": {
    "iotEdge": false
  },
  "deviceScope": "ms-azure-iot-edge://"
}
```

Any insight into why my old devices receive messages but the new devices do not would be greatly appreciated.

ronniesa, 2020-05-15:
From a first look this class C device did not join the network as this part should contain the valid keys that were generated during a join and not empty:
"reported": {
      "AppSKey": "",
      "NwkSKey": "", 

Did the device try to do the join?

gerfen, 2020-05-15:
@ronniesa I just forced a rejoin on one of the devices.  Here's the log for that event:

```
2020-05-15 17:09:24.302 Statistic: {"stat":{"time":"2020-05-15 17:09:24 GMT","rxnb":2,"rxok":2,"rxfw":2,"ackr":100.0,"dwnb":1,"txnb":1}}
2020-05-15 17:09:28.280 Physical dataUp {"rxpk":[{"tmst":4203425820,"chan":5,"rfch":1,"freq":903.300000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":12.0,"rssi":-21,"size":21,"data":"QGta4wIAAAA3LuDYtYwzXiPc1mxz"}]}
2020-05-15 17:09:28.281 24E1611693965996: device in cache
2020-05-15 17:09:28.283 24E1611693965996: converted 16bit FCnt 0 to 32bit FCnt 0
2020-05-15 17:09:28.283 24E1611693965996: valid frame counter, msg: 0 server: 0
2020-05-15 17:09:28.283 24E1611693965996: decoding with: http://decoder/api/DecodeUrsaLinkUc1114 port: 55
2020-05-15 17:09:28.289 24E1611693965996: sending message {"time":null,"tmms":0,"tmst":4203425820,"freq":903.3,"chan":5,"rfch":1,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","rssi":-21,"lsnr":12.0,"size":21,"data":{"reportType":"UnknownMessageReport","message":"�\ba\u0016��Y�"},"port":55,"fcnt":0,"rawdata":"/whhFpOWWZY=","eui":"24E1611693965996","gatewayid":"evolution-indoor","edgets":1589562568279} to hub
2020-05-15 17:09:28.316 24E1611693965996: message '{"reportType":"UnknownMessageReport","message":"�\ba\u0016��Y�"}' sent to hub
2020-05-15 17:09:28.317 24E1611693965996: checking cloud to device message for 00:00:00.5623506
2020-05-15 17:09:28.885 24E1611693965996: done checking cloud to device message, found no message
2020-05-15 17:09:28.885 24E1611693965996: processing time: 00:00:00.6061176
2020-05-15 17:09:33.371 Physical dataUp {"rxpk":[{"tmst":4208511892,"chan":3,"rfch":0,"freq":902.900000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":12.2,"rssi":-21,"size":21,"data":"QGta4wIAAQA3yfI8NeT7FhiW7Oi/"}]}
2020-05-15 17:09:33.380 24E1611693965996: device in cache
2020-05-15 17:09:33.381 24E1611693965996: converted 16bit FCnt 1 to 32bit FCnt 1
2020-05-15 17:09:33.381 24E1611693965996: valid frame counter, msg: 1 server: 0
2020-05-15 17:09:33.381 24E1611693965996: decoding with: http://decoder/api/DecodeUrsaLinkUc1114 port: 55
2020-05-15 17:09:33.384 24E1611693965996: sending message {"time":null,"tmms":0,"tmst":4208511892,"freq":902.9,"chan":3,"rfch":0,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","rssi":-21,"lsnr":12.2,"size":21,"data":{"reportType":"UnknownMessageReport","message":"�\t\u0002\u0010�\n\u0002\u0015"},"port":55,"fcnt":1,"rawdata":"/wkCEP8KAhU=","eui":"24E1611693965996","gatewayid":"evolution-indoor","edgets":1589562573370} to hub
2020-05-15 17:09:33.411 24E1611693965996: message '{"reportType":"UnknownMessageReport","message":"�\t\u0002\u0010�\n\u0002\u0015"}' sent to hub
2020-05-15 17:09:33.412 24E1611693965996: checking cloud to device message for 00:00:00.5585188
2020-05-15 17:09:33.973 24E1611693965996: done checking cloud to device message, found no message
2020-05-15 17:09:33.973 24E1611693965996: processing time: 00:00:00.6032244
2020-05-15 17:09:38.422 Physical dataUp {"rxpk":[{"tmst":4213562732,"chan":4,"rfch":1,"freq":903.100000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":11.0,"rssi":-22,"size":19,"data":"QGta4wIAAgA37J0hoLgz+RBjCg=="}]}
2020-05-15 17:09:38.423 24E1611693965996: device in cache
2020-05-15 17:09:38.423 24E1611693965996: converted 16bit FCnt 2 to 32bit FCnt 2
2020-05-15 17:09:38.423 24E1611693965996: valid frame counter, msg: 2 server: 1
2020-05-15 17:09:38.424 24E1611693965996: decoding with: http://decoder/api/DecodeUrsaLinkUc1114 port: 55
2020-05-15 17:09:38.429 24E1611693965996: sending message {"time":null,"tmms":0,"tmst":4213562732,"freq":903.1,"chan":4,"rfch":1,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","rssi":-22,"lsnr":11.0,"size":19,"data":{"reportType":"UnknownMessageReport","message":"\u0001\u0000\u0000\u0002\u0000\u0000"},"port":55,"fcnt":2,"rawdata":"AQAAAgAA","eui":"24E1611693965996","gatewayid":"evolution-indoor","edgets":1589562578421} to hub
2020-05-15 17:09:38.470 24E1611693965996: message '{"reportType":"UnknownMessageReport","message":"\u0001\u0000\u0000\u0002\u0000\u0000"}' sent to hub
2020-05-15 17:09:38.470 24E1611693965996: checking cloud to device message for 00:00:00.5506551
2020-05-15 17:09:39.022 24E1611693965996: done checking cloud to device message, found no message
2020-05-15 17:09:39.022 24E1611693965996: processing time: 00:00:00.6009541
2020-05-15 17:09:43.506 Physical dataUp {"rxpk":[{"tmst":4218644804,"chan":0,"rfch":0,"freq":902.300000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":12.2,"rssi":-23,"size":19,"data":"QGta4wIAAwA3UttII9RF0E3HrQ=="}]}
2020-05-15 17:09:43.506 24E1611693965996: device in cache
2020-05-15 17:09:43.507 24E1611693965996: converted 16bit FCnt 3 to 32bit FCnt 3
2020-05-15 17:09:43.507 24E1611693965996: valid frame counter, msg: 3 server: 2
2020-05-15 17:09:43.507 24E1611693965996: decoding with: http://decoder/api/DecodeUrsaLinkUc1114 port: 55
2020-05-15 17:09:43.512 24E1611693965996: sending message {"time":null,"tmms":0,"tmst":4218644804,"freq":902.3,"chan":0,"rfch":0,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","rssi":-23,"lsnr":12.2,"size":19,"data":{"reportType":"UnknownMessageReport","message":"\t\u0001\u0000\n\u0001\u0000"},"port":55,"fcnt":3,"rawdata":"CQEACgEA","eui":"24E1611693965996","gatewayid":"evolution-indoor","edgets":1589562583505} to hub
[... cut ...]
# TX rejected (collision beacon): 0.00% (req:82, rej:0)
# TX rejected (too late): 0.00% (req:82, rej:0)
# TX rejected (too early): 0.00% (req:82, rej:0)
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 84270150
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# GPS sync is disabled
##### END #####

JSON up: {"stat":{"time":"2020-05-15 17:12:24 GMT","rxnb":2,"rxok":1,"rxfw":1,"ackr":100.0,"dwnb":0,"txnb":0}}
INFO: [up] PUSH_ACK received in 0 ms
```

ronniesa, 2020-05-15:
Mhhh it is not joining but sending an upstream msg.... there is no join in the above log

gerfen, 2020-05-15:
@ronniesa  Also I removed those keys from the twins data.  There were keys there.

gerfen, 2020-05-15:
Sorry... Let's try that again.

```
2020-05-15 17:30:24.332 Statistic: {"stat":{"time":"2020-05-15 17:30:24 GMT","rxnb":1,"rxok":1,"rxfw":1,"ackr":100.0,"dwnb":0,"txnb":0}}
2020-05-15 17:30:54.335 Statistic: {"stat":{"time":"2020-05-15 17:30:54 GMT","rxnb":0,"rxok":0,"rxfw":0,"ackr":100.0,"dwnb":0,"txnb":0}}
2020-05-15 17:31:17.695 Physical dataUp {"rxpk":[{"tmst":1217869804,"chan":1,"rfch":0,"freq":902.500000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":11.2,"rssi":-21,"size":23,"data":"AExrbmlMQHJVllmWkxZh4SR7/uL7yu0="}]}
2020-05-15 17:31:17.697 24E1611693965996: join request received
2020-05-15 17:31:17.697 24E1611693965996: querying the registry for OTAA device
2020-05-15 17:31:17.762 24E1611693965996: updating twin
2020-05-15 17:31:17.781 24E1611693965996: twin updated
2020-05-15 17:31:17.782 24E1611693965996: device added to cache
2020-05-15 17:31:17.782 24E1611693965996: previous device devAddr (02E35A6B) removed from cache.
2020-05-15 17:31:17.783 UDP: sending message with ID 3B99, to 172.18.0.1:52297
2020-05-15 17:31:17.783 UDP: message sent with ID 3B99
2020-05-15 17:31:17.783 24E1611693965996: processing time: 00:00:00.0885244
2020-05-15 17:31:17.783 24E1611693965996: JoinAccept {"txpk":{"imme":false,"data":"IJj12K2dF7o0RlTgisPQqYA=","tmst":1222869804,"size":17,"freq":923.9,"rfch":0,"modu":"LORA","datr":"SF10BW500","codr":"4/5","powe":14,"ipol":true}}
2020-05-15 17:31:17.784 UDP: packet with id 3B99 successfully transmitted by the aggregator
2020-05-15 17:31:20.029 Physical dataUp {"rxpk":[{"tmst":1220208956,"chan":5,"rfch":1,"freq":903.300000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":9.0,"rssi":-33,"size":18,"data":"QJ+o/QMAl+gE+1ME6e+u4wmB"}]}
2020-05-15 17:31:20.030 03FDA89F: querying the registry for device
2020-05-15 17:31:20.072 03FDA89F: querying the registry for devices by devAddr 03FDA89F found 1 devices
2020-05-15 17:31:20.072 03FDA89F: with devAddr 03FDA89F check MIC failed
2020-05-15 17:31:20.072 03FDA89F: processing time: 00:00:00.0434441
2020-05-15 17:31:24.328 Statistic: {"stat":{"time":"2020-05-15 17:31:24 GMT","rxnb":4,"rxok":2,"rxfw":2,"ackr":100.0,"dwnb":1,"txnb":1}}
2020-05-15 17:31:26.960 Physical dataUp {"rxpk":[{"tmst":1227133436,"chan":6,"rfch":1,"freq":903.500000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":11.2,"rssi":-21,"size":21,"data":"QMsmwgMAAAA3C+FNrvjzXKyU64Cr"}]}
2020-05-15 17:31:26.961 24E1611693965996: device in cache
2020-05-15 17:31:26.961 24E1611693965996: converted 16bit FCnt 0 to 32bit FCnt 0
2020-05-15 17:31:26.962 24E1611693965996: valid frame counter, msg: 0 server: 0
2020-05-15 17:31:26.962 24E1611693965996: decoding with: http://decoder/api/DecodeUrsaLinkUc1114 port: 55
2020-05-15 17:31:26.971 24E1611693965996: sending message {"time":null,"tmms":0,"tmst":1227133436,"freq":903.5,"chan":6,"rfch":1,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","rssi":-21,"lsnr":11.2,"size":21,"data":{"reportType":"UnknownMessageReport","message":"�\ba\u0016��Y�"},"port":55,"fcnt":0,"rawdata":"/whhFpOWWZY=","eui":"24E1611693965996","gatewayid":"evolution-indoor","edgets":1589563886959} to hub
2020-05-15 17:31:26.997 24E1611693965996: message '{"reportType":"UnknownMessageReport","message":"�\ba\u0016��Y�"}' sent to hub
2020-05-15 17:31:26.997 24E1611693965996: checking cloud to device message for 00:00:00.5624368
2020-05-15 17:31:27.560 24E1611693965996: done checking cloud to device message, found no message
2020-05-15 17:31:27.561 24E1611693965996: processing time: 00:00:00.6012521
2020-05-15 17:31:32.040 Physical dataUp {"rxpk":[{"tmst":1232219236,"chan":7,"rfch":1,"freq":903.700000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":12.5,"rssi":-23,"size":21,"data":"QMsmwgMAAQA3invZDrdbqPNos/1L"}]}
2020-05-15 17:31:32.041 24E1611693965996: device in cache
2020-05-15 17:31:32.041 24E1611693965996: converted 16bit FCnt 1 to 32bit FCnt 1
2020-05-15 17:31:32.041 24E1611693965996: valid frame counter, msg: 1 server: 0
2020-05-15 17:31:32.042 24E1611693965996: decoding with: http://decoder/api/DecodeUrsaLinkUc1114 port: 55
2020-05-15 17:31:32.048 24E1611693965996: sending message {"time":null,"tmms":0,"tmst":1232219236,"freq":903.7,"chan":7,"rfch":1,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","rssi":-23,"lsnr":12.5,"size":21,"data":{"reportType":"UnknownMessageReport","message":"�\t\u0002\u0010�\n\u0002\u0015"},"port":55,"fcnt":1,"rawdata":"/wkCEP8KAhU=","eui":"24E1611693965996","gatewayid":"evolution-indoor","edgets":1589563892039} to hub
2020-05-15 17:31:32.073 24E1611693965996: message '{"reportType":"UnknownMessageReport","message":"�\t\u0002\u0010�\n\u0002\u0015"}' sent to hub
2020-05-15 17:31:32.073 24E1611693965996: checking cloud to device message for 00:00:00.5665280
2020-05-15 17:31:32.640 24E1611693965996: done checking cloud to device message, found no message
2020-05-15 17:31:32.641 24E1611693965996: processing time: 00:00:00.6015306
2020-05-15 17:31:37.081 Physical dataUp {"rxpk":[{"tmst":1237258620,"chan":4,"rfch":1,"freq":903.100000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":11.8,"rssi":-23,"size":19,"data":"QMsmwgMAAgA33TXREdYq3iSb0g=="}]}
2020-05-15 17:31:37.081 24E1611693965996: device in cache
2020-05-15 17:31:37.082 24E1611693965996: converted 16bit FCnt 2 to 32bit FCnt 2
2020-05-15 17:31:37.082 24E1611693965996: valid frame counter, msg: 2 server: 1
[... cut ...]
2020-05-15 17:31:42.211 24E1611693965996: message '{"reportType":"UnknownMessageReport","message":"\t\u0001\u0000\n\u0001\u0000"}' sent to hub
2020-05-15 17:31:42.211 24E1611693965996: checking cloud to device message for 00:00:00.5498702
2020-05-15 17:31:42.765 24E1611693965996: done checking cloud to device message, found no message
2020-05-15 17:31:42.765 24E1611693965996: processing time: 00:00:00.6042740
2020-05-15 17:31:54.327 Statistic: {"stat":{"time":"2020-05-15 17:31:54 GMT","rxnb":4,"rxok":4,"rxfw":4,"ackr":100.0,"dwnb":0,"txnb":0}}
```

ronniesa, 2020-05-15:
Ok much better :-) can you check if in the reported twin the keys are populated?

gerfen, 2020-05-15:
Yes they are.  I can share them if they would be helpful.

ronniesa, 2020-05-15:
Join looks good then. Does the device has setting for rx2? If yes can you post it. My guess is some mismatch on dr etc as the msg leaves the gateway but seems to be ignored by the device

gerfen, 2020-05-15:
This is how I have the devices configured:

[image]
[image]

gerfen, 2020-05-18:
@ronniesa  I'm still not able to receive messages with these new Class C devices.  Do you have any other ideas as what the issue might be?

gerfen, 2020-05-18:
@ronniesa I'm not sure if this will be helpful or not but the tech support team from Ursalink sent me this document for sending direct messages:  

[MQTT Remote Control DO.pdf](https://github.com/Azure/iotedge-lorawan-starterkit/files/4646701/MQTT.Remote.Control.DO.pdf)

ronniesa, 2020-05-19:
@gerfen So the device is able to join and send msg upstream so we know that the device can receive msg (join accept) and can send msg like join request and normal call home msg. This means that all the keys etc. are correct.

The RX2 DR and Freq seems also to match between the down msg and the settings you posted. 
Are you sure that the device expect the down msg using port 55?
I would try with a confirmed down msg to see if the device is answering back.

Also does the device have kind of terminal/debug that you can see what is going on?

ronniesa, 2020-05-19:
@gerfen in the doc above you can see that they send msg on port 85 and as confirmed:

For UC1122&UC1152
Set DO1 high
{"confirmed": true,"fport": 85,"data": "CQEA/w=="}
Set DO1 low
{"confirmed": true,"fport": 85,"data": "CQAA/w=="}
For UC1114
Set DO1 high
{"confirmed": true,"fport": 85,"data": "CQEA/w=="}
Set DO1 low
{"confirmed": true,"fport": 85,"data": "CQAA/w=="}
Set DO2 high
{"confirmed": true,"fport": 85,"data": "CgEA/w=="}
Set DO2 low
{"confirmed": true,"fport": 85,"data": "CgAA/w=="}

gerfen, 2020-05-19:
@ronniesa  This morning I was able to get the device to respond to the DO1 high command but I had to take the base64 encoded string and decode it first before sending.  

[image]

The gateway sees this:
```
2020-05-19 17:36:17.767 24E1611693965996: received cloud to device message from direct method: {"devEUI":"24E1611693965996","fport":"85","confirmed":true,"payload":"\t\u0001\u0000�","MessageId":"b596b47f1d254ac89c21b827f0b8e4d3"}
2020-05-19 17:36:17.768 24E1611693965996: [class-c] down frame counter: 23
2020-05-19 17:36:17.768 24E1611693965996: using device rx2: 10, datr: SF10BW500, region: US915
2020-05-19 17:36:17.768 24E1611693965996: cloud to device message: 090100EFBFBD, id: b596b47f1d254ac89c21b827f0b8e4d3, fport: 85, confirmed: True, cidType: Zero
2020-05-19 17:36:17.769 24E1611693965996: sending a downstream message with ID 3F5C
2020-05-19 17:36:17.769 24E1611693965996: ConfirmedDataDown {"txpk":{"imme":true,"data":"oI89SAIAFwBVHAsefjNL1mEiFw==","tmst":0,"size":19,"freq":923.3,"rfch":0,"modu":"LORA","datr":"SF10BW500","codr":"4/5","powe":14,"ipol":true}}
2020-05-19 17:36:17.769 UDP: sending message with ID C9E2, to 172.18.0.1:39550
2020-05-19 17:36:17.770 UDP: message sent with ID C9E2
2020-05-19 17:36:17.773 UDP: packet with id C9E2 successfully transmitted by the aggregator
2020-05-19 17:36:19.358 Physical dataUp {"rxpk":[{"tmst":1115911876,"chan":0,"rfch":0,"freq":902.300000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":9.5,"rssi":-37,"size":16,"data":"QI89SAIgRABVjNjjP0D2EQ=="}]}
2020-05-19 17:36:19.358 24E1611693965996: device in cache
2020-05-19 17:36:19.359 24E1611693965996: converted 16bit FCnt 68 to 32bit FCnt 68
2020-05-19 17:36:19.359 24E1611693965996: valid frame counter, msg: 68 server: 67
2020-05-19 17:36:19.360 24E1611693965996: decoding with: http://decoder/api/DecodeUrsaLinkUc1114 port: 85
2020-05-19 17:36:19.366 24E1611693965996: message ack received for cloud to device message id b596b47f1d254ac89c21b827f0b8e4d3
2020-05-19 17:36:19.366 24E1611693965996: sending message {"time":null,"tmms":0,"tmst":1115911876,"freq":902.3,"chan":0,"rfch":0,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","rssi":-37,"lsnr":9.5,"size":16,"data":{"reportType":"UnknownMessageReport","message":"\t\u0001\u0001"},"port":85,"fcnt":68,"rawdata":"CQEB","eui":"24E1611693965996","gatewayid":"evolution-indoor","edgets":1589909779357} to hub
2020-05-19 17:36:19.391 24E1611693965996: message '{"reportType":"UnknownMessageReport","message":"\t\u0001\u0001"}' sent to hub
2020-05-19 17:36:19.391 24E1611693965996: checking cloud to device message for 00:00:00.5660342
2020-05-19 17:36:19.959 24E1611693965996: done checking cloud to device message, found no message
2020-05-19 17:36:19.959 24E1611693965996: processing time: 00:00:00.6019250
2020-05-19 17:36:27.391 Statistic: {"stat":{"time":"2020-05-19 17:36:27 GMT","rxnb":2,"rxok":2,"rxfw":2,"ackr":100.0,"dwnb":1,"txnb":1}}

```

Using a COM sniffer I see the data from the devices perspective:

```
=====SX1276OnDio0Irq:1616 1
date: 1970-1-4 , 22:19:29
******* LoRaWAN Receive data******
 rssi: -51
 snr : 24
 size: 19
 UpLinkCounter: 61
 DownLinkCounter: 18
 AdrAckCounter: 1
 MType: 5
 data[ a08f3d480200130055190de906e1379a57bb54 ]
************************************
======= McpsIndication start ===========
  Port: 85
  RxData: 1
  size: 6
  data: [ 090100efbfbd]
======= McpsIndication end   ===========
===LoRaWAN receive : 090100efbfbd
===setDigitalOutput, channel = 9
===report config:090101
===send data(3):090101
======= sendframe start ===========
  Port: 85
  Type: 0
  size: 3
  NbTrials: 3
  Confirm:0
  data: [ 090101]
date: 1970-1-4 , 22:19:29
=======RegionUS915TxConfig: start=========
Channel: 5
Frequency: 903300000
phyTxPower: 20
Datarate: 0
bandwidth: 0
phyDr: 10
=======RegionUS915TxConfig: end=========
date: 1970-1-4 , 22:19:29
******* LoRaWAN Send data ********
 size: 16
 data[ 408f3d4802203d0055cbf784ad25164b ]
************************************
=====SX1276OnDio0Irq:1616 2
=====..\..\public\Ursalink\LoRaWAN\radio\sx1276.c SX1276OnTimeoutIrq:1530 1
```
So that's some progress.  What I can't wrap my head around is why the older versions of the devices respond to just sending a command string like "CurtainUp".    I probably need to encode that string differently but I'm not sure how.

ronniesa, 2020-05-19:
Cool, I thought it was not a ntwsrv issue as the join and dr are all good. Honestly I find it strange that the device use strings as it is a waste of bandwidth normally you would use some bits for this like 0 and 1.... you should ask the device manufacturer what did they change.

gerfen, 2020-05-19:
Unfortunately they claim they didn't change anything.  That said I can live with sending the DO commands along with a duration.  If I can do that  I will be good to go.  

Thanks for the assistance.

gerfen, 2020-05-20:
@ronniesa So Ursalink has indeed change their firmware.  I need to send the command as a hex string which starts with FF12 followed by the length of the command string.  For example if I want to send "test" as a command the hex string would look like this:  FF120474657374.  

After getting the hex string I needed to convert it to base64 encode and then use the "rawPayload" property to send the direct method. With all of those changes I can now control the devices as I had before. 

Thank you so much for helping me dig through all of this.  I really appreciate the help!

ronniesa, 2020-05-20:
@gerfen glad that you found the issue, no problem you also helped us a lot in the past especially with the US frequencies.

closed 2020-05-20 by gerfen