Azure/iotedge-lorawan-starterkit #238: Redis cache usage

levalencia, 2020-06-19:
Hello

I just read the documenation and I know a redis cache is needed, however it doesnt mention whats the usage of it in the entire solution, can you please explain me?  What do we store in there?

Thanks in advance?

ronniesa, 2020-06-19:
Hi, 
Redis is used for the following functionality:
- in a mulligatawny scenario to coordinate who will send the downstream join, down msg or ack.
- for message deduplication strategy. We don't send the message to redis but with the deveui, fcnt and gtwid we can detect if the message was already processed by another gateway.
- we also cache some twins data to avoid the throttling of the device registry in very large deployments.

Hope this helps

levalencia, 2020-06-19:
so If I leave the redis connection string empty, it wont hurt?

as I am doing a PoC, I will only have one field gateway

ronniesa, 2020-06-19:
Unfortunately currently it is not possible to run the kit without redis.

levalencia, 2020-07-13:
can you please let me know what this means:

deveui, fcnt and gtwid 

thanks in advance

gerfen, 2020-07-13:
@levalencia  Here is my understanding of those names:

deveui - is an ID in the IEEE EUI64 address space used to identify a device
fcnt - is the frame counter
gtwid - is the gateway ID

Perhaps someone from the project team can confirm when it's convenient?  :)

Mandur, 2020-08-10:
FYI, I investigated the requirements to remove the Redis Cache dependency in case of single gateway and that would take substantial time to properly test the stability of the system. Additionally if we really want to support it we would need to include this configuration in our E2E tests, that would require even more work. For now I will park this and we might revisit the suggestion when we have more time

levalencia, 2020-08-10:
thanks, dont worry not urgent.
I added a 2nd gateway in my PoC for better testing and I will use redis as well.