Azure/iotedge-lorawan-starterkit #244: Aaeon support for US915?

donniekerr, 2020-07-14:
https://www.aaeon.com/en/p/intel-lora-gateway-system-server 
Can you confirm that this device supports US915?  I had a sensor vendor tell me it only supports EU, but your Read.me here says both.  Their specs don't say US915 exactly. 

In my use case, I'll have Lora devices on a farm in Ohio.  I want to get access to their data on an Azure IoT Edge device in the greenhouse.  That is why I'm looking at using this starter kit.  

Just wanted to confirm that this device is still good and will work with your kit.
Thanks!
Donnie

gerfen, 2020-07-14:
@donniekerr  I have built out a lorawan system using the AAEON gateway you mentioned.  Checkout NurserySentry.com

donniekerr, 2020-07-14:
Very cool @gerfen .  But does this device work in North America? says it is less than 900 Mhz band in their specs.

gerfen, 2020-07-14:
@donniekerr Yes.  We're based in Oregon.

donniekerr, 2020-07-14:
Thank you!  It is confusing because two of the recommended ones here only say 868 MHz in the spec, but this one says https://www.aaeon.com/en/p/iot-gateway-systems-aiot-ip6801 
915 is optional.  

Did you have to do a custom order to get US915?  Or did it just work out of the box?  Not clear because when you go to buy it, it doesn't let you configure which OS or which frequency band.

gerfen, 2020-07-14:
I ended up contacting the North American sales group to order the gateways I have in production.  My point of contact at AAEON is Tim Sterling (tims (at) aaeon.com).  If you contact Tim please let him know I referred you.  Make sure you also explicitly order a power supply or specify POE if desired.

After receiving the units I had to install Ubuntu manually and configure the SPI functionality.  I followed the instructions [here](https://wiki.up-community.org/Ubuntu).

If you would like to chat further I can be reached via Skype (michael.gerfen)

gerfen, 2020-07-14:
One other point, if you order POE you will need to order a special RJ45 cable to get an IP68 seal.

donniekerr, 2020-07-14:
Thanks! I will reach out to Tim and tell him you referred me. 

This process of getting Lora to work with Azure IoT seems overly complicated.  I was hoping there was a Lora Edge Module that accessed the Lora device data on the iotedge.  But, no luck!

One cool option I found was the Air Pace https://www.etherwan.com/products/wireless/industrial-smart-iot-edge-computing-gateway .  It has an Azure IoTEdge Runtime UI built in and it even stores all Lora data in a local SQL Server. Unfortunately, it isn't available in US. :(

donniekerr, 2020-07-15:
Tim fixed me up!  Thanks for the referral @gerfen !  I sent him a link to this page and told him about you. 

I will close this issue. I just suggest adding the supported frequency bands next to each Gateway list here, because only one supported US.

closed 2020-07-15 by donniekerr