Azure/iotedge-lorawan-starterkit #245: Windows IoT Enterprise?

donniekerr, 2020-07-18:
Does this starterkit work with an Edge Device running Windows?  It doesn't say or at I don't see OS requirement stated.

Thanks,
Donnie

donniekerr, 2020-07-18:
Found my answer https://techcommunity.microsoft.com/t5/internet-of-things/linux-modules-with-azure-iot-edge-on-windows-10-iot-enterprise/ba-p/1407066  and it is not yet.

What would it take to convert these two modules from linux to windows containers?  is it a huge deal?  I basically have the same requirement this video is trying to solve.  Both windows and linux modules running on IoT Enterprise.

donniekerr, 2020-07-19:
Which OS is required on the Gateway device? https://docs.microsoft.com/en-us/azure/iot-edge/support#operating-systems . Ubuntu 16.04 or 18.08? ARM or AMD?  I need to replace my Windows IoT Enterprise with one that works with this starterkit.  Thanks

gerfen, 2020-07-19:
Is this for an AAEON gateway?  If so the you want the AMD version.  I've been running Ubuntu 16.04 server.  Follow the instructions here: https://wiki.up-community.org/Ubuntu_16.04#Install_Ubuntu_for_UP.2C_UP2.2C_UP_Core_and_UP_Core_Plus

For Ubuntu 18.04 follow the instructions here: https://wiki.up-community.org/Ubuntu_18.04

--Michael

gerfen, 2020-07-19:
You might find this helpful as well: https://github.com/Azure/iotedge-lorawan-starterkit/issues/152

donniekerr, 2020-07-19:
Thanks again @gerfen !  I got this from Tim 
AIOT-IP6801 E3940 w/4G RAM.32G eMMC, -20 ~ 70 °CRS-232.Wifi.BT(801.1ac+BT4.0) Wall-mount + LoRa 915 MHz

Given your issues, I'm not encouraged that installation and setup will be easy. :(

gerfen, 2020-07-19:
I can forward my notes and the SPI files via Skype if you like 

My Skype handle is michael.gerfen.

donniekerr, 2020-07-19:
@gerfen I was testing just the install of iotedge and these modules on a Linux VM so I can learn how to do on the real box when it arrives.  I think I got the same error as you with the concentrator not starting in PktFwdModule fails in module status. But, this may be because I'm not on a real compatible edge gateway for this test.

I'm also not sure what you mean by SPI Files?  I don't even know what they are. In my case, we are not using Arduino devices (yet).  We just want to test connecting to a couple off-the-shelf Lora devices to learn how Lora devices can connect to Azure IoT Hub. 

Anyway, below is the log of the PktFwd module.  If there is an issue that is not related to being on a VM, I can't tell. It looks like a directory is missing. 

azureuser@linuxlora:~$ sudo iotedge logs LoRaWanPktFwdModule
US region detected.
sh: echo: I/O error
Accessing concentrator reset pin through GPIO7...
./reset_lgw.sh: 30: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent
./reset_lgw.sh: 33: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
./reset_lgw.sh: 34: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
./reset_lgw.sh: 37: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent
No SPI Dev version detected in environment variables on x86, defaulting to SPI Dev 2
*** Beacon Packet Forwarder for Lora Gateway ***
Version: 4.0.1
*** Lora concentrator HAL library version info ***
Version: 5.0.1;
***
INFO: Little endian host
INFO: found global configuration file global_conf.json, parsing it
INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
INFO: lorawan_public 1, clksrc 1
INFO: no configuration for LBT
INFO: antenna_gain 0 dBi
INFO: Configuring TX LUT with 16 indexes
INFO: radio 0 enabled (type SX1257), center frequency 902700000, RSSI offset -166.000000, tx enabled 1, tx_notch_freq 0
INFO: radio 1 enabled (type SX1257), center frequency 903400000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
INFO: Lora multi-SF channel 0>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 1>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 2>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 4>  radio 1, IF -300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 5>  radio 1, IF -100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 6>  radio 1, IF 100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 7>  radio 1, IF 300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora std channel> radio 0, IF 300000 Hz, 500000 Hz bw, SF 8
INFO: FSK channel 8 disabled
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000000
INFO: server hostname or IP address is configured to "192.168.0.17"
INFO: upstream port is configured to "1680"
INFO: downstream port is configured to "1680"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: found local configuration file local_conf.json, parsing it
INFO: redefined parameters will overwrite global parameters
INFO: local_conf.json does not contain a JSON object named SX1301_conf
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000101
INFO: server hostname or IP address is configured to "172.17.0.1"
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
ERROR: [main] failed to start the concentrator
US region detected.
sh: echo: I/O error
Accessing concentrator reset pin through GPIO7...
./reset_lgw.sh: 30: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent
./reset_lgw.sh: 33: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
./reset_lgw.sh: 34: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
./reset_lgw.sh: 37: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent
No SPI Dev version detected in environment variables on x86, defaulting to SPI Dev 2
*** Beacon Packet Forwarder for Lora Gateway ***
Version: 4.0.1
[... 389 lines cut ...]
INFO: Lora multi-SF channel 6>  radio 1, IF 100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 7>  radio 1, IF 300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora std channel> radio 0, IF 300000 Hz, 500000 Hz bw, SF 8
INFO: FSK channel 8 disabled
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000000
INFO: server hostname or IP address is configured to "192.168.0.17"
INFO: upstream port is configured to "1680"
INFO: downstream port is configured to "1680"
INFO: downstream keep-alive interval is configured to 10 seconds
[... cut ...]
INFO: local_conf.json does not contain a JSON object named SX1301_conf
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000101
INFO: server hostname or IP address is configured to "172.17.0.1"
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
ERROR: [main] failed to start the concentrator

gerfen, 2020-07-19:
@donniekerr You're not going to be able to get the PktFwd module to start on a VM.  SPI (serial peripheral interface) is used to talk to the SX1301 Digital Baseband Chip which is the heart of the Lorawan gateway.  I've uploaded the SPI files -- you'll eventually need to get the AAEON gateway properly configured -- to this issue.  Below are the curated instructions I use to configure my AAEON gateways - YMMV:

```
	11. Install Ubuntu - follow instructions here:  https://wiki.up-community.org/Ubuntu
	2. Create the bootable USB drive with Rufus
		a. https://tutorials.ubuntu.com/tutorial/tutorial-create-a-usb-stick-on-windows#9
	3. Boot to the USB drive 
	3. Boot to the USB drive 
		a. Press F7 and choose the bootable USB drive from the menu
	4. Choose Install Ubuntu Server
	5. Name the host - <YOUR HOST NAME>
	6. Full name - <YOUR FULL NAME>
	7. Username: <YOUR USER NAME>
	8. PW: <YOUR PASSWORD>
	9. Encrypt home directory (you'll need to decide what you want to do here -- Yes/No)
	10. Force UEFI - yes
	11. Partition - use entire disk and set up LVM
	12. Proxy - blank
	13. No automatic updates (NB:  This is very important!)
	14. Software
		a. Standard system utilities
		b. OpenSSH server
	15. Follow instructions here: https://wiki.up-community.org/Ubuntu
		a. sudo add-apt-repository ppa:ubilinux/up
		b. sudo apt update
		c. sudo apt-get autoremove --purge 'linux-.*generic'
			i. Respond 'Y'
		d. sudo apt-get install linux-image-generic-hwe-16.04-upboard
			i. Respond 'Y'
		e. sudo apt dist-upgrade -y
		f. sudo reboot now
		g. Confirm the kernel has been installed:
			i. uname -a
			ii. See "4.15.0-37-generic #40~upboard03-Ubuntu SMP Wed Dec 12 16:21:24 UTC 2018 x86_64 x86_64 x86_64 GNU/Linux"
	16. Install Ampak firmware for WiFi and Bluetooth
		a. sudo apt install firmware-ampak
		b. sudo reboot now
	17. Enable the HAT functionality
		a. sudo apt install upboard-extras
		b. sudo usermod -a -G gpio ${USER}
		c. sudo usermod -a -G leds ${USER}
		d. sudo usermod -a -G spi ${USER}
		e. sudo usermod -a -G i2c ${USER}
		f. sudo usermod -a -G dialout ${USER}
		g. sudo reboot now
	18. Copy spi-enable to USB drive (attached to this issue)
		a. Mount the drive 
			i. Sudo mkdir /media/usb
			ii. sudo mount /dev/sda1 /media/usb
			iii. cd /media/usb/spi-enable
			iv. sudo ./install_hooks
			v. sudo acpi-add spidev*
			vi. sudo reboot now
			vii. ls /dev
				1) Confirm spidev0.0 and spidev1.1 exist
	19. Follow instructions here: https://docs.microsoft.com/en-us/azure/iot-edge/how-to-install-iot-edge-linux
		a. https://docs.microsoft.com/en-us/azure/iot-edge/quickstart-linux (back up)
		b. curl https://packages.microsoft.com/config/ubuntu/16.04/multiarch/prod.list > ./microsoft-prod.list
		c. sudo cp ./microsoft-prod.list /etc/apt/sources.list.d/
		d. curl https://packages.microsoft.com/keys/microsoft.asc | gpg --dearmor > microsoft.gpg
		sudo cp ./microsoft.gpg /etc/apt/trusted.gpg.d/
		e. sudo apt-get update
		f. sudo apt-get install moby-engine 
			i. answer 'Y' when prompted
		g. sudo apt-get install moby-cli
			i. Answer 'Y' when prompted
		h. sudo apt-get update
		i. sudo apt-get install iotedge
	20. Docker log rotation: https://sandro-keil.de/blog/logrotate-for-docker-container/
		a. Create a file named /etc/logrotate.d/docker-container
			sudo nano /etc/logrotate.d/docker-container
		b. Add the following to the file:
			/var/lib/docker/containers/*/*.log {
			  rotate 7
			  daily
			  compress
			
			  missingok
			  delaycompress
			  copytruncate
			}
		c. Test by executing: logrotate -fv /etc/logrotate.d/docker-container

```
[Spi-enable.zip](https://github.com/Azure/iotedge-lorawan-starterkit/files/4944794/Spi-enable.zip)

donniekerr, 2020-07-19:
Thank you.  

I'm having serious concerns about using this kit.  Instructions are way too complicated and now you have all of these additional instructions to make it work.  I appreciate your input.  But, I'm going to have a pause and re-evaluate the use of Lora with Azure IoT.  There are way too many points of potential failure in this architecture.

gerfen, 2020-07-19:
It’s really not that bad – just a lot of steps.  Once it’s been configured it’s been rock solid for me. 😊

ronniesa, 2020-07-20:
> I'm having serious concerns about using this kit. Instructions are way too complicated and now you have all of these additional instructions to make it work. I appreciate your input. But, I'm going [...]

The goal of this kit is to integrate Lora on top of IoT Hub and IoT Edge and be as lean as possible and use what the platform offers. 
It is a platform and not a ready to use solution, it is made for people that plan to build custom software on top. It is attractive for people that are building on top of IoT Edge and IoT Hub as most of the concept and learnings are the same for any IoT solution and does not require third party servers, licenses and different device registry, queue etc.
There are Lora end to end solution out of the box ready to use but are domain specific. Most commercial Lora solution still needs a lot of configuration and know how and the same applies to this solution. The current status of Lora is not a plug and play like is a WiFi router.
Gateway are fairly complex and requires quite a bit of configuration and if you use our stuff or not the challenge is still there and you cannot easily update the stack like the pktfrw that with IoT Edge is easier.

Bottom-line if you want to build a custom solution that needs Lora on top of IoT Hub/Edge this kit is compelling as there are no other solution with this kind of integration. And the learnings on using IoT Edge, IoT Hub and Azure will be very beneficial for various IoT scenarios
If your needs is more a pure Lora solution and you don’t need the capability of IoT Hub or IoT Edge or you want an out of the box ready to use solution then this may not be the best approach.

Hope this clarify a bit the target of this kit.

ronniesa, 2020-07-20:
> Which OS is required on the Gateway device? https://docs.microsoft.com/en-us/azure/iot-edge/support#operating-systems . Ubuntu 16.04 or 18.08? ARM or AMD? I need to replace my Windows IoT Enterprise with one that works with this starterkit. Thanks

Currently we support only Linux, we support ARM and AMD and the HW needs to be capable of running IoT Edge. We strongly recommend at least 2GB RAM on ARM and 4GB Ram on AMD. We test with 1GB and 2GB but a bit more ram is beneficial with a doker container solution.

donniekerr, 2020-07-20:
Thanks guys.  I apologize for venting my frustrations here. I appreciate the fact this is the only option out there and will try to make it work.  I'm sure I will have more questions.  I will close this since my original Windows questions was answered.  Thanks!

maxstefaniv, 2021-03-15:
> 18. Copy spi-enable to USB drive (attached to this issue)
> 		a. Mount the drive
> 			i. Sudo mkdir /media/usb
> 			ii. sudo mount /dev/sda1 /media/usb
> 			iii. cd /media/usb/spi-enable
> 			iv. sudo [...]

Hello, I Am stuck on trying to make AIOT-ILRA01-A10-CA40232EU work, and I followed the detailed steps provided, yet last point only displays apidev2.0.

donniekerr, 2021-03-15:
Mine worked fine with upgraded 2.0 version number in the file name.

Donnie

maxstefaniv, 2021-03-15:
@donniekerr yeah, later I have seen that it was working anyway. That is the reason I deleted the question. Thank you for answering.

sanistar, 2021-04-13:
@donniekerr  I am now trying to get a new gateway set up and the LoRaWanPktFwdModule is not starting.  Could you clarify which file name you are referring to in this comment? 

_'Mine worked fine with upgraded 2.0 version number in the file name.'_

Thanks!

donniekerr, 2021-04-13:
@sanistar that was referring to @gerfen input on enabling spi 

Copy spi-enable to USB drive (attached to this issue)
		a. Mount the drive 
			i. Sudo mkdir /media/usb
			ii. sudo mount /dev/sda1 /media/usb
			iii. cd /media/usb/spi-enable
			iv. sudo ./install_hooks
			v. sudo acpi-add spidev*
			vi. sudo reboot now
			vii. ls /dev
				1) Confirm spidev0.0 and spidev1.1 exist

Step vii, running ls /dev .  I saw version 2 instead, but it was still fine. 
There is a lot of conversation in various threads about getting the forwarder to start. Can't point you directly there right now but should be easy to find.

gerfen, 2021-04-15:
@donniekerr Did you see spidev2.0?  And did you use GPIO430 and SPI_DEV 1 in the config for your LoRaWanPktFwdModule in the Azure IoT Hub?

donniekerr, 2021-04-15:
@gerfen , honestly, I don't recall and I left that company so I don't have access anymore. I do remember seeing spidev2.0 or 2.1 after completing the steps.  I think there was also spidev0, but not sure.  Sorry.  All I know is that it still worked using these instructions. Maybe by magic. :)

gerfen, 2021-04-15:
Thanks @donniekerr.  Getting that particular AAEON gateway properly configured has always involved a bit of voodoo. :)  I was just curious what you saw.