Azure/iotedge-lorawan-starterkit #249: At least one resource deployment operation failed

levalencia, 2020-08-07:
I tried to use the Deploy to Azure button and everything works fine, except for the Function, I get this error, but the error does not really have any details:

My question is: maybe is an sporadic error and trying it again would fix it? or should I have to deploy it manually?

{"code":"DeploymentFailed","message":"At least one resource deployment operation failed. Please list deployment operations for details. Please see https://aka.ms/DeployOperations for usage details.","details":[{"code":"BadRequest","message":"{\r\n \"Code\": \"BadRequest\",\r\n \"Message\": \"There was a conflict. The remote server returned an error: (400) Bad Request.\",\r\n \"Target\": null,\r\n \"Details\": [\r\n {\r\n \"Message\": \"There was a conflict. The remote server returned an error: (400) Bad Request.\"\r\n },\r\n {\r\n \"Code\": \"BadRequest\"\r\n },\r\n {\r\n \"ErrorEntity\": {\r\n \"ExtendedCode\": \"01020\",\r\n \"MessageTemplate\": \"There was a conflict. {0}\",\r\n \"Parameters\": [\r\n \"The remote server returned an error: (400) Bad Request.\"\r\n ],\r\n \"Code\": \"BadRequest\",\r\n \"Message\": \"There was a conflict. The remote server returned an error: (400) Bad Request.\"\r\n }\r\n }\r\n ],\r\n \"Innererror\": null\r\n}"}]}

thank you

Mandur, 2020-08-10:
Hello,

Thank you for reporting the issue. Would you per case have a screenshot of the deployment blade windows? 
Could I also ask what was the name you selected for your lora deployment? A supposition is that the template tried to deploy a resource with a name that already exists on Azure.
When you say it is sporadic, could I ask how often does it occurs?

Thank you ! ,Best regards,

Mikhail

donniekerr, 2020-08-11:
Yes, I experienced this as well. Took me 4 or 5 tries with different names to get past it.  Then it all was successful with a rare name like gh1ed1.   Think it failed with a dash because one of the services didn't allow a dash in the name.

Mandur, 2020-08-12:
I see, this is definitelly an annoying issue. I found no way around it when I last investigated and I reported it to the ARM team. I saw there are new updates, maybe I could check if something was done to add a kind of validation before the template is ran.