Azure/iotedge-lorawan-starterkit #253: Failed to deploy path that does not exist

levalencia, 2020-08-13:
I got the master branch and I am trying to deploy the Azure Facade Function.
I changed the remote function setting from 3 to 2 as in the previous Issue I just marked as closed.

However I get this error:
Failed to deploy path that does not exist: c:\Users\username\Downloads\Repos\iotedge-lorawan-starterkit\LoRaEngine\LoraKeysManagerFacade\bin\Release\netstandard2.0\publish

Build is working fine:

```

Build succeeded.
    0 Warning(s)
    0 Error(s)

Time Elapsed 00:00:01.24

Terminal will be reused by tasks, press any key to close it.

> Executing task: dotnet publish --configuration Release <

Microsoft (R) Build Engine version 16.6.0+5ff7b0c9e for .NET Core
Copyright (C) Microsoft Corporation. All rights reserved.

  Determining projects to restore...
  All projects are up-to-date for restore.
  Logger -> C:\Users\username\Downloads\Repos\iotedge-lorawan-starterkit\LoRaEngine\modules\LoRaWanNetworkSrvModule\Logger\bin\Release\netcoreapp3.1\Logger.dll
  LoRaTools -> C:\Users\username\Downloads\Repos\iotedge-lorawan-starterkit\LoRaEngine\modules\LoRaWanNetworkSrvModule\LoraTools\bin\Release\netcoreapp3.1\LoRaTools.dll
  LoraKeysManagerFacade -> C:\Users\username\Downloads\Repos\iotedge-lorawan-starterkit\LoRaEngine\LoraKeysManagerFacade\bin\Release\netcoreapp3.1\bin\LoraKeysManagerFacade.dll
  LoraKeysManagerFacade -> C:\Users\username\Downloads\Repos\iotedge-lorawan-starterkit\LoRaEngine\LoraKeysManagerFacade\bin\Release\netcoreapp3.1\publish\
```

I am not sure what I am doing wrong from my side

Thanks

Mandur, 2020-08-13:
From which branch are you trying to deploy? Master or dev? It looks to me like you are somewhat using .net 3.1 locally?

levalencia, 2020-08-13:
master branch, when I checked the bin folder in VS Code, there is not netstandard2.0 in the folder structure.

Mandur, 2020-08-13:
Okay, I think you might not have .NET 2 installed. If you let me 15 minutes I was working on the 1.0.4 release that should simplify this for you.

Mandur, 2020-08-13:
Okay, Revert the setting from your Azure function (should be ~3 not ~2) and pull the latest master. That should solve your issue

levalencia, 2020-08-13:
are you sure you made a change on the master? I dont see any recent changes, latest was 3hours ago:
https://github.com/Azure/iotedge-lorawan-starterkit/commits/master

levalencia, 2020-08-13:
I still have this warning after pulling 
The remote version "~3" for function app "be-lva-iotpoc-fnFacade" does not match your local version "~2".

Mandur, 2020-08-13:
the commit is aged 3 hours ago from the dev branch, I just pushed it now on master. So I am pretty sure it's new on master. Did you try to run git pull ?

Mandur, 2020-08-13:
I tried to reproduce on my side but I am unable to. Would you agree to have a live session so you could show me? my Skype handle is mandurlevrai

Mandur, 2020-08-17:
We updated the how-to to better describe the new deployment experience for visual studio and visual studio code. Thank you @levalencia for raising the issue! If there are any additional concern feel free to reopen or add other issues.

closed 2020-08-17 by Mandur