Azure/iotedge-lorawan-starterkit #255: Expose metrics in LoRa Network Server module

fbeltrao, 2020-08-14:
**Is your feature request related to a problem? Please describe.**
With IoT Edge release 1.0.10 modules can [expose metrics](https://labs.iotedge.dev/codelabs/monitor-iotedge/#0) that can be used to improve system observability. Having the LoRa Network Server Module exposing metrics would enable a better visibility of the module deployment.

**Describe the solution you'd like**
Add support in LoRa Network Server module to expose metrics. This feature, at least in the beginning, would need to be explicitly enabled.

Example metrics:
- Messages / sec
- Message processing time
- Connected devices
- # authentication fails
- # authentication success
- # Delivered C2D messages

**Describe alternatives you've considered**
N/a

**Additional context**
N/a

Mandur, 2020-08-17:
Very cool idea!

bastbu, 2021-11-08:
With .NET 6 [including the Metrics API](https://docs.microsoft.com/en-us/dotnet/api/system.diagnostics.metrics?view=net-6.0), maybe we get the abstraction for metrics for free if we move to .NET 6. We should definitely pick up this work item, now that we have #737 almost done. Independently of the .NET 6 release/Metrics API, we should consider this for the sprint planning in two days @Mandur.

bastbu, 2021-11-11:
As the story had a lot of scope we track the custom metrics in #747 and the metrics collector/Azure Monitor integration in #739.

closed 2021-11-12 by bastbu