Azure/iotedge-lorawan-starterkit #256: Decoding advice:  Sanitizing errant characters in payloads

gerfen, 2020-08-17:
I have a couple of devices which send "%2b" instead of "+" as part of the payload.  For example: 

````
fvQMAHZ3AABLAAB%2b
````

The devices do this randomly which is very frustrating.  I'm looking for advice for the best way to sanitize the payload data before trying to decode it.  In the code below[1], I've added some brute force code to sanitize the string before attempting to decode it.

I'm wondering if there is a better way or a better place to handle this situation?  Whatever ends up being the proper fix, is this something that can be rolled into the decoding sample?

Thanks!

[1]  Updated DecodedController
````
    [Route("api")]
    [ApiController]
    public class DecoderController : ControllerBase
    {
        // GET: api/TestDecoder
        [HttpGet("{decoder}", Name = "Get")]
        [ProducesResponseType(200)]
        [ProducesResponseType(400)]
        public IActionResult Get(string devEUI, string decoder, string fport, string payload)
        {
            var sanitizedPayload = payload.Replace("%2B", "+").Replace("%2b", "+");
            // Validate that fport and payload URL parameters are present.
            Validator.ValidateParameters(fport, sanitizedPayload);

            Type decoderType = typeof(LoraDecoders);
            MethodInfo toInvoke = decoderType.GetMethod(decoder, BindingFlags.Static | BindingFlags.NonPublic);

            if (toInvoke != null)
            {
                var decoderResult = (string)toInvoke.Invoke(null, new object[] { devEUI, Convert.FromBase64String(sanitizedPayload), Convert.ToByte(fport) });
                return this.Ok(decoderResult);
            }
            else
            {
                throw new WebException($"Decoder {decoder} not found.");
            }
        }
    }
````

Mandur, 2020-08-18:
At the current time, I think that is what I would have done. I might have used Regex to avoid string sensitivity but otherwise LGTM.

Have you seen this kind of behavior occurs between different range of devices and how frequently? In case it does we might evaluate a solution to be more flexible here

gerfen, 2020-08-18:
@Mandur Thanks for the response.

I only have one type of device that has this kind of behavior.  The behavior occurs about one in ten times the device sends an upstream message.

Mandur, 2020-08-19:
I would suggest the solution you described before for now. If you see this kind of need arise between multiple device range, let us know and we might investigate further.

gerfen, 2020-08-19:
Sounds good.  Thanks for the advice!

closed 2020-08-19 by gerfen