Azure/iotedge-lorawan-starterkit #257: MIC Failed

gerfen, 2020-08-20:
## Expected Behavior

Devices should reconnect after a gateway reboot.

## Current Behavior

I have many devices which fail to reconnect after a gateway reboot.  If I manually reboot the devices they connect.  Note: I have tried clearing the Redis cache and then rebooting the gateway but the devices still refuse to connect without restarting them.  I should also note that I have been rebooting the gateway once a week for over a year and only recently has this issue become a problem for one class of devices.

See the logs below for details.

## Steps to Reproduce

1. Reboot the gateway
2. Check the logs for the LoRaWanNetworkSrvModule

## Context (Environment)
I'm running on the 1.0.3 release.
### Device (Host) Operating System

Ubuntu 16.04
### Architecture

AMD64
### Container Operating System

Linux

## Logs

````
2020-08-19 18:54:08.777 02D9CD05: querying the registry for device
2020-08-19 18:54:08.854 02D9CD05: querying the registry for devices by devAddr 02D9CD05 found 1 devices
2020-08-19 18:54:08.854 02D9CD05: with devAddr 02D9CD05 check MIC failed
2020-08-19 18:54:08.854 02D9CD05: processing time: 00:00:00.0775013
2020-08-19 18:54:11.185 Physical dataUp {"rxpk":[{"tmst":410477852,"chan":5,"rfch":1,"freq":903.300000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":10.0,"rssi":-58,"size":24,"data":"QIHP0gKA9xoGhhdlbljDC6PmHuDWkbHc"}]}
2020-08-19 18:54:11.186 02D2CF81: querying the registry for device
2020-08-19 18:54:11.244 02D2CF81: querying the registry for devices by devAddr 02D2CF81 found 1 devices
2020-08-19 18:54:11.244 02D2CF81: with devAddr 02D2CF81 check MIC failed
2020-08-19 18:54:11.244 02D2CF81: processing time: 00:00:00.0597021
2020-08-19 18:54:11.283 Physical dataUp {"rxpk":[{"tmst":410573204,"chan":7,"rfch":1,"freq":903.700000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":9.2,"rssi":-71,"size":24,"data":"QCMQFAKA9hoGOqdj1KQPF4uS8mcsgRf2"}]}
2020-08-19 18:54:11.284 02141023: querying the registry for device
2020-08-19 18:54:11.335 02141023: querying the registry for devices by devAddr 02141023 found 1 devices
2020-08-19 18:54:11.335 02141023: with devAddr 02141023 check MIC failed
2020-08-19 18:54:11.335 02141023: processing time: 00:00:00.0527491
2020-08-19 18:54:12.729 Physical dataUp {"rxpk":[{"tmst":412023372,"chan":1,"rfch":0,"freq":902.500000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":11.2,"rssi":-88,"size":24,"data":"QIb64QOA1SMGQurK9rgcH0uBOLtRhOQV"}]}
````

## Additional Information

gerfen, 2020-08-20:
I have an update... We restarted all 100 devices after clearing the Redis cache and restarting the gateway.  Unfortunately only 44 of the devices were able to reconnect.  I'm at a loss for why this is happening and what to do to get the devices to reconnect.  FWIW, all of the devices that are having issues reconnecting are the same type of device from one manufacturer.

Mandur, 2020-08-24:
Hello @Gerfen,

Quick question, are your devices using OTAA or ABP?

gerfen, 2020-08-24:
OTAA

ronniesa, 2020-08-24:
The twin of one of the non working device would be great. 

Also the log while they are joining could give us some useful info.

gerfen, 2020-08-24:
I'm working on getting the logs form the gateway.  Unfortunately I don't have remote access to the gateway and it is over 130 miles from my current location so I'll need to get support from my client's IT department to the log file.

In the meantime here is the redacted twin:

````
{
  "deviceId": "00137A1000004254",
  "etag": "AAAAAAAAAAI=",
  "deviceEtag": "NDE5MDI5OTk4",
  "status": "enabled",
  "statusUpdateTime": "0001-01-01T00:00:00Z",
  "connectionState": "Disconnected",
  "lastActivityTime": "2020-08-19T19:59:03.2520575Z",
  "cloudToDeviceMessageCount": 0,
  "authenticationType": "sas",
  "x509Thumbprint": {
    "primaryThumbprint": null,
    "secondaryThumbprint": null
  },
  "version": 6093,
  "properties": {
    "desired": {
      "AppEUI": "xxxxxxxxxxxxxxxx",
      "AppKey": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
      "SensorDecoder": "http://decoder/api/DecoderNetvoxTemperatureHumidity",
      "GatewayID": "xxxx-xx-xxxxxxx-xxxxx",
      "$metadata": {
        "$lastUpdated": "2019-12-02T20:56:02.850897Z",
        "$lastUpdatedVersion": 2,
        "AppEUI": {
          "$lastUpdated": "2019-12-02T20:56:02.850897Z",
          "$lastUpdatedVersion": 2
        },
        "AppKey": {
          "$lastUpdated": "2019-12-02T20:56:02.850897Z",
          "$lastUpdatedVersion": 2
        },
        "SensorDecoder": {
          "$lastUpdated": "2019-12-02T20:56:02.850897Z",
          "$lastUpdatedVersion": 2
        },
        "GatewayID": {
          "$lastUpdated": "2019-12-02T20:56:02.850897Z",
          "$lastUpdatedVersion": 2
        }
      },
      "$version": 2
    },
    "reported": {
      "AppSKey": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
      "NwkSKey": "xxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxxx",
      "DevAddr": "033A7CDD",
      "FCntDown": 0,
      "FCntUp": 0,
      "DevEUI": "00137A1000004254",
      "NetId": "010000",
      "DevNonce": "3B6A",
      "GatewayID": "xxxx-xx-xxxxxxx-xxxxx",
      "$metadata": {
        "$lastUpdated": "2020-08-19T19:58:59.626528Z",
        "AppSKey": {
          "$lastUpdated": "2020-08-19T19:58:59.626528Z"
        },
        "NwkSKey": {
          "$lastUpdated": "2020-08-19T19:58:59.626528Z"
        },
        "DevAddr": {
          "$lastUpdated": "2020-08-19T19:58:59.626528Z"
        },
        "FCntDown": {
          "$lastUpdated": "2020-08-19T19:58:59.626528Z"
        },
        "FCntUp": {
          "$lastUpdated": "2020-08-19T19:58:59.626528Z"
        },
        "DevEUI": {
          "$lastUpdated": "2020-08-19T19:58:59.626528Z"
        },
        "NetId": {
          "$lastUpdated": "2020-08-19T19:58:59.626528Z"
        },
        "DevNonce": {
          "$lastUpdated": "2020-08-19T19:58:59.626528Z"
        },
        "GatewayID": {
          "$lastUpdated": "2019-12-03T21:26:16.414395Z"
        }
      },
      "$version": 6091
    }
  },
  "capabilities": {
    "iotEdge": false
  },
  "deviceScope": "ms-azure-iot-edge://xxxx-xx-xxxxxxx-xxxxx-xxxxxxxxxxx"
}
````

Mandur, 2020-08-25:
You can set the environment variable LOG_TO_HUB to true on your edge gateway, and the device log will flow as the gateway's device messages in IoT Hub. 
I fail to understand what is going wrong for now. Would it be possible to have a remote session one of these days to troubleshoot together?

ronniesa, 2020-08-25:
Just to be sure are you restarting the lora devices all at the same time or one by one?

gerfen, 2020-08-25:
@ronniesa The devices have to be restarted one by one.  There's one per greenhouse - which are quite large and the devices are installed such that the person restarting the devices needs to carry a ladder into the greenhouse, set it up, climb it and then hold the power button on the device to 20 seconds to get it to shut down and then push the button again to get it to boot up - all of which takes a finite amount of time.

@Mandur I know I can turn on logging but I've experienced the LoRaWanNetworkSrvModule not restarting when I've changed the variable.  Unfortunately I do not have remote access to the gateway due to IT policies and there is not dedicated IT staff at the location so I'm hesitant to potentially make things worse.  

@ronniesa  and @mandur  The folks who work at the nursery are not native English speakers so there was a miscommunication WRT to rebooting the devices.  It appears that many of the devices were not properly restarted.  The staff took a pass through the nursery yesterday and managed to get 77 (up from 44 from late last week) of the devices to reconnect.  There are 47 more which have not been reconnected.  My hope is that the reboot sequence was improperly completed.  I have identified these devices and have ask that they be rebooted. 

What remains to be answered is why the devices were not able to reconnect with out being rebooted.  I am trying to gain access to the complete logs so we can see the join requests.  I have a support request in to get some IT support to get the logs.  I hope to get those soon.

gerfen, 2020-08-27:
I was able to get access to the logs.  I'm seeing this [1] which is indicating that the OTAA AppKey is not correct which is strange as the keys have not been touched since we moved the gateway and devices to a new Azure account about a year ago.  

I've uploaded the complete log file for your review.

[8ab60dd8b6ca9eb91c506f48cd849a6c68514bd261617424875830858b33d04b-json.log.zip](https://github.com/Azure/iotedge-lorawan-starterkit/files/5138930/8ab60dd8b6ca9eb91c506f48cd849a6c68514bd261617424875830858b33d04b-json.log.zip)

Please let me know if I can provide any further information.

[1] Log file excerpt

````
{"log":"2020-08-26 16:05:27.568 Physical dataUp {\"rxpk\":[{\"tmst\":3803189032,\"chan\":8,\"rfch\":0,\"freq\":903.000000,\"stat\":1,\"modu\":\"LORA\",\"datr\":\"SF8BW500\",\"codr\":\"4/5\",\"lsnr\":-2.2,\"rssi\":-108,\"size\":23,\"data\":\"AAAAAAAQehMATUIAABB6EwBnSQv85+I=\"}]}\n","stream":"stdout","time":"2020-08-26T16:05:27.569255553Z"}
{"log":"2020-08-26 16:05:27.569 00137A100000424D: join request received\n","stream":"stdout","time":"2020-08-26T16:05:27.569606124Z"}
{"log":"2020-08-26 16:05:27.569 00137A100000424D: querying the registry for OTAA device\n","stream":"stdout","time":"2020-08-26T16:05:27.569655855Z"}
{"log":"2020-08-26 16:05:27.713 00137A100000424D: using edgeHub local queue\n","stream":"stdout","time":"2020-08-26T16:05:27.713593512Z"}
{"log":"2020-08-26 16:05:27.713 00137A100000424D: getting device twin\n","stream":"stdout","time":"2020-08-26T16:05:27.714016329Z"}
{"log":"2020-08-26 16:05:27.816 00137A100000424D: done getting device twin\n","stream":"stdout","time":"2020-08-26T16:05:27.817016858Z"}
{"log":"2020-08-26 16:05:27.816 00137A100000424D: failed to initialize device from twin: Missing AppKey for OTAA device\n","stream":"stdout","time":"2020-08-26T16:05:27.817199463Z"}
{"log":"2020-08-26 16:05:27.817 00137A100000424D: processing time: 00:00:00.2493792\n","stream":"stdout","time":"2020-08-26T16:05:27.817812869Z"}
{"log":"2020-08-26 16:05:27.817 00137A100000424D: join refused: error initializing OTAA device. Missing AppKey for OTAA device\n","stream":"stderr","time":"2020-08-26T16:05:27.817814946Z"}
{"log":"2020-08-26 16:05:33.924 Physical dataUp {\"rxpk\":[{\"tmst\":3809543772,\"chan\":4,\"rfch\":1,\"freq\":903.100000,\"stat\":1,\"modu\":\"LORA\",\"datr\":\"SF10BW125\",\"codr\":\"4/5\",\"lsnr\":2.5,\"rssi\":-107,\"size\":24,\"data\":\"QFJlUwOADwIGvUG489i3nh6ptjEvMqjO\"}]}\n","stream":"stdout","time":"2020-08-26T16:05:33.924785709Z"}
{"log":"2020-08-26 16:05:33.924 00137A1000004291: device in cache\n","stream":"stdout","time":"2020-08-26T16:05:33.924990198Z"}
{"log":"2020-08-26 16:05:33.925 00137A1000004291: converted 16bit FCnt 527 to 32bit FCnt 527\n","stream":"stdout","time":"2020-08-26T16:05:33.925409786Z"}
{"log":"2020-08-26 16:05:33.925 00137A1000004291: valid frame counter, msg: 527 server: 526\n","stream":"stdout","time":"2020-08-26T16:05:33.925456489Z"}
````

Mandur, 2020-08-28:
Thank you Gerfen. 

Could you please check the desired properties of the device 00137A100000424D ? Based on the log and code, I am suspecting that the desired property AppKey might be missing.

We are investigating the MIC issue, that indeed look to be happening quite consistently...

gerfen, 2020-08-31:
@Mandur Do you need any other info from me other than the info I sent to you via Skype late last week?

Mandur, 2020-09-01:
No I think I have most of what I need, I am struggling to find the exact origin of the bug, but I am investigating

gerfen, 2020-09-16:
@Mandur - I've been able to recreate this issue in my lab - with a different type of sensor - this time a Strega smart valve.  If I reboot my gateway, the device refuses to connect as shown below:

````
2020-09-14 21:58:24.965 Physical dataUp {"rxpk":[{"tmst":4091009796,"chan":4,"rfch":1,"freq":903.100000,"stat":1,"modu":"LORA","datr":"SF10BW125","codr":"4/5","lsnr":9.8,"rssi":-48,"size":18,"data":"QJ+o/QMAvpgEdgKdj6hirsDN"}]}
2020-09-14 21:58:24.968 03FDA89F: querying the registry for device
2020-09-14 21:58:25.095 03FDA89F: querying the registry for devices by devAddr 03FDA89F found 1 devices
2020-09-14 21:58:25.095 03FDA89F: with devAddr 03FDA89F check MIC failed
2020-09-14 21:58:25.095 03FDA89F: processing time: 00:00:00.1309645
````

I have not manually reset the device as I wanted let you know that we have an opportunity to debug in real-time when and if that is convenient for you.

gerfen, 2020-09-23:
@Mandur - any progress on this issue?  Thanks!

Mandur, 2020-09-28:
@gerfen  Sorry for the delay, I was on vacations :) 
Let's have a live session to try to see what is happening there. I should be fairly available this week expect for wednesday and thursday.

maxstefaniv, 2021-05-03:
Hello @gerfen I think I might have experienced the same issue with the same kind of device. How have you solved your issue?
I sent a message to device (Netvox R311W) with new password (32 randomly generated chars which are numbers and letters in hexadecimal). After rebooting it will not connect only getting the error:
`
2021-05-03 10:49:31.578 00137A100000B453: join refused: invalid MIC
2021-05-03 10:49:38.547 00137A100000B453: join request received
`

I have tried going back to previous password which was 32 ones "11111111111111111111111111111111" but still nothing...

How can I see what AppKey is using the device when trying to connect to gateway? @Mandur maybe you could help with this one.

Thank you in advance.

Mandur, 2021-05-03:
Could I ask what is the framecounter value you have on the device? Is the device up for a long time?

maxstefaniv, 2021-05-03:
@Mandur yeah it is up for a while trying to connect and always I get the same message. I can only suspect that it had some issue while changing AppKey  so it would be interesting to see with what AppKey he is trying to connect.

Mandur, 2021-05-05:
No Problem @maxstefaniv , is the issue solved now?

Mandur, 2021-09-06:
As there were no additional discussion I will close the issue, please feel free to reopen if you still have the issue.

closed 2021-09-06 by Mandur