Azure/iotedge-lorawan-starterkit #258: LoraKeysManagerFacade fails when run locally with VS2019

Feiko, 2020-08-25:
## Expected Behavior
Debugging LoraKeysManagerFacade locally using Azure Functions

## Current Behavior
Launching LoraKeysManagerFacade with Visual Studio or Visual Studio Code outputs the following error:

[25-8-2020 08:01:56] A host error has occurred during startup operation '2b8f3a21-cd48-4742-9ffa-6dbebe4972d4'.
[25-8-2020 08:01:56] func: Invalid host services. Microsoft.Azure.WebJobs.Script.WebHost: The following service registrations did not match the expected services:
[25-8-2020 08:01:56]   [Invalid] ServiceType: Microsoft.Extensions.Hosting.IHostedService, Lifetime: Singleton, ImplementationType: LoraKeysManagerFacade.DevAddressCacheManager.
Value cannot be null. (Parameter 'provider')

## Steps to Reproduce

1. Clone the repository 
2. Open LoraEngine Solution
3. Add local.settings.json to LoraKeysManagerFacade project
4. added the connectionstrings to IoTHub Redis and WebjobsStorage
5. Run LoRaKeysManagerFacade

## Context (Environment)
C:\Users\xxxxxx\AppData\Local\AzureFunctionsTools\Releases\3.10.0\cli_x64>func --version
3.0.2798

### Device (Host) Operating System
Windows 10 x64

### Architecture
x64

### Container Operating System

none

### LoRaWAN Module Version

#### Docker

none

## Logs
Azure Functions Core Tools (3.0.2798 Commit hash: d9d99872a51c04b1c2783198b1ee1104da6b064f)
Function Runtime Version: 3.0.14191.0
[25-8-2020 08:14:56] Building host: startup suppressed: 'False', configuration suppressed: 'False', startup operation id: '5ca9f522-d9df-4281-b059-e62ca5fef475'
[25-8-2020 08:14:56] Reading host configuration file 'C:\Users\gorte506\Desktop\iotedge-lorawan-starterkit-1.0.4\LoRaEngine\LoraKeysManagerFacade\bin\Debug\netcoreapp3.1\host.json'
[25-8-2020 08:14:56] Host configuration file read:
[25-8-2020 08:14:56] {
[25-8-2020 08:14:56]   "version": "2.0"
[25-8-2020 08:14:56] }
[25-8-2020 08:14:56] Loading functions metadata
[25-8-2020 08:14:56] Reading functions metadata
[25-8-2020 08:14:56] 6 functions found
[25-8-2020 08:14:56] 6 functions loaded
[25-8-2020 08:14:56] Loading extensions from C:\Users\xxxxxx\Desktop\iotedge-lorawan-starterkit-1.0.4\LoRaEngine\LoraKeysManagerFacade\bin\Debug\netcoreapp3.1\bin. BundleConfigured: False, PrecompiledFunctionApp: False, LegacyBundle: False
[25-8-2020 08:14:56] Loading startup extension 'FacadeStartup'
[25-8-2020 08:14:56] Loaded extension 'FacadeStartup' (1.0.0.0)
[25-8-2020 08:14:56] Reading host configuration file 'C:\Users\xxxxxx\Desktop\iotedge-lorawan-starterkit-1.0.4\LoRaEngine\LoraKeysManagerFacade\bin\Debug\netcoreapp3.1\host.json'
[25-8-2020 08:14:56] Host configuration file read:
[25-8-2020 08:14:56] {
[25-8-2020 08:14:56]   "version": "2.0"
[25-8-2020 08:14:56] }
[25-8-2020 08:14:57] A host error has occurred during startup operation '5ca9f522-d9df-4281-b059-e62ca5fef475'.
[25-8-2020 08:14:57] func: Invalid host services. Microsoft.Azure.WebJobs.Script.WebHost: The following service registrations did not match the expected services:
[25-8-2020 08:14:57]   [Invalid] ServiceType: Microsoft.Extensions.Hosting.IHostedService, Lifetime: Singleton, ImplementationType: LoraKeysManagerFacade.DevAddressCacheManager.
Value cannot be null. (Parameter 'provider')
Press any to continue....[25-8-2020 08:14:58] Building host: startup suppressed: 'False', configuration suppressed: 'False', startup operation id: '95bc08b6-6846-473e-ab2d-4936ed844f9b'
[25-8-2020 08:14:58] Reading host configuration file 'C:\Users\xxxxxx\Desktop\iotedge-lorawan-starterkit-1.0.4\LoRaEngine\LoraKeysManagerFacade\bin\Debug\netcoreapp3.1\host.json'
[25-8-2020 08:14:58] Host configuration file read:
[25-8-2020 08:14:58] {
[25-8-2020 08:14:58]   "version": "2.0"
[25-8-2020 08:14:58] }
[25-8-2020 08:14:58] Loading functions metadata
[25-8-2020 08:14:58] Reading functions metadataHosting environment: Production
Content root path: C:\Users\xxxxxx\Desktop\iotedge-lorawan-starterkit-1.0.4\LoRaEngine\LoraKeysManagerFacade\bin\Debug\netcoreapp3.1

Now listening on: http://0.0.0.0:7071
Application started. Press Ctrl+C to shut down.
[25-8-2020 08:14:59] 6 functions found
[25-8-2020 08:14:59] 6 functions loaded
[25-8-2020 08:14:59] Loading extensions from C:\Users\xxxxxx\Desktop\iotedge-lorawan-starterkit-1.0.4\LoRaEngine\LoraKeysManagerFacade\bin\Debug\netcoreapp3.1\bin. BundleConfigured: False, PrecompiledFunctionApp: False, LegacyBundle: False
[25-8-2020 08:14:59] Loading startup extension 'FacadeStartup'
[25-8-2020 08:14:59] Loaded extension 'FacadeStartup' (1.0.0.0)
[25-8-2020 08:14:59] Reading host configuration file 'C:\Users\xxxxxx\Desktop\iotedge-lorawan-starterkit-1.0.4\LoRaEngine\LoraKeysManagerFacade\bin\Debug\netcoreapp3.1\host.json'
[25-8-2020 08:14:59] Host configuration file read:
[25-8-2020 08:14:59] {
[25-8-2020 08:14:59]   "version": "2.0"
[25-8-2020 08:14:59] }
[25-8-2020 08:14:59] A host error has occurred during startup operation '95bc08b6-6846-473e-ab2d-4936ed844f9b'.
[25-8-2020 08:14:59] func: Invalid host services. Microsoft.Azure.WebJobs.Script.WebHost: The following service registrations did not match the expected services:
[25-8-2020 08:14:59]   [Invalid] ServiceType: Microsoft.Extensions.Hosting.IHostedService, Lifetime: Singleton, ImplementationType: LoraKeysManagerFacade.DevAddressCacheManager.

## Additional Information
If I comment out 'builder.Services.AddSingleton<IHostedService, DevAddressCacheManager>();' in FacadeStartup.cs (line 52) the azure functions starts without errors. 

Perhaps something to do with the update to v3 and this issue? https://github.com/Azure/azure-functions-host/issues/5447 ??

Mandur, 2020-08-27:
Thanks for raising this. I could replicate it locally, and I am still astonished that the deployment is actually working in the current setup. I will prioritize a fix with the team, and keep you updated on the steps

Feiko, 2020-08-27:
Thanks @Mandur . I thought perhaps the issue had something to do with the update to Functions v3 in the latest release. So I tried to run the source from v1.0.3  and got the same exceptions. Not sure what to make of this. Was the problem already present in earlier versions? This would be unlikely right?
So something to do with the local development environment???

Mandur, 2020-08-27:
To be honest I thought as well, but I think it has been transient before. Today I saw it both work and not work on the same function app. Anyway the code is conceptually not appropriate for function, I migrated it to use a cron azure function, it seems to work but I want to be careful with the implication and discuss it with my colleagues.
You can try it out in the branch [mik-dev-fix-function-di](https://github.com/Azure/iotedge-lorawan-starterkit/tree/mik-dev-fix-function-di). if you want.

Thank you for spotting this

Feiko, 2020-08-28:
I'm able to start the LoraKeysManagerFacade using the mik-dev-fix-function-di. I do get a couple of ArgumentNullExceptions from System.IO.FileSystem.dll at startup and a Exception thrown: 'Microsoft.Azure.Storage.StorageException' in System.Private.CoreLib.dll ("The specified blob does not exist.").

I don't know if these messages are expected behavior. Is the missing blob due to a config/setup step that I missed?

Mandur, 2020-08-28:
When breaking on every CLR exception, I can indeed see the FileSystemException, but I don't think the are problematic. The Storage exception is more intriguing, did the blob storage you inputed as the AzureWebJobsStorage exist? Does this error make the debugging break?

Feiko, 2020-08-29:
The error did not break the debugging. Also I just ran it again without the StorageException. Not sure what happend the first time, every seems to work now.

Feiko, 2020-09-01:
I'm still having some trouble getting the solution to run locally. I've hooked up a LoRa gateway over UDP and connected a LoRa device. I created the LoRa device in the IoT Hub. 

When the device tries to join, I get following error. 

2020-09-01 19:23:06.346 Physical dataUp {"rxpk": [{"data": "AP07APB+1bNweHf/7t3Mu6rMokICNGk=", "time": "2020-09-01T17:23:05.793355Z", "chan": 0, "tmst": 1849516944, "stat": 1, "modu": "LORA", "lsnr": 6.0, "rssi": -36, "rfch": 0, "codr": "4/5", "freq": 868.1, "datr": "SF7BW125", "size": 23}]}
2020-09-01 19:23:06.348 AABBCCDDEEFF7778: join request received
2020-09-01 19:23:06.348 AABBCCDDEEFF7778: querying the registry for OTAA device
2020-09-01 19:23:06.566 AABBCCDDEEFF7778: failed to get join devices from api. Missing IOT hub name
2020-09-01 19:23:06.566 AABBCCDDEEFF7778: processing time: 00:00:00.2203330

I tried to debug the error, but I'm not sure where it's thrown. I do see the API returning the device and a  iot hub connectionstring. I've set and checked the IOTEDGE_IOTHUBHOSTNAME in the environment variables. 

Any idea what I am doing wrong?

Mandur, 2020-09-01:
Are you only launching the function or also the LoRa Engine Edge part?

Feiko, 2020-09-01:
I also run the LoRa Engine (LoraWanNetworkSrvModule). But I don't run it in a IoT Edge (simulated) environment.

Mandur, 2020-09-01:
Did you added the properties described in point 4 [here](https://github.com/Azure/iotedge-lorawan-starterkit/blob/dev/Docs/devguide.md#debugging-in-visual-studio-outside-of-iot-edge-and-docker)?

Feiko, 2020-09-01:
The error is caught in the function GetDeviceForJoinRequestAsync in LoraDeviceRegistr.cs

var loader = this.GetOrCreateJoinDeviceLoader(matchingDeviceInfo);
var loRaDevice = await loader.WaitCompleteAsync();

It looks like the exception originates to line 39 in LoraDeviceClient, however the exception is thrown multiple times by different pieces of code. 
 
Line 39: this.deviceClient = DeviceClient.CreateFromConnectionString(this.connectionString, this.transportSettings);

Mandur, 2020-09-01:
Not sure it's the error you are facing but worth a shot

Feiko, 2020-09-01:
Yes, the 4 properties are added.

I can even find the connectionstring being used when debugging the code in LoraDeviceClient. It contains the right hostname. I've edited away the sensitive keys. 

"HostName=*******hub;DeviceId=AABBCCDDEEFF7778;SharedAccessKey=**********"

Mandur, 2020-09-02:
And just to be sure, the iot hub connection string you gave to the function is the iothubowner connection string right?

Feiko, 2020-09-03:
Ah I found the mistake. I had a typo in the .azure-devices.net suffix. A big thanks @Mandur for your help!

The LoRa device connects and I see data coming through 👍 . 

On connecting the device LoRaEngine gives get the outputmessage:
2020-09-03 10:58:53.056 AABBCCDDEEFF7778: the provided RX2DataRate is not valid

The DesiredRX2DataRate = null. Is this a setting I need to set in the Twin desiredproperty? I don't see any mention in the documentation.

Mandur, 2020-09-04:
Great!
Regarding the DesiredRX2DR, it's a glitch in the logging, I will submit a PR to fix it. That should not have an impact on the process as this is only the logging, thanks for bringing this up

PR #263 (2020-10-14): Prepare Release 1.0.5
Prepare release 1.0.5 with latest improvements from master and upgrade iot edge to latest security version. Please refer to the draft release note for more details.

This new release will fix #258 ,
closed 2020-10-14 by Mandur