Azure/iotedge-lorawan-starterkit #262: invalid argument "/lorawannetworksrvmodule:-amd64" for "-t, --tag" flag: invalid reference format

levalencia, 2020-10-08:
## Expected Behavior
Right Click Buiild and Push IoT Edge should not throw erros.

## Current Behavior
It throws this error:
invalid argument "/lorawannetworksrvmodule:-amd64" for "-t, --tag" flag: invalid reference format

## Steps to Reproduce
1. Right click on deployment.template.json and select build and push iot edge
2.
3.
4.

## Context (Environment)

### Device (Host) Operating System
Windows 10

### Architecture
Not sure, how do I check this?

### Container Operating System
Windows 10

### LoRaWAN Module Version
Not sure, how do I check this?

#### Docker
Client: Docker Engine - Community
 Cloud integration  0.1.18
 Version:           19.03.13
 API version:       1.40
 Go version:        go1.13.15
 Git commit:        4484c46d9d
 Built:             Wed Sep 16 17:00:27 2020
 OS/Arch:           windows/amd64
 Experimental:      false

Server: Docker Engine - Community
 Engine:
  Version:          19.03.13
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.13.15
  Git commit:       4484c46d9d
  Built:            Wed Sep 16 17:07:04 2020
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          v1.3.7
  GitCommit:        8fba4e9a7d01810a393d5d25a3621dc101981175
 runc:
  Version:          1.0.0-rc10
  GitCommit:        dc9208a3303feef5b3839f4323d9beb36df0a9dd
 docker-init:
  Version:          0.18.0
  GitCommit:        fec3683

## Logs

## Additional Information

Mandur, 2020-10-09:
Hello, 

Thanks for raising this issue.
Question: Did you set the local .env file as described [here](https://github.com/Azure/iotedge-lorawan-starterkit/blob/dev/Docs/devguide.md#follow-these-next-steps-in-both-cases) ?

levalencia, 2020-10-12:
I duplicated example.env and then filled the parameters, azure container registry, passwor, facade url and password.

something I noticed its that when you right click build and deploy the following command is generated

`PS C:\Users\valencil\Downloads\Repos\iotedge-lorawan-starterkit\LoRaEngine> docker build  --rm -f "c:\Users\valencil\Downloads\Repos\iotedge-lorawan-starterkit\LoRaEngine\modules\LoRaWanNetworkSrvModule\Dockerfile.amd64" -t $CONTAINER_REGISTRY_ADDRESS/lorawannetworksrvmodule:$NET_SRV_VERSION-amd64 "c:\Users\valencil\Downloads\Repos\iotedge-lorawan-starterkit" ; if ($?) { docker push $CONTAINER_REGISTRY_ADDRESS/lorawannetworksrvmodule:$NET_SRV_VERSION-amd64 } if ($?) { docker build  --rm -f "c:\Users\valencil\Downloads\Repos\iotedge-lorawan-starterkit\LoRaEngine\modules\LoRaWanPktFwdModule\Dockerfile.amd64" -t $CONTAINER_REGISTRY_ADDRESS/lorawanpktfwdmodule:$PKT_FWD_VERSION-amd64 "c:\Users\valencil\Downloads\Repos\iotedge-lorawan-starterkit\LoRaEngine\modules\LoRaWanPktFwdModule" } if ($?) { docker push $CONTAINER_REGISTRY_ADDRESS/lorawanpktfwdmodule:$PKT_FWD_VERSION-amd64 }`

the -t is usually for tags, but its tryng to append -amd64 to 2 variables@
` -t $CONTAINER_REGISTRY_ADDRESS/lorawanpktfwdmodule:$PKT_FWD_VERSION-amd64`

It looks like the problem is in the -amd64 append only.

But is this an error on the vs code extension?

levalencia, 2020-10-12:
I noticed this works depending on the architecture

It works for:  arm64v8 
but for the others I get the same error.

So the json file is generated, but I get a 2nd error (reported as a new bug)
This one can be closed

Mandur, 2020-10-12:
The problem was that the environment file was not named ".env". Correct naming solved the issue

closed 2020-10-12 by Mandur