Azure/iotedge-lorawan-starterkit #264: Right CLick Create deployment for single device.

levalencia, 2020-10-12:
Hello

After I created the gateway (iot edge) device on the portal and right click create deployment for single device in VS CODE
I get this error:

There are errors in deployment json file: data.modulesContent.$edgeAgent['properties.desired'].runtime.settings.registryCredentials should NOT have additional properties

Any idea how to fix this?

Mandur, 2020-10-12:
The problem was that the environment file was not named ".env". Correct naming solved the issue

closed 2020-10-12 by Mandur

michaelaelise, 2021-02-19:
Hm, mine is `.env` but I have the same error.

mcohen97, 2021-03-09:
I had this same issue, I saw that in the deployment.json file, there was a "settings" field which was truncated into three extra fields: "settings01", "settings02" and "settings03". I manually merged all of them in "settings" and it worked.