Azure/iotedge-lorawan-starterkit #266: sh: echo: I/O error ./reset_lgw.sh: 30: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent

levalencia, 2020-10-13:
## Expected Behavior
After selecting on VS Code Create deployment for single device, the lorawanpktfwd module should be up and running

## Current Behavior
Its failed

## Steps to Reproduce
Followed documentation, but I have Ubuntu 18.04

## Context (Environment)

### Device (Host) Operating System
Ubuntu 18.04

### Architecture
rmv7

### Container Operating System
Linux containers

### LoRaWAN Module Version

#### Docker
```
ubuntu@ubuntu:/sys/class/gpio$ docker version
Client:
 Version:           19.03.13+azure
 API version:       1.40
 Go version:        go1.13.15
 Git commit:        cd8016b6bcc51a51f577bcffb37e6a870f7813f9
 Built:             Tue Sep 15 23:02:04 2020
 OS/Arch:           linux/arm64
 Experimental:      false
Got permission denied while trying to connect to the Docker daemon socket at unix:///var/run/docker.sock: Get http://%2Fvar%2Frun%2Fdocker.sock/v1.40/version: dial unix /var/run/docker.sock: connect: permission denied
ubuntu@ubuntu:/sys/class/gpio$ sudo docker version
Client:
 Version:           19.03.13+azure
 API version:       1.40
 Go version:        go1.13.15
 Git commit:        cd8016b6bcc51a51f577bcffb37e6a870f7813f9
 Built:             Tue Sep 15 23:02:04 2020
 OS/Arch:           linux/arm64
 Experimental:      false

Server:
 Engine:
  Version:          19.03.13+azure
  API version:      1.40 (minimum version 1.12)
  Go version:       go1.13.15
  Git commit:       bd33bbf0497b2327516dc799a5e541b720822a4c
  Built:            Mon Mar 12 00:00:00 2018
  OS/Arch:          linux/arm64
  Experimental:     false
 containerd:
  Version:          1.3.7+azure
  GitCommit:        8fba4e9a7d01810a393d5d25a3621dc101981175
 runc:
  Version:          1.0.0-rc92
  GitCommit:        ff819c7e9184c13b7c2607fe6c30ae19403a7aff
 docker-init:
  Version:          0.18.0
  GitCommit:
```

## Logs
```
ubuntu@ubuntu:/sys/class/gpio$ sudo docker logs LoRaWanPktFwdModule
EU region detected.
Accessing concentrator reset pin through GPIO7...
sh: echo: I/O error
./reset_lgw.sh: 30: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent
./reset_lgw.sh: 32: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
./reset_lgw.sh: 33: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
./reset_lgw.sh: 34: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
./reset_lgw.sh: 37: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent
Using SPI dev 1 from environment variables
./start_pktfwd.sh: line 33: ./lora_pkt_fwd_spidev1: No such file or directory
EU region detected.
Accessing concentrator reset pin through GPIO7...
```

## Additional Information
I have the Seeduino lorawan hardware

Any help is appreciated

If I check the /sys/class/gpio folder, indeed there is no gpio7 folders, just this
```
ubuntu@ubuntu:/sys/class/gpio$ ls -al
total 0
drwxr-xr-x  2 root root    0 Oct 13 09:11 .
drwxr-xr-x 73 root root    0 Oct 13 09:03 ..
--w-------  1 root root 4096 Oct 13 08:31 export
lrwxrwxrwx  1 root root    0 Oct 13 09:17 gpiochip0 -> ../../devices/platform/soc/3f200000.gpio/gpio/gpiochip0
lrwxrwxrwx  1 root root    0 Oct 13 09:17 gpiochip100 -> ../../devices/gpiochip1/gpio/gpiochip100
lrwxrwxrwx  1 root root    0 Oct 13 09:17 gpiochip504 -> ../../devices/platform/soc/soc:firmware/soc:firmware:expgpio/gpio/gpiochip504
--w-------  1 root root 4096 Oct 13 09:17 unexport
```

Mandur, 2020-10-14:
Hello, 

Were you able to successfully run the packet forwarder outside of Docker? https://github.com/Lora-net/packet_forwarder

 If yes, please paste your global/local config files here.

If no, This bug is occurring at a lower error than our stack is operating. I fear we won't be able to provide much assistance there. you should probably open a bug on the [seeedstudio repo](https://github.com/Seeed-Studio/Seeed_Platform)

Mandur, 2020-10-19:
Based on #268, I deduct that this issue comes from spi interfaces on Ubuntu. Note that as mentioned above this operates below our stack. Therefore we can't provide support for it and we suggest to check directly with the hardware vendor.
Closing it for now, feel free to reopen if any questions.

closed 2020-10-19 by Mandur