Azure/iotedge-lorawan-starterkit #268: 404 Error on LorawanPktForwarder on raspberry stretch

levalencia, 2020-10-15:
so I decided to install rpi stretch insted of trying in Ubuntu, I did everything from scratch.
SPI is enabled and I can see GPIO7 under /sys/class/gpio so that seems to be fine.

However after restarting multiple times, I get this error:
```

[INFO] - Edge runtime is running.
[INFO] - [mgmt] - - - [2020-10-15 17:16:32.220772748 UTC] "GET /modules?api-version=2019-11-05 HTTP/1.1" 200 OK 2229 "-" "-" auth
[INFO] - [mgmt] - - - [2020-10-15 17:16:37.313402229 UTC] "GET /modules?api-version=2019-11-05 HTTP/1.1" 200 OK 2229 "-" "-" auth
[INFO] - [mgmt] - - - [2020-10-15 17:16:42.398163502 UTC] "GET /modules?api-version=2019-11-05 HTTP/1.1" 200 OK 2229 "-" "-" auth
[INFO] - [mgmt] - - - [2020-10-15 17:16:47.488300223 UTC] "GET /modules?api-version=2019-11-05 HTTP/1.1" 200 OK 2229 "-" "-" auth
[INFO] - [mgmt] - - - [2020-10-15 17:16:52.592329036 UTC] "GET /modules?api-version=2019-11-05 HTTP/1.1" 200 OK 2229 "-" "-" auth
[INFO] - [mgmt] - - - [2020-10-15 17:16:52.592329036 UTC] "GET /modules?api-version=2019-11-05 HTTP/1.1" 200 OK 2229 "-" "-" auth
[INFO] - [mgmt] - - - [2020-10-15 17:16:57.685492274 UTC] "GET /modules?api-version=2019-11-05 HTTP/1.1" 200 OK 2229 "-" "-" auth
[INFO] - Getting logs for module LoraWanPktFwdModule...
[WARN] - Could not get logs for module LoraWanPktFwdModule
[WARN] -         caused by: Container runtime error - Api(ApiError { code: 404, content: None })
[INFO] - [mgmt] - - - [2020-10-15 17:17:00.529667197 UTC] "GET /modules/LoraWanPktFwdModule/logs?api-version=2020-07-07&follow=fa
[INFO] - [mgmt] - - - [2020-10-15 17:17:02.772257167 UTC] "GET /modules?api-version=2019-11-05 HTTP/1.1" 200 OK 2229 "-" "-" auth
[INFO] - [mgmt] - - - [2020-10-15 17:17:07.876061203 UTC] "GET /modules?api-version=2019-11-05 HTTP/1.1" 200 OK 2229 "-" "-" auth
[INFO] - [mgmt] - - - [2020-10-15 17:17:12.974199988 UTC] "GET /modules?api-version=2019-11-05 HTTP/1.1" 200 OK 2229 "-" "-" auth
[INFO] - [mgmt] - - - [2020-10-15 17:17:18.058979529 UTC] "GET /modules?api-version=2019-11-05 HTTP/1.1" 200 OK 2229 "-" "-" auth
[INFO] - [mgmt] - - - [2020-10-15 17:17:23.151968364 UTC] "GET /modules?api-version=2019-11-05 HTTP/1.1" 200 OK 2229 "-" "-" auth
[INFO] - Stopping module LoRaWanPktFwdModule...
[WARN] - Could not stop module LoRaWanPktFwdModule
[WARN] -         caused by: Target of operation already in this state
[INFO] - [mgmt] - - - [2020-10-15 17:17:23.192509463 UTC] "POST /modules/LoRaWanPktFwdModule/stop?api-version=2019-11-05 HTTP/1.1
[INFO] - Starting module LoRaWanPktFwdModule...
[INFO] - Successfully started module LoRaWanPktFwdModule
[INFO] - [mgmt] - - - [2020-10-15 17:17:23.864849107 UTC] "POST /modules/LoRaWanPktFwdModule/start?api-version=2019-11-05 HTTP/1.
[INFO] - [work] - - - [2020-10-15 17:17:23.981975342 UTC] "POST /modules/%24edgeAgent/genid/637371606687179420/encrypt?api-versio
[INFO] - [mgmt] - - - [2020-10-15 17:17:29.177010444 UTC] "GET /modules?api-version=2019-11-05 HTTP/1.1" 200 OK 2229 "-" "-" auth
[INFO] - Checking edge runtime status
[INFO] - Edge runtime is running.
[INFO] - [mgmt] - - - [2020-10-15 1
```

I dont know why at the end of the logs it says  Successfully started module LoRaWanPktFwdModule, because in VSCODE and  in the portal it appears as backoff, and in dockerps it does not even appear

Mandur, 2020-10-19:
Could you please paste the log of the loraPktFwd? (docker logs LoRaWanPktFwdModule)?
Additionally if you could provide the module twins of the LoRaWanPktFwdModule module on azure we might be able to get a better idea .

levalencia, 2020-10-19:
on the docker logs, I got this multiple times

Accessing concentrator reset pin through GPIO7...
Currently only a customized value of 2Mbps is supported as custom SPI speed
EU region detected.
Accessing concentrator reset pin through GPIO7...
Currently only a customized value of 2Mbps is supported as custom SPI speed
EU region detected.

levalencia, 2020-10-19:
and the twin json

{
  "deviceId": "loragtw_bxl_G1",
  "moduleId": "LoRaWanPktFwdModule",
  "etag": "AAAAAAAAAAE=",
  "deviceEtag": "abcd",
  "status": "enabled",
  "statusUpdateTime": "0001-01-01T00:00:00Z",
  "connectionState": "Disconnected",
  "lastActivityTime": "0001-01-01T00:00:00Z",
  "cloudToDeviceMessageCount": 0,
  "authenticationType": "sas",
  "x509Thumbprint": {
    "primaryThumbprint": null,
    "secondaryThumbprint": null
  },
  "version": 2,
  "properties": {
    "desired": {
      "$metadata": {
        "$lastUpdated": "0001-01-01T00:00:00Z"
      },
      "$version": 1
    },
    "reported": {
      "$metadata": {
        "$lastUpdated": "0001-01-01T00:00:00Z"
      },
      "$version": 1
    }
  }
}

levalencia, 2020-10-19:
If you feel like a remote session would help, I m happy to do it and share my screen

closed 2021-01-20 by levalencia