Azure/iotedge-lorawan-starterkit #272: packet forwarder not running

donniekerr, 2020-11-10:
I've installed iotedge and set modules on my edge Gateway device. All runs except pktfwd
Below is the log of the packet forwarder.  What am I missing to get it to run? 

I installed the spi-enable files per @gerfen 
when I ls/dev I see there two
 Confirm spidev1.0 and spidev1.1 exist

Here is the full support bundle 
[support_bundle (1).zip](https://github.com/Azure/iotedge-lorawan-starterkit/files/5519918/support_bundle.1.zip)

Any ideas?  Thanks!

`US region detected.
Accessing concentrator reset pin through GPIO7...
sh: echo: I/O error
./reset_lgw.sh: 30: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent
./reset_lgw.sh: 33: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
./reset_lgw.sh: 34: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
./reset_lgw.sh: 37: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent
No SPI Dev version detected in environment variables on x86, defaulting to SPI Dev 2
*** Beacon Packet Forwarder for Lora Gateway ***
Version: 4.0.1
*** Lora concentrator HAL library version info ***
Version: 5.0.1;
***
INFO: Little endian host
INFO: found global configuration file global_conf.json, parsing it
INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
INFO: lorawan_public 1, clksrc 1
INFO: no configuration for LBT
INFO: antenna_gain 0 dBi
INFO: Configuring TX LUT with 16 indexes
INFO: radio 0 enabled (type SX1257), center frequency 902700000, RSSI offset -166.000000, tx enabled 1, tx_notch_freq 0
INFO: radio 1 enabled (type SX1257), center frequency 903400000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
INFO: Lora multi-SF channel 0>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 1>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 2>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 4>  radio 1, IF -300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 5>  radio 1, IF -100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 6>  radio 1, IF 100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 7>  radio 1, IF 300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora std channel> radio 0, IF 300000 Hz, 500000 Hz bw, SF 8
INFO: FSK channel 8 disabled
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000000
INFO: server hostname or IP address is configured to "192.168.0.17"
INFO: upstream port is configured to "1680"
INFO: downstream port is configured to "1680"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: found local configuration file local_conf.json, parsing it
INFO: redefined parameters will overwrite global parameters
INFO: local_conf.json does not contain a JSON object named SX1301_conf
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000101
INFO: server hostname or IP address is configured to "172.17.0.1"
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
ERROR: [main] failed to start the concentrator
US region detected.
sh: echo: I/O error
Accessing concentrator reset pin through GPIO7...
./reset_lgw.sh: 30: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent
./reset_lgw.sh: 33: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
[... 182 lines cut ...]
INFO: Lora multi-SF channel 6>  radio 1, IF 100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 7>  radio 1, IF 300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora std channel> radio 0, IF 300000 Hz, 500000 Hz bw, SF 8
INFO: FSK channel 8 disabled
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000000
INFO: server hostname or IP address is configured to "192.168.0.17"
INFO: upstream port is configured to "1680"
INFO: downstream port is configured to "1680"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: found local configuration file local_conf.json, parsing it
INFO: redefined parameters will overwrite global parameters
INFO: local_conf.json does not contain a JSON object named SX1301_conf
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000101
INFO: server hostname or IP address is configured to "172.17.0.1"
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
ERROR: [main] failed to start the concentrator`

gerfen, 2020-11-10:
If you have the AAEON AIOT-IP68 outdoor gateway, the reset pin need to be set to 430.

--Michael

donniekerr, 2020-11-10:
Thanks Michael,  I do have outdoor version.  I changed the reset pin to 430 and reset modules.  It changed the module status from failed to backoff. 
Log looks better but still says failed to start at the end.
Here is new log ...

NFO: Little endian host
INFO: found global configuration file global_conf.json, parsing it
INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
INFO: lorawan_public 1, clksrc 1
INFO: no configuration for LBT
INFO: antenna_gain 0 dBi
INFO: Configuring TX LUT with 16 indexes
INFO: radio 0 enabled (type SX1257), center frequency 902700000, RSSI offset -166.000000, tx enabled 1, tx_notch_freq 0
INFO: radio 1 enabled (type SX1257), center frequency 903400000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
INFO: Lora multi-SF channel 0>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 1>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 2>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 4>  radio 1, IF -300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 5>  radio 1, IF -100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 6>  radio 1, IF 100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 7>  radio 1, IF 300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora std channel> radio 0, IF 300000 Hz, 500000 Hz bw, SF 8
INFO: FSK channel 8 disabled
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000000
INFO: server hostname or IP address is configured to "192.168.0.17"
INFO: upstream port is configured to "1680"
INFO: downstream port is configured to "1680"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: found local configuration file local_conf.json, parsing it
INFO: redefined parameters will overwrite global parameters
INFO: local_conf.json does not contain a JSON object named SX1301_conf
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000101
INFO: server hostname or IP address is configured to "172.17.0.1"
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
ERROR: [main] failed to start the concentrator

gerfen, 2020-11-10:
The other setting you may need to check in your .env is PKT_FWD_SPI_DEV.  Set it like this:

PKT_FWD_SPI_DEV=1

donniekerr, 2020-11-10:
@gerfen , well, don't know what .env you are referring to.  Where do I change PKT_FWD_SPI_DEV=1 ? 

Also, where in the docs does it talk about setting the reset pin to 430?  I don't recall seeing anything about it. I need to learn what it does and how the reset pin works?

gerfen, 2020-11-11:
You can set both the RESET_PIN and the PKT_FWD_SPI_DEV variables from the Azure IoT hub.

[image]

Regarding setting the RESET_PIN to 430... I got that information from an AAEON engineer three years ago when I first set up the AAEON AIOT-IP68 gateway.  I don't recall if it was on the up-community.org forum or not.  Sorry.  If I can find the discussion I'll add a link in another reply.  Sorry I don't have anything more specific.

donniekerr, 2020-11-11:
I changed SPI_DEV to 1 and re-ran set modules.  But it doesn't seem to help resolve the issue.  My assumption is that I expect to see the concentrator start and for the module status to be running. (even if no lora devices is transmitting yet). 

`ccessing concentrator reset pin through GPIO7...
sh: echo: I/O error
./reset_lgw.sh: 30: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent
./reset_lgw.sh: 33: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
./reset_lgw.sh: 34: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
./reset_lgw.sh: 37: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent
No SPI Dev version detected in environment variables on x86, defaulting to SPI Dev 2
*** Beacon Packet Forwarder for Lora Gateway ***
Version: 4.0.1
*** Lora concentrator HAL library version info ***
Version: 5.0.1;
***
INFO: Little endian host
INFO: found global configuration file global_conf.json, parsing it
INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
INFO: lorawan_public 1, clksrc 1
INFO: no configuration for LBT
INFO: antenna_gain 0 dBi
INFO: Configuring TX LUT with 16 indexes
INFO: radio 0 enabled (type SX1257), center frequency 902700000, RSSI offset -166.000000, tx enabled 1, tx_notch_freq 0
INFO: radio 1 enabled (type SX1257), center frequency 903400000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
INFO: Lora multi-SF channel 0>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 1>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 2>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 4>  radio 1, IF -300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 5>  radio 1, IF -100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 6>  radio 1, IF 100000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 7>  radio 1, IF 300000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora std channel> radio 0, IF 300000 Hz, 500000 Hz bw, SF 8
INFO: FSK channel 8 disabled
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000000
INFO: server hostname or IP address is configured to "192.168.0.17"
INFO: upstream port is configured to "1680"
INFO: downstream port is configured to "1680"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: found local configuration file local_conf.json, parsing it
INFO: redefined parameters will overwrite global parameters
INFO: local_conf.json does not contain a JSON object named SX1301_conf
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000101
INFO: server hostname or IP address is configured to "172.17.0.1"
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
ERROR: [main] failed to start the concentrator`

gerfen, 2020-11-11:
It appears that you are still using GPIO Pin 7 and not GPIO PIN 430 as the 'RESET_PIN' 

```
Accessing concentrator reset pin through GPIO7...
```
and SPI_DEV 2 (which is the default) instead of SPI_DEV 1.
```
No SPI Dev version detected in environment variables on x86, defaulting to SPI Dev 2
```

What do you see if you putty or SSH into the gateway and run 

```
sudo iotedge list
```

donniekerr, 2020-11-11:
yes, it looks like the environment variables aren't getting down to the edge device.  Not sure why.  I can re-run set modules again. 

When I run iotedge list I see all modules running except PktFwd which says failed.

donniekerr, 2020-11-11:
It is hard to tell when the previous log ends and the latest ones begins because there is no timestamps. So, I'm not sure if the reset pin is just a previous one or not.

donniekerr, 2020-11-11:
I'll re-run the support-bundle tomorrow so I can look into it

donniekerr, 2020-11-11:
Unfortunately, the support bundle doesn't indicate why the Module's Env variables are not getting updated.  I set modules a few times and still see the wrong reset pin and dev setting in the logs. Any idea why they aren't updated on the edge device?

gerfen, 2020-11-11:
I manually set up my system so I set the variables in the .env file I mentioned earlier.  I'll see if I can figure out where those variables get set on the gateway and will get back to you.  I'm on the road today so it will be several more hours before I can look.

--Michael

donniekerr, 2020-11-11:
ok, so I looked at the code I downloaded from this site. (I just deployed the same code as is).
the LoraEngine/example.env is the only env in the folder structure.  It contains:

# Reset pin (GPIO to reset Packet Forwarder Antenna)
RESET_PIN=7

##################
# edgeHub settings
##################
EDGEHUB_ROUTE='FROM /* INTO $upstream'

##############################
# LoRaWanPktFwdModule settings
##############################
PKT_FWD_VERSION=1.0.0
#SPI DEV version information. Only needed for x86 gateways. Possible values are 1 or 2.
PKT_FWD_SPI_DEV=2
#Set a custom speed for your SPI or leave commented if not in use. Supported custom speed: 2Mb/sec.
#PKT_FWD_SPI_SPEED=2

deployment.template.json has this ...
"env": {
              "RESET_PIN": {
                "value": "$RESET_PIN"
              },
              "REGION": {
                "value": "$REGION"
              },
              "SPI_DEV":{
                "value":"$PKT_FWD_SPI_DEV"
              },
              "SPI_SPEED":
              {
               "value":"$PKT_FWD_SPI_SPEED"
              }
            },

and my set modules deployment has:
"env": {
                            "RESET_PIN": {
                                "value": "430"
                            },
                            "REGION": {
                                "value": "US"
                            },
                            "SPI_DEV": {
                                "value": "1"
                            }
                        },

Does this look correct? Or was I supposed to create a separate env file before deploying the code?

gerfen, 2020-11-11:
I have always created the .env file prior to deployment.  Without the .env file I think the code defaults to SPI_DEV=2 and RESET_PIN=7.

donniekerr, 2020-11-11:
ah.  I see what I did wrong.  When I used the Quick Start ARM to automatically set everything up, I kept the defaults for these values.  But, that still doesn't explain why the updated values aren't getting to the edge.

So, I can either re-run the Quick Start or manually create a .env file or both

gerfen, 2020-11-11:
I think I would try creating the .env file first.

--Michael

donniekerr, 2020-11-12:
I got it working by re-running the Quick Start and using the settings recommended by @gerfen .  Thanks Michael!
[image]

closed 2020-11-12 by donniekerr