Azure/iotedge-lorawan-starterkit #276: another architecture question - connection strings and certificates required?

donniekerr, 2020-11-19:
When I register a downstream device and set the keys in the device twin, is that all I need to do to allow the device to connect to the Gateway?  The downstream device code does have the same keys when it sends. 

I'm just not sure if I have to use the downstream device Connection Strings or not? Or if you have to do this complicated certificate stuff?
https://docs.microsoft.com/en-us/azure/iot-edge/how-to-create-transparent-gateway?view=iotedge-2018-06#set-up-the-device-ca-certificate 
It doesn't look like this kit requires all of that, am I correct?  Connecting Lora Devices is made much easier with your kit, right?
Looks like the Azure Function manages which downstream devices are allowed to connect to the Gateway and connection strings for each downstream device are not required? As long as the join process works with the NwkSkey, etc. 

Just want to be sure I'm not missing anything with the downstream device setup requirements.  I don't see anything in the arduino sample code related to the connection strings or certs.  Just the NwkSkey and the AppSkey.

donniekerr, 2020-11-25:
The Azure Function gets the device twin and connection strings from IoT Hub and enables the connection to the device, right?  As long the device twin keys match the device keys, then it should be able to join via ABP, right?

I'm asking this because I'm not seeing any join/connect entries or data traffic in the PktFwd logs when my lora node device is transmitting. How do I verify the Lora node (32u4) is trying to connect to the gateway(AAEON)?  Or even if it is getting to the Concentrator? I also don't see any requests in the Azure Function logs.  I would expect to see something there if it was able to join, right?

donniekerr, 2020-12-01:
Anyone get ABP working? How do you verify a device join attempt? Or should we switch to OTTA to see if that works instead?

ronniesa, 2020-12-01:
ABP and OTAA both works and we use real devices in our pipeline to test the system both in ABP and OTTA.

ABT normally is even easier to make it work. Pay attention to the keys and some devices needs the key inverted so to do a quick test is to use a mirroed key like FF00000000FF.

Best is to look on the Network Server docker log to se what is going on.

donniekerr, 2020-12-02:
Thank you @ronniesa .  I have my electronics guy give it a try.  He is getting CRC errrors.  I've asked him to send me the exact errors since I don't know what those errors mean.   I'll post them when I get it. I think it is on his arduino side, not the Gateway side.

donniekerr, 2020-12-02:
CRC issue is resolved.  And packets are getting through.  We just need to add in the decoder.

But, any idea why we see this network id error?
**67B4942B: device is using another network id, ignoring this message (network: 1, devAddr: 102)** 

don't know where network 1 or devAddr 102 comes from. 

our twin has these test keys for ABP 
"AppSKey": "FD92015E88AB091A4462B06E857F0C0B",
      "NwkSKey": "FB6A63864701CD596501A9A93D942AC3",
      "DevAddr": "67B4942B",

I also don't see any calls the Key Manager function.  I would expect to see something in the function logs, but nothing. 

Any assistance is appreciated.  Thanks

ronniesa, 2020-12-02:
The devaddr contains also a network id and for private network it must be 0 or 1, we use 1 as default. For ABT you need to generate a valid devaddr with the network id.  Our  cmd utility can generate one for you. https://github.com/Azure/iotedge-lorawan-starterkit/tree/dev/Tools/Cli-LoRa-Device-Provisioning

For a quick test you can try this one 0228B1B1 both in the lora device and the twin needs to match.

donniekerr, 2020-12-03:
ah ha!  I must have missed this point! Thank you.  I will try this test devAddr.  The table that defines these keys just says a 4 bit Hex string.   And I will use the cmd to add future devices.  This is just for a prototype. 

Will let you know how it goes tomorrow!

donniekerr, 2020-12-03:
Progress! That worked to at least call GetDevice in the function.  Don't know what the object error means in the logs though.  

[image]

Mandur, 2020-12-03:
May I ask what function version you are currently using?

donniekerr, 2020-12-03:
@Mandur , I'm not sure.  I used Quick Start to generate it about a month ago.  I looked at the function app and don't see any reference to a version number.  Where can I find it? All the code for it is only in the Portal.

donniekerr, 2020-12-03:
Related questions. 
1) How do I see which devices were created in the Key Manager/Redis Cache/Storage account? I just want to see if the new devAddr that @ronniesa gave me has updated there.  
2) what do I use for GatewayID in a private network?  currently I'm using the id of the edge device the node connects to. 
Just wondering if either of these might be related to the above azure function error or not?

donniekerr, 2020-12-03:
This doesn't look right in the trace with the null ...
PublicKeyToken=nullLoraKeysManagerFacade 

what is missing that is throwing the NullReferenceException?

gerfen, 2020-12-03:
From the redis console you can type the following:

```keys *```

which will give you a list which looks like this...

```
KEYS command is unsupported. Sending "SCAN 0 COUNT 1000 MATCH *" instead.
1) "0"
2)   1) "devAddrTable:02C2B40E"
     2) "devAddrTable:0242E26A"
     3) "devAddrTable:0269B414"
     4) "devAddrTable:031A5587"
... reset elided
```

Then you can see the info for a specific device by typing this:

```hgetall devAddrTable:03FDA89F ```

```
>hgetall devAddrTable:03FDA89F
1) "0004A30B00240C03"
2) "{\"GatewayId\":\"<your-gateway-id-if-set>\",\"LastUpdatedTwins\":\"2019-12-02T20:56:18.9252032Z\",\"NwkSKey\":\"<your NwkSKey >\",\"DevAddr\":\"03FDA89F\",\"DevEUI\":\"0004A30B00240C03\",\"PrimaryKey\":\"<your PrimaryKey>\"}"
```
Regarding the GatewayID... It can be whatever you like, for example I use ``` evolution-indoor``` for one of my test gateways.

ronniesa, 2020-12-03:
GatewayID if specified must match the iotedge name. You could leave it blank. If you specify it on the device twin you force that only that gateway will process messages for that device. Is a bit faster if you specify it

ronniesa, 2020-12-03:
There are two caches one on the module, we have a directmethod to clear it or you can restart the module and we have redis that is refreshed by a scheduled task.

In case if you change the devaddr is like a new device for the system so there is no need to clear the cache as the devaddr is the key

donniekerr, 2020-12-03:
Thanks! Looks like cache updated just fine to the test devAddr.  
Don't know why PrimaryKey is null.  Is that what throws the null error in GetDevice?
`>keys *
KEYS command is unsupported. Sending "SCAN 0 COUNT 1000 MATCH *" instead.
1) "0"
2) 1) "lastDeltaUpdateKeyValue"
   2) "devAddrTable:0228B1B1"
   3) "fullUpdateKey"
>hgetall devAddrTable:0228B1B1
1) "wanda1"
2) "{\"GatewayId\":\"edgegateway\",\"LastUpdatedTwins\":\"2020-12-03T03:20:45.9790535Z\",\"NwkSKey\":\"FB6A63864701CD596501A9A93D942AC3\",\"DevAddr\":\"0228B1B1\",\"DevEUI\":\"wanda1\",\"PrimaryKey\":null}"
`

ronniesa, 2020-12-03:
It can be null as the way to get the twin is different than the one to get the IoT Hub key we only do when is needed:

Here is the piece of code that will return the primarykey if is null in the cache.

if (!string.IsNullOrEmpty(devAddressesInfo[i].DevEUI))
                                {
                                    // device was not yet populated
                                    if (!string.IsNullOrEmpty(devAddressesInfo[i].PrimaryKey))
                                    {
                                        results.Add(devAddressesInfo[i]);
                                    }
                                    else
                                    {
                                        // we need to load the primaryKey from IoTHub
                                        // Add a lock loadPrimaryKey get lock get
                                        devAddressesInfo[i].PrimaryKey = await this.LoadPrimaryKeyAsync(devAddressesInfo[i].DevEUI);
                                        results.Add(devAddressesInfo[i]);
                                        devAddrCache.StoreInfo(devAddressesInfo[i]);
                                    }

                                    // even if we fail to acquire the lock we wont enter in the next condition as devaddressinfo is not null
                                }

ronniesa, 2020-12-03:
I don't think it should matter for ABP but for OTAA for sure the deviceid in iot hub needs to be compliant and wnada1 is not a valid DevEUI.

Can you please try to create a new device with a valid deviceid like this 46AAC86800430028 instead of wanda1. Please delete then wanda1 or change the devaddr of wanda1 and clear the cache.

I'm not sure it will solve the issue but I'm sure we only tested with valid deviceid so we can rule out this eventually.

donniekerr, 2020-12-04:
I deleted wanda1 device
Cleared Redis cache
created new device with new id

cache was empty for quite awhile and then it updated to this 

>keys *
KEYS command is unsupported. Sending "SCAN 0 COUNT 1000 MATCH *" instead.
1) "0"
2) 1) "fullUpdateKey"
   2) "lastDeltaUpdateKeyValue"
   3) "devAddrTable:0228B1B1"
>hgetall devAddrTable:0228B1B1
1) "46AAC86800430028"
2) "{\"GatewayId\":\"edgegateway\",\"LastUpdatedTwins\":\"2020-12-03T21:54:41.0101821Z\",\"NwkSKey\":\"FB6A63864701CD596501A9A93D942AC3\",\"DevAddr\":\"0228B1B1\",\"DevEUI\":\"46AAC86800430028\",\"PrimaryKey\":null}"

Was PrimaryKey supposed to be there?

donniekerr, 2020-12-04:
Here is the trace after I made the changes suggested by @ronniesa 

Same error 
at LoraKeysManagerFacade.VersionValidator.Validate (LoraKeysManagerFacade, Version=1.0.0.0, Culture=neutral, PublicKeyToken=nullLoraKeysManagerFacade, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null: C:\Users\mikha\iotedge-lorawan-starterkit\LoRaEngine\LoraKeysManagerFacade\VersionValidator.csLoraKeysManagerFacade, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null: 14)

`Microsoft.Azure.WebJobs.Host.FunctionInvocationException:
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw (System.Private.CoreLib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e)
   at Microsoft.Azure.WebJobs.Host.Executors.FunctionExecutor+<ExecuteWithLoggingAsync>d__19.MoveNext (Microsoft.Azure.WebJobs.Host, Version=3.0.23.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35Microsoft.Azure.WebJobs.Host, Version=3.0.23.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35: C:\projects\azure-webjobs-sdk-rqm4t\src\Microsoft.Azure.WebJobs.Host\Executors\FunctionExecutor.csMicrosoft.Azure.WebJobs.Host, Version=3.0.23.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35: 293)
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw (System.Private.CoreLib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e)
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Private.CoreLib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Private.CoreLib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult (System.Private.CoreLib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e)
   at Microsoft.Azure.WebJobs.Host.Executors.FunctionExecutor+<TryExecuteAsyncCore>d__16.MoveNext (Microsoft.Azure.WebJobs.Host, Version=3.0.23.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35Microsoft.Azure.WebJobs.Host, Version=3.0.23.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35: C:\projects\azure-webjobs-sdk-rqm4t\src\Microsoft.Azure.WebJobs.Host\Executors\FunctionExecutor.csMicrosoft.Azure.WebJobs.Host, Version=3.0.23.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35: 97)
Inner exception System.NullReferenceException handled at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw:
   at LoraKeysManagerFacade.VersionValidator.Validate (LoraKeysManagerFacade, Version=1.0.0.0, Culture=neutral, PublicKeyToken=nullLoraKeysManagerFacade, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null: C:\Users\mikha\iotedge-lorawan-starterkit\LoRaEngine\LoraKeysManagerFacade\VersionValidator.csLoraKeysManagerFacade, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null: 14)
   at LoraKeysManagerFacade.DeviceGetter+<GetDevice>d__3.MoveNext (LoraKeysManagerFacade, Version=1.0.0.0, Culture=neutral, PublicKeyToken=nullLoraKeysManagerFacade, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null: C:\Users\mikha\iotedge-lorawan-starterkit\LoRaEngine\LoraKeysManagerFacade\DeviceGetter.csLoraKeysManagerFacade, Version=1.0.0.0, Culture=neutral, PublicKeyToken=null: 41)
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw (System.Private.CoreLib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e)
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Private.CoreLib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Private.CoreLib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult (System.Private.CoreLib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e)
   at Microsoft.Azure.WebJobs.Host.Executors.FunctionInvoker`2+<InvokeAsync>d__10.MoveNext (Microsoft.Azure.WebJobs.Host, Version=3.0.23.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35Microsoft.Azure.WebJobs.Host, Version=3.0.23.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35: C:\projects\azure-webjobs-sdk-rqm4t\src\Microsoft.Azure.WebJobs.Host\Executors\FunctionInvoker.csMicrosoft.Azure.WebJobs.Host, Version=3.0.23.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35: 52)
   at System.Runtime.ExceptionServices.ExceptionDispatchInfo.Throw (System.Private.CoreLib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e)
   at System.Runtime.CompilerServices.TaskAwaiter.ThrowForNonSuccess (System.Private.CoreLib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e)
   at System.Runtime.CompilerServices.TaskAwaiter.HandleNonSuccessAndDebuggerNotification (System.Private.CoreLib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e)
   at System.Runtime.CompilerServices.TaskAwaiter`1.GetResult (System.Private.CoreLib, Version=4.0.0.0, Culture=neutral, PublicKeyToken=7cec85d7bea7798e)
   at Microsoft.Azure.WebJobs.Host.Executors.FunctionExecutor+<InvokeAsync>d__27.MoveNext (Microsoft.Azure.WebJobs.Host, Version=3.0.23.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35Microsoft.Azure.WebJobs.Host, Version=3.0.23.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35: C:\projects\azure-webjobs-sdk-rqm4t\src\Microsoft.Azure.WebJobs.Host\Executors\FunctionExecutor.csMicrosoft.Azure.WebJobs.Host, Version=3.0.23.0, Culture=neutral, PublicKeyToken=31bf3856ad364e3 [... cut ...]
   at Microsoft.Azure.WebJobs.Host.Executors.FunctionExecutor+<ExecuteWithLoggingAsync>d__19.MoveNext (Microsoft.Azure.WebJobs.Host, Version=3.0.23.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35Microsoft.Azure.WebJobs.Host, Version=3.0.23.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35: C:\projects\azure-webjobs-sdk-rqm4t\src\Microsoft.Azure.WebJobs.Host\Executors\FunctionExecutor.csMicrosoft.Azure.WebJobs.Host, Version=3.0.23.0, Culture=neutral, PublicKeyToken=31bf3856ad364e35: 245)
`

donniekerr, 2020-12-04:
I'm not a .Net developer (python, node) ... which is why I'm using the Quick Start and not modifying your code.  but it looks like it doesn't like line 14
 req.HttpContext.Response.Headers.Add(ApiVersion.HttpHeaderName, currentApiVersion.Version);

Don't know what ApiVersion is, but I'm assuming one of these inputs is null.

ronniesa, 2020-12-04:
> Was PrimaryKey supposed to be there?

As posted above, it is fine that is null in the cache. Getting the key is a costly operation that we do when we process the first message and as the function is crashing before we don't get the key. The value in the cache are refreshed by a scheduled job this is why in cache we have the info but the Edge module do not get them. Is a separate code path.

donniekerr, 2020-12-04:
Oh, sorry didn't catch that before.  Thank you for explaining again.  I'm trying to learn how the flow works for ABP.  So, you are saying the crash is expected on first message and I can basically ignore it until cache catches up?  And upon receipt of another message, getDevice will be successful, the module will pick up the device, and subsequent messages will get through? 

What should I expect to see when getDevice is successful? Because I'm only seeing repeated failed calls in the logs after days.  I was thinking it was an issue with the deviceId and devAddr not being in correct format or something.
All I keep seeing on the endpoint is a repeat of this sequence and my goal, I thought, was to get rid of the "device is not our device" error. I thought the nullexception was preventing it from ever finding the device. If that is the case, then I'm not sure what this is telling me below.

`"Physical dataUp {\"rxpk\":[{\"tmst\":3327998667,\"chan\":2,\"rfch\":0,\"freq\":902.700000,\"stat\":1,\"modu\":\"LORA\",\"datr\":\"SF7BW125\",\"codr\":\"4/5\",\"lsnr\":0.5,\"rssi\":-106,\"size\":17,\"data\":\"QLGxKAKAxA8BzVTB/6y6kHY=\"}]}"
[IoTHubMonitor] [7:33:09 AM] Message received from [edgegateway/LoRaWanNetworkSrvModule]:
"0228B1B1: querying the registry for device"
[IoTHubMonitor] [7:33:09 AM] Message received from [edgegateway/LoRaWanNetworkSrvModule]:
"0228B1B1: error calling get device function api: Internal Server Error, status: InternalServerError, check the azure function log"
[IoTHubMonitor] [7:33:09 AM] Message received from [edgegateway/LoRaWanNetworkSrvModule]:
"0228B1B1: querying the registry for devices by devAddr 0228B1B1 found 0 devices"
[IoTHubMonitor] [7:33:09 AM] Message received from [edgegateway/LoRaWanNetworkSrvModule]:
"0228B1B1: device is not our device, ignore message"`

Mandur, 2020-12-04:
Hey @donniekerr, Would you be available to have a quick screensharing session? We might have few idea about the issue you are currently facing but it would help to see it running live for a better understanding and fix. Would you be available around 17:00 (Houston time) today? My Skype is Mandurlevrai

donniekerr, 2020-12-04:
That sounds great. Can I send you a Teams invite though? I want my device guy to join too and that is easier. Just need your email.

Mandur, 2020-12-04:
Sure, chmikhai@microsoft.com!

Mandur, 2020-12-04:
Let me know when you copied it ;)

donniekerr, 2020-12-04:
done

donniekerr, 2020-12-04:
@Mandur let me know if you got the teams invite.  For some reason Teams didn't find your email, so I sent it to you via email.

Mandur, 2020-12-04:
I got the team invite, It's for in 55 min right? :P (I am very bad with timetzones)

donniekerr, 2020-12-04:
Oh.  I thought you said 5pm Central.  It is 9:15 am here now.  I'm free any time after 11:30 am today.  does 2 hours from now work for you?

Mandur, 2020-12-04:
Yes Sure!

donniekerr, 2020-12-04:
Thank you @Mandur for taking the time to share screen.  We installed the bug fix in the Function App and I will let you know if it resolved the error or not.

donniekerr, 2020-12-04:
[image]
@Mandur like what you see?  Yay!!!  Thanks for the bug fix!

donniekerr, 2020-12-04:
"Physical dataUp {\"rxpk\":[{\"tmst\":3442102851,\"chan\":2,\"rfch\":0,\"freq\":902.700000,\"stat\":1,\"modu\":\"LORA\",\"datr\":\"SF7BW125\",\"codr\":\"4/5\",\"lsnr\":8.2,\"rssi\":-41,\"size\":17,\"data\":\"QLGxKAKATQABoG86/obgB5M=\"}]}"
[IoTHubMonitor] [5:56:17 PM] Message received from [edgegateway/LoRaWanNetworkSrvModule]:
"46AAC86800430028: device in cache"
[IoTHubMonitor] [5:56:17 PM] Message received from [edgegateway/LoRaWanNetworkSrvModule]:
"46AAC86800430028: converted 16bit FCnt 77 to 32bit FCnt 77"
[IoTHubMonitor] [5:56:17 PM] Message received from [edgegateway/LoRaWanNetworkSrvModule]:
"46AAC86800430028: invalid frame counter, message ignored, msg: 77 server: 5102"
[IoTHubMonitor] [5:56:17 PM] Message received from [edgegateway/LoRaWanNetworkSrvModule]:
"46AAC86800430028: processing time: 00:00:00.0053535"

donniekerr, 2020-12-05:
How did you say I reset the counter? maybe the 5102 is the limit like you said?  see error above.

Getting close!!

ronniesa, 2020-12-05:
For resetting the counter on a device https://github.com/Azure/iotedge-lorawan-starterkit/blob/dev/Docs/quickstart.md#optional-device-properties under reset counter 

only for ABP (OTAA reset automatically at every join)

donniekerr, 2020-12-05:
Thank you @Mandur @ronniesa @gerfen for all the help!  We are finally error free and successfully getting data from the Feather Lora Node  to the hub!  I will close this for now.  Please let me know when the bug fix is live and what the zip url should be changed to in the function.

closed 2020-12-05 by donniekerr