Azure/iotedge-lorawan-starterkit #282: Remove Comment from #PKT_FWD_SPI_SPEED=2

levalencia, 2021-01-20:
The LoraWanPktFwd fails if this line is commented! I learnt it the hard way.  

It should always be uncommented!

#PKT_FWD_SPI_SPEED=2

maxstefaniv, 2021-02-03:
Hello, I am experiencing the same issue trying to install and run everything from the manual, and I am getting the same probplem... 
Where should I locate the file on the raspberry Pi folders? in order to change that line.
The error I am geting is :
`INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: found local configuration file local_conf.json, parsing it
INFO: redefined parameters will overwrite global parameters
INFO: local_conf.json does not contain a JSON object named SX1301_conf
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000101
INFO: server hostname or IP address is configured to "172.17.0.1"
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
ERROR: Failed to load fw 1
ERROR: Version of calibration firmware not expected, actual:0 expected:2
ERROR: [main] failed to start the concentrator
`

levalencia, 2021-02-03:
One of the steps in the instructions is to take the example.env and rename to .env, That file contains the line I mentioned in this issue.

maxstefaniv, 2021-02-03:
Ty! (Gracias)
I deployed through the step of ARM template so I didn't have to modify that .env but I changed environment variables through the Twin:
[image]
and still fails :(
this is the error:
[image]

I am deploying on a Raspberry Pi using the LoRa/LoRaWAN Gateway Kit

Thanks in advance!

levalencia, 2021-02-03:
are you using Raspberry stretch OS?

I had issues like this when I tried ubuntu or debian.

maxstefaniv, 2021-02-03:
Using the default raspberry software:
[image]
[image]

levalencia, 2021-02-03:
According to this link:
https://docs.microsoft.com/en-us/azure/iot-edge/how-to-install-iot-edge?view=iotedge-2018-06

iotedge is only compatible with Raspberry Stretch (an old version of Raspberry OS).

I did tried Ubuntu, Debian, and the normal Raspberry OS.  I couldnt make it work on ANY.

Finally I installed my edge device with Raspberry Stretch:
Guide:
https://howchoo.com/raspbian/raspbian-stretch-download

Then you need to create an SD card with that OS and follow the guide for the iotedge above.

Mandur, 2021-09-17:
Sorry for the lack of updates in this topic. THis was fixed in !331, you can now keep those variable blanks

closed 2021-09-17 by Mandur