Azure/iotedge-lorawan-starterkit #285: Issue "7A7A7A00000014E2: join refused: no devices found matching join request"

maxstefaniv, 2021-02-04:
Hi everyone, I managed to set everything up, so now I have the seeduino rhf76-052 sending messages (trying to execute OTAA) and the gateway on a Raspberry pi receiving data and forwarding it.
To achieve this I followed the instructions and deployed Azure resources, set up the Raspberry and on the Arduino I loaded the sample code for OTAA as in [this repo](https://github.com/Azure/iotedge-lorawan-starterkit/blob/dev/Arduino/EU/GpsOtaaLora/GpsOtaaLora.ino)
And I did create a device manually on IoT Hub devices with the ID indicated in the sample code ant updated the desired in Device Twin.
[image]

I am getting the log from gateway as shown below:
```
JSON up: {"stat":{"time":"2021-02-04 09:25:16 GMT","rxnb":1,"rxok":1,"rxfw":1,"ackr":100.0,"dwnb":0,"txnb":0}}
INFO: [up] PUSH_ACK received in 0 ms
INFO: [down] PULL_ACK received in 1 ms

INFO: Received pkt from mote: 000014E2 (fcnt=31232)

JSON up: {"rxpk":[{"tmst":4055808411,"chan":2,"rfch":1,"freq":868.500000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":8.5,"rssi":-81,"size":23,"data":"AOIUAAAAAHq+4hQAAAB6enoFJ3O7Rf0="}]}
INFO: [up] PUSH_ACK received in 0 ms
INFO: [down] PULL_ACK received in 0 ms
INFO: [down] PULL_ACK received in 1 ms

INFO: Disabling GPS mode for concentrator's counter...
INFO: host/sx1301 time offset=(1612426663s:603301µs) - drift=-331µs
INFO: Enabling GPS mode for concentrator's counter.

##### 2021-02-04 09:25:46 GMT #####
### [UPSTREAM] ###
# RF packets received by concentrator: 2
# CRC_OK: 50.00%, CRC_FAIL: 50.00%, NO_CRC: 0.00%
# RF packets forwarded: 1 (23 bytes)
# PUSH_DATA datagrams sent: 2 (319 bytes)
# PUSH_DATA acknowledged: 100.00%
### [DOWNSTREAM] ###
# PULL_DATA sent: 3 (100.00% acknowledged)
# PULL_RESP(onse) datagrams received: 0 (0 bytes)
# RF packets sent to concentrator: 0 (0 bytes)
# TX errors: 0
# BEACON queued: 0
# BEACON sent so far: 0
# BEACON rejected: 0
### [JIT] ###
# SX1301 time (PPS): 4082692407
src/jitqueue.c:448:jit_print_queue(): INFO: [jit] queue is empty
### [GPS] ###
# GPS sync is disabled
##### END #####
```
And as a result I am getting data on the IoT Hub, after decoding from base64 the contents are:
**b'7A7A7A00000014E2: join request received'**
b'7A7A7A00000014E2: querying the registry for OTAA device'
b'error calling get device function api: Internal Server Error, status: InternalServerError, check the azure function log'
**b'7A7A7A00000014E2: join refused: no devices found matching join request'**
b'7A7A7A00000014E2: processing time: 00:00:00.0878072'
b'Statistic: {"stat":{"time":"2021-02-04 08:42:16 GMT","rxnb":1,"rxok":1,"rxfw":1,"ackr":100.0,"dwnb":0,"txnb":0}}'

Please help me figure out where should I look, because it is supposed to connect out of the box following the steps.
What should I do to communicate for example without authentication?

maxstefaniv, 2021-02-04:
Hopefully my last Issue related to this end to end SetUp:
So I managed to fix the previous issue, and after that TransmissionTestABP worked perfectly. Yey!
So I went for the OTAA and I got a strange messages which after decoding:
```
b'47AAC86800430028: join request received'
b'47AAC86800430028: querying the registry for OTAA device'
b'47AAC86800430028: using edgeHub local queue'
b'47AAC86800430028: getting device twin'
b'47AAC86800430028: done getting device twin'
b'47AAC86800430028: updating twin'
b'47AAC86800430028: twin updated'
b'47AAC86800430028: the provided RX2DataRate is not valid'
b'47AAC86800430028: device added to cache'
b'UDP: sending message with ID 1421, to 172.18.0.1:40970'
b'UDP: message sent with ID 1421'
b'UDP: packet with id 1421 successfully transmitted by the aggregator'
b'47AAC86800430028: processing time: 00:00:02.3748516'
b'47AAC86800430028: JoinAccept {"txpk":{"imme":false,"data":"IAu1GvlBjn+FlvQ4gq3ZCAg=","tmst":3387505307,"size":17,"freq":868.3,"rfch":0,"modu":"LORA","datr":"SF7BW125","codr":"4/5","powe":14,"ipol":true}}'
b'Statistic: {"stat":{"time":"2021-02-04 15:39:28 GMT","rxnb":1,"rxok":1,"rxfw":1,"ackr":100.0,"dwnb":1,"txnb":1}}'
b'Physical dataUp {"rxpk":[{"tmst":3415545371,"chan":1,"rfch":1,"freq":868.300000,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","lsnr":8.8,"rssi":-99,"size":14,"data":"gLnMCQIAAAAIZfdzRNs="}]}'
b'47AAC86800430028: device in cache'
b'47AAC86800430028: converted 16bit FCnt 0 to 32bit FCnt 0'
b'47AAC86800430028: down frame counter: 1'
b'47AAC86800430028: valid frame counter, msg: 0 server: 0'
b'47AAC86800430028: decoding with: DecoderValueSensor port: 8'
b'47AAC86800430028: sending message {"time":null,"tmms":0,"tmst":3415545371,"freq":868.3,"chan":1,"rfch":1,"stat":1,"modu":"LORA","datr":"SF7BW125","codr":"4/5","rssi":-99,"lsnr":8.8,"size":14,"data":{"value":0},"port":8,"fcnt":0,"rawdata":"MA==","eui":"47AAC86800430028","gatewayid":"rpi3eg","edgets":1612453176418} to hub'
b'47AAC86800430028: message \'{"value":0}\' sent to hub'
b'47AAC86800430028: checking cloud to device message for 00:00:00.5521692'
b'47AAC86800430028: done checking cloud to device message, found no message'
b'47AAC86800430028: sending a downstream message with ID 27F8'
b'47AAC86800430028: UnconfirmedDataDown {"txpk":{"imme":false,"data":"YLnMCQIgAQB8+TbA","tmst":3416545371,"size":12,"freq":868.3,"rfch":0,"modu":"LORA","datr":"SF7BW125","codr":"4/5","powe":14,"ipol":true}}'        
b'UDP: sending message with ID C586, to 172.18.0.1:40970'
b'UDP: message sent with ID C586'
b'UDP: packet with id C586 successfully transmitted by the aggregator'
```

I mean line 8 of the log: **the provided RX2DataRate is not valid'**
So The set up is the same, everything deployed through the button and after that functions are redeployed from the repo using VS.
I am in EU so that is the configuration and I am running the sample code.
The device twin does get updated its reported properties so as far as I understand it has successfully communicated with the cloud and got his DevAddr... From this point it should be sending messages as nothing happened into IotHub but there are no messages:
[image]
The only communication is from gateway (and I checked the device is running)

Also this is what I have on the device:
`  lora.setDeciveMode(LWOTAA);
  lora.setDataRate(DR0, EU868);`
And I did try to change it to DR5 -> obtaining an error also...

Could you please point me in the right direction heare??

ronniesa, 2021-02-05:
The deviceid: 47AAC86800430028 is correctly OTAA joining and sending the message. We need to look in to 'the provided RX2DataRate is not valid' as it strange. But the device correctly joined and sent a first message.

How are you monitoring the messages in IoT Hub?

maxstefaniv, 2021-02-05:
Hi @ronniesa well in a very ineffective way:
 I am saving all the messages from IoT Hub in blob storage getting for example this data on the storage:
 ```
{"EnqueuedTimeUtc":"2021-02-04T16:19:58.8260000Z","Properties":{},"SystemProperties":{"connectionDeviceId":"rpi3eg","connectionModuleId":"LoRaWanNetworkSrvModule","connectionAuthMethod":"{\"scope\":\"module\",\"type\":\"sas\",\"issuer\":\"iothub\",\"acceptingIpFilterRule\":null}","connectionDeviceGenerationId":"637479412847121814","enqueuedTime":"2021-02-04T16:19:58.8260000Z"},"Body":"U3RhdGlzdGljOiB7InN0YXQiOnsidGltZSI6IjIwMjEtMDItMDQgMTY6MTk6NTggR01UIiwicnhuYiI6MCwicnhvayI6MCwicnhmdyI6MCwiYWNrciI6MTAwLjAsImR3bmIiOjAsInR4bmIiOjB9fQ=="}
{"EnqueuedTimeUtc":"2021-02-04T16:20:28.8150000Z","Properties":{},"SystemProperties":{"connectionDeviceId":"rpi3eg","connectionModuleId":"LoRaWanNetworkSrvModule","connectionAuthMethod":"{\"scope\":\"module\",\"type\":\"sas\",\"issuer\":\"iothub\",\"acceptingIpFilterRule\":null}","connectionDeviceGenerationId":"637479412847121814","enqueuedTime":"2021-02-04T16:20:28.8150000Z"},"Body":"U3RhdGlzdGljOiB7InN0YXQiOnsidGltZSI6IjIwMjEtMDItMDQgMTY6MjA6MjggR01UIiwicnhuYiI6MCwicnhvayI6MCwicnhmdyI6MCwiYWNrciI6MTAwLjAsImR3bmIiOjAsInR4bmIiOjB9fQ=="}
{"EnqueuedTimeUtc":"2021-02-04T16:20:58.8240000Z","Properties":{},"SystemProperties":{"connectionDeviceId":"rpi3eg","connectionModuleId":"LoRaWanNetworkSrvModule","connectionAuthMethod":"{\"scope\":\"module\",\"type\":\"sas\",\"issuer\":\"iothub\",\"acceptingIpFilterRule\":null}","connectionDeviceGenerationId":"637479412847121814","enqueuedTime":"2021-02-04T16:20:58.8240000Z"},"Body":"U3RhdGlzdGljOiB7InN0YXQiOnsidGltZSI6IjIwMjEtMDItMDQgMTY6MjA6NTggR01UIiwicnhuYiI6MCwicnhvayI6MCwicnhmdyI6MCwiYWNrciI6MTAwLjAsImR3bmIiOjAsInR4bmIiOjB9fQ=="}
{"EnqueuedTimeUtc":"2021-02-04T16:21:09.1850000Z","Properties":{},"SystemProperties":{"connectionDeviceId":"rpi3eg","connectionModuleId":"LoRaWanNetworkSrvModule","connectionAuthMethod":"{\"scope\":\"module\",\"type\":\"sas\",\"issuer\":\"iothub\",\"acceptingIpFilterRule\":null}","connectionDeviceGenerationId":"637479412847121814","enqueuedTime":"2021-02-04T16:21:09.1850000Z"},"Body":"UGh5c2ljYWwgZGF0YVVwIHsicnhwayI6W3sidG1zdCI6MTYxMzI5NjgxMSwiY2hhbiI6MSwicmZjaCI6MSwiZnJlcSI6ODY4LjMwMDAwMCwic3RhdCI6MSwibW9kdSI6IkxPUkEiLCJkYXRyIjoiU0Y3QlcxMjUiLCJjb2RyIjoiNC81IiwibHNuciI6OS4yLCJyc3NpIjotOTYsInNpemUiOjIzLCJkYXRhIjoiQU9JVUFBQUFBSHErNGhRQUFBQjZlbnJ0dGlvVy9NND0ifV19"}
{"EnqueuedTimeUtc":"2021-02-04T16:21:09.2790000Z","Properties":{},"SystemProperties":{"connectionDeviceId":"rpi3eg","connectionModuleId":"LoRaWanNetworkSrvModule","connectionAuthMethod":"{\"scope\":\"module\",\"type\":\"sas\",\"issuer\":\"iothub\",\"acceptingIpFilterRule\":null}","connectionDeviceGenerationId":"637479412847121814","enqueuedTime":"2021-02-04T16:21:09.2790000Z"},"Body":"N0E3QTdBMDAwMDAwMTRFMjogam9pbiByZXF1ZXN0IHJlY2VpdmVk"}
{"EnqueuedTimeUtc":"2021-02-04T16:21:09.2790000Z","Properties":{},"SystemProperties":{"connectionDeviceId":"rpi3eg","connectionModuleId":"LoRaWanNetworkSrvModule","connectionAuthMethod":"{\"scope\":\"module\",\"type\":\"sas\",\"issuer\":\"iothub\",\"acceptingIpFilterRule\":null}","connectionDeviceGenerationId":"637479412847121814","enqueuedTime":"2021-02-04T16:21:09.2790000Z"},"Body":"N0E3QTdBMDAwMDAwMTRFMjogcXVlcnlpbmcgdGhlIHJlZ2lzdHJ5IGZvciBPVEFBIGRldmljZQ=="}
{"EnqueuedTimeUtc":"2021-02-04T16:21:09.2790000Z","Properties":{},"SystemProperties":{"connectionDeviceId":"rpi3eg","connectionModuleId":"LoRaWanNetworkSrvModule","connectionAuthMethod":"{\"scope\":\"module\",\"type\":\"sas\",\"issuer\":\"iothub\",\"acceptingIpFilterRule\":null}","connectionDeviceGenerationId":"637479412847121814","enqueuedTime":"2021-02-04T16:21:09.2790000Z"},"Body":"RGV2aWNlIGNhY2hlIGNsZWFyZWQ="}
{"EnqueuedTimeUtc":"2021-02-04T16:21:12.0300000Z","Properties":{},"SystemProperties":{"connectionDeviceId":"rpi3eg","connectionModuleId":"LoRaWanNetworkSrvModule","connectionAuthMethod":"{\"scope\":\"module\",\"type\":\"sas\",\"issuer\":\"iothub\",\"acceptingIpFilterRule\":null}","connectionDeviceGenerationId":"637479412847121814","enqueuedTime":"2021-02-04T16:21:12.0300000Z"},"Body":"N0E3QTdBMDAwMDAwMTRFMjogdXBkYXRpbmcgdHdpbg=="}
{"EnqueuedTimeUtc":"2021-02-04T16:21:12.1390000Z","Properties":{"iothub-message-schema":"twinChangeNotification"},"SystemProperties":{"connectionDeviceId":"7A7A7A00000014E2","connectionAuthMethod":"{\"scope\":\"device\",\"type\":\"sas\",\"issuer\":\"iothub\",\"acceptingIpFilterRule\":null}","connectionDeviceGenerationId":"637479478688859964","enqueuedTime":"2021-02-04T16:21:12.1390000Z"},"Body":"eyJBcHBTS2V5IjoiMkZEODI3NDkxOTNFOTQ1OEExOEVCNTA4QzhBMUFGOUQiLCJOd2tTS2V5IjoiNUFCNzE2ODc4NTE0NTI0QUMyQzg5RTlCNDg1QkYxQkYiLCJEZXZBZGRyIjoiMDNCQ0M1RDkiLCJGQ250RG93biI6MCwiRkNudFVwIjowLCJEZXZFVUkiOiI3QTdBN0EwMDAwMDAxNEUyIiwiTmV0SWQiOiIwMTAwMDAiLCJEZXZOb25jZSI6IkVEQjYiLCJSWDFEUk9mZnNldCI6bnVsbCwiUlgyRGF0YVJhdGUiOm51bGwsIlJYRGVsYXkiOm51bGx9"}
{"EnqueuedTimeUtc":"2021-02-04T16:21:12.1550000Z","Properties":{},"SystemProperties":{"connectionDeviceId":"rpi3eg","connectionModuleId":"LoRaWanNetworkSrvModule","connectionAuthMethod":"{\"scope\":\"module\",\"type\":\"sas\",\"issuer\":\"iothub\",\"acceptingIpFilterRule\":null}","connectionDeviceGenerationId":"637479412847121814","enqueuedTime":"2021-02-04T16:21:12.1550000Z"},"Body":"N0E3QTdBMDAwMDAwMTRFMjogdHdpbiB1cGRhdGVk"}
[... cut ...]
b'7A7A7A00000014E2: JoinAccept {"txpk":{"imme":false,"data":"IAsfGNxMxnb33frawkriMZg=","tmst":1618296811,"size":17,"freq":868.3,"rfch":0,"modu":"LORA","datr":"SF7BW125","codr":"4/5","powe":14,"ipol":true}}'
b'Statistic: {"stat":{"time":"2021-02-04 16:21:28 GMT","rxnb":1,"rxok":1,"rxfw":1,"ackr":100.0,"dwnb":1,"txnb":1}}'
```

So I totally agree with you, the device achieves the first message transimission (pruf is that the twin is also updated) but after that I am not getting any new messages from the device...

Mandur, 2021-02-05:
The 'message b'47AAC86800430028: the provided RX2DataRate is not valid' is a logging error in our current Code base, I will take a note to fix it asap. Thank you for reporting it! It should however not have any impact on the logic

ronniesa, 2021-02-05:
I recommend installing iot hub device explorer so you can directly monitor a specific device message: https://docs.microsoft.com/en-us/azure/iot-pnp/howto-use-iot-explorer 

You should create a consumer group in iot hub just for this to avoid competing for messages by other consumers (if you have any)

What lora device are you using? The seeedstudio arduino? 
If yes then you should monitor the serial output and see if the device is sending messages continusly e.g 0,1,2..... then check the log as per above that the NetworkServer is getting the message and sending upstream to iot hub b'47AAC86800430028: message \'{"value":0}\' sent to hub' where the value should increase and check with iot hub device exploer the messages for device 47AAC86800430028

maxstefaniv, 2021-02-07:
Hello @ronniesa  thank you for your suggestion.
I gave it a chance this weekend and found part of my issue. Somehow the Seeeduino LoRaWAN does not get GPS signal, so the code that you have in the sample does not send messages and that is shy previously I was getting only the first update message... 

After modifying the Arduino code, it seems to kind of work. In case of no GPS I am sending a string "NoGPS" so in the Hub I receive the message as follows:
`{"EnqueuedTimeUtc":"2021-02-07T18:18:18.7440000Z","Properties":{},"SystemProperties":{"connectionDeviceId":"7A7A7A00000014E2","connectionAuthMethod":"{\"scope\":\"device\",\"type\":\"sas\",\"issuer\":\"iothub\",\"acceptingIpFilterRule\":null}","connectionDeviceGenerationId":"637483159898912923","contentType":"application/json","contentEncoding":"utf-8","enqueuedTime":"2021-02-07T18:18:18.7440000Z"},"Body":{"time":null,"tmms":0,"tmst":2762465340,"freq":868.1,"chan":0,"rfch":1,"stat":1,"modu":"LORA","datr":"SF12BW125","codr":"4/5","rssi":-36,"lsnr":10.2,"size":18,"data":{"error":"'DecoderGpsSensor' decoder not found"},"port":8,"fcnt":0,"rawdata":"Tm9HUFM=","eui":"7A7A7A00000014E2","gatewayid":"rpi3eg","edgets":1612721898502}}
`
So just left to understand where that decoding is being done and why it is not executed in my case... any hint?

But anyhow I still have the issue of the edgeHub unsolved because then I `iotedge logs edgeHub` I get the same persistent error:
````
<6> 2021-02-07 18:17:50.497 +00:00 [INF] - Entering periodic task to reauthenticate connected clients
<6> 2021-02-07 18:17:50.498 +00:00 [INF] - Unable to authenticate client 7A7A7A00000014E2 with cached service identity 7A7A7A00000014E2. Resyncing service identity...
<4> 2021-02-07 18:17:51.040 +00:00 [WRN] - Error while refreshing the service identity for 7A7A7A00000014E2
Microsoft.Azure.Devices.Edge.Hub.CloudProxy.DeviceScopeApiException: Message: Error getting device scope result from IoTHub, HttpStatusCode: Unauthorized, Content: {"Message":"ErrorCode:IotHubUnauthorizedAccess;Unauthorized","ExceptionMessage":"Tracking ID:89fac4095d65419e89a7e251e89d93c4-G:23-TimeStamp:02/07/2021 18:17:51"}
   at Microsoft.Azure.Devices.Edge.Hub.CloudProxy.DeviceScopeApiClient.GetIdentitiesInScope(Uri uri) in /home/vsts/work/1/s/edge-hub/src/Microsoft.Azure.Devices.Edge.Hub.CloudProxy/DeviceScopeApiClient.cs:line 131
   at Microsoft.Azure.Devices.Edge.Hub.CloudProxy.DeviceScopeApiClient.GetIdentitiesInScopeWithRetry(Uri uri) in /home/vsts/work/1/s/edge-hub/src/Microsoft.Azure.Devices.Edge.Hub.CloudProxy/DeviceScopeApiClient.cs:line 99
   at Microsoft.Azure.Devices.Edge.Hub.CloudProxy.ServiceProxy.GetServiceIdentity(String deviceId) in /home/vsts/work/1/s/edge-hub/src/Microsoft.Azure.Devices.Edge.Hub.CloudProxy/ServiceProxy.cs:line 29
   at Microsoft.Azure.Devices.Edge.Hub.Core.DeviceScopeIdentitiesCache.RefreshServiceIdentity(String id) in /home/vsts/work/1/s/edge-hub/src/Microsoft.Azure.Devices.Edge.Hub.Core/DeviceScopeIdentitiesCache.cs:line 72
<6> 2021-02-07 18:18:50.406 +00:00 [INF] - Processing pending subscriptions for 7A7A7A00000014E2
<6> 2021-02-07 18:18:50.599 +00:00 [INF] - Updated reported properties for 7A7A7A00000014E2
```
And about this error I honestly don't know where to start

gerfen, 2021-02-07:
Hi @maxstefaniv,  

I'm facing the same issue -- it appears to be an issue with the Azure IoT Hub.  If you read through #286, you'll see @ronniesa suggested that I set ENABLE_GATEWAY = false for the LoraWanNetworkSrvModule via the Azure IoT Hub portal.  It's a temporary workaround until the issue can be resolved.  So far it has worked well for me.

maxstefaniv, 2021-02-07:
Hello @gerfen TY a lot, seems to work fine!! 

Yet the message is not decoded, where is decoding done? 
in the data on IoT hub I am getting **"data":{"error":"'DecoderGpsSensor' decoder not found"}**
Sad story I am tumbling from error into error XD

gerfen, 2021-02-07:
I am decoding at the edge using the sample decoding code from the starter kit as a start.  If you haven't done so you can read about decoding [here](https://github.com/Azure/iotedge-lorawan-starterkit/blob/dev/Docs/devguide.md).  you'll add your decoder code to the sample module and then will publish that to your own Docker container registry.  After the module has been published, you can push it to the gateway through the same page I referenced above in the Azure IoT Hub portal.

ronniesa, 2021-02-07:
The DecoderGpsSensor is not in the engine as is specific for the sample. Not sure we have an example for that, tomorrow I will check with my colleagues. 

I assume you are using this in your arduino https://github.com/Azure/iotedge-lorawan-starterkit/blob/dev/Arduino/EU/GpsOtaaLora/GpsOtaaLora.ino 

As we send the gps value as a string with ":" to separate lat long you can keep the default decoder "DecoderValueSensor"

We have done a more complex GPS encoding (optimize bandwidth) but not sure we can share It I will check.

gerfen, 2021-02-07:
And to more fully answer your question... I typically grab the raw payload from the LoRaWanNetworkSrvModule logs and write unit tests (I'm using xUnit) to test the decoding prior to adding my method to the sample decoder.

For example here's a decoder for a Strega Smart Valve:
```
    public class SmartValveDecoder
    {
        public static string Decode(string devEui, byte[] payload, uint fport)
        {
            var hex = BitConverter.ToString(payload);
            var hexValues = hex.Split('-').Select(x=>(Convert.ToInt16(x)-30).ToString()).ToArray();

            return JsonConvert.SerializeObject(DecodePeriodicUplink(hexValues, payload));

        }

        private static Dictionary<string, object> DecodePeriodicUplink(string[] hexValues, byte[] payload)
        {
            //(value & checkBit) == checkBit;
            //var valveState = (payload[4] & 0x0001) == 0x0001;
            //var enlosureState = (payload[4] & 0x0002) == 0x0002;
            //var valveOpen = (Convert.ToInt32($"{hexValues[4]}") & 0x0001) == 0x0001;
            //var enclosureOpen = (Convert.ToInt32($"{hexValues[4]}") & 0x0002) == 0x0002;
            var dictionary = new Dictionary<string, object>
            {
                { "battery", Convert.ToInt32($"{hexValues[0]}{hexValues[1]}{hexValues[2]}{hexValues[3]}")},
                { "enclosureIsOpen",  IsBitSet(hexValues[4], 0X0002) },
                { "valveIsOpen",  IsBitSet(hexValues[4], 0X0001) }
            };

            return dictionary;
        }

        private static bool IsBitSet(string value, int checkBit)
        {
            return IsBitSet(Convert.ToInt32(value), checkBit);
        }

        private static bool IsBitSet(int value, int checkBit)
        {
            return (value & checkBit) == checkBit;
        }
    }
```

Here's a test:
```
    public class SmartValveDecoderTests : DecodingTests
    {
        public SmartValveDecoderTests(ITestOutputHelper output) : base(output)
        {
        }

        protected override string DecodeData(byte[] data, string deviceEui, uint port)
        {
            return SmartValveDecoder.Decode(deviceEui, data, port);
        }
        [Fact]
        public void BasicDecodingTests()
        {
               var result = DecodeData("MzYxNzI=", port: 4);
                
       }
}
```
Here's the DecodingTests base class
```
  public abstract class DecodingTests
    {
        protected ITestOutputHelper Output { get; set; }

        protected DecodingTests(ITestOutputHelper output)
        {
            Output = output;
        }

        protected virtual Dictionary<string, object> DecodeData(string data, string deviceEui = "00000000-0000-0000-0000-000000000000", uint port = 1)
        {
            Output.WriteLine($"Data: {data}");
            var bytes = Convert.FromBase64String(data);

            var hex = BitConverter.ToString(bytes);
            //var hexValues = hex.Split('-');
            Output.WriteLine($"hex: {hex}");

            var json = DecodeData(bytes, deviceEui, port);
            Output.WriteLine($"JSON: {json}");

            return JsonConvert.DeserializeObject<Dictionary<string, object>>(json);
        }

        protected Tuple<string, string> ReverseKeys(string appEui, string appKey)
        {
            return new Tuple<string, string>(string.Concat(appEui.Reverse()), string.Concat(appKey.Reverse()));
        }

        protected abstract string DecodeData(byte[] data,string deviceEui, uint port);
    }
```

And finally here is the Decoder integrated in to the decoding module:

```
 using StregaSmartValveDecoder = Evolution.IoT.LoraWan.Decoders.Strega.SmartValveDecoder;
    internal static class LoraDecoders
    {
        // rest ode code elided for clarity...
        private static string DecodeStregaSmartValve(string devEUI, byte[] payload, uint fport)
        {
            return StregaSmartValveDecoder.Decode(devEUI, payload, fport);
        }
   }
```