Azure/iotedge-lorawan-starterkit #288: [main] failed to start the concentrator (AIOT-ILRA01-A10-CA40232EU)

maxstefaniv, 2021-03-03:
## Expected Behavior
LoRaWanPktFwdModule should start.

## Current Behavior
When `iotedge logs LoRaWanPktFwdModule`
getting:
error [main] failed to start the concentrator

## Steps to Reproduce

1. I already have set up an END-2-END with a Raspberry Pi3 (has some issues, but I was helped with everything)
2. Currently, I am trying to repeat the set-up on a AIOT-ILRA01-A10-CA40232EU
3. I installed UBUNTU Server 18.04.5 (including updates) and followed the steps on [Install or uninstall Azure IoT Edge for Linux](https://docs.microsoft.com/en-us/azure/iot-edge/how-to-install-iot-edge?view=iotedge-2018-06)
4. The Connection string I am using is the same that worked on my first set up.

## Context (Environment)

### Device (Host) Operating System
Ubuntu 18.04.5

### LoRaWAN Module Version
loraedge/lorawanpktfwdmodule:1.0.5

## Logs

When checking for modules `sudo iotedge list`:
NAME                     STATUS           DESCRIPTION                CONFIG
LoRaWanNetworkSrvModule  running          Up 42 minutes              loraedge/lorawannetworksrvmodule:1.0.5
LoRaWanPktFwdModule      failed           Failed (1) 27 minutes ago  loraedge/lorawanpktfwdmodule:1.0.5
edgeAgent                running          Up 42 minutes              mcr.microsoft.com/azureiotedge-agent:1.0.9.5
edgeHub                  running          Up 42 minutes              mcr.microsoft.com/azureiotedge-hub:1.0.9.5
[image]
Checking the logs `iotedge logs LoRaWanPktFwdModule`:
EU region detected.
Accessing concentrator reset pin through GPIO7...
sh: echo: I/O error
./reset_lgw.sh: 30: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent
./reset_lgw.sh: 33: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
./reset_lgw.sh: 34: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/value: Directory nonexistent
./reset_lgw.sh: 37: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio7/direction: Directory nonexistent
No SPI Dev version detected in environment variables on x86, defaulting to SPI Dev 2
*** Beacon Packet Forwarder for Lora Gateway ***
Version: 4.0.1
*** Lora concentrator HAL library version info ***
Version: 5.0.1;
***
INFO: Little endian host
INFO: found global configuration file global_conf.json, parsing it
INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
INFO: lorawan_public 1, clksrc 1
INFO: LBT is disabled
INFO: antenna_gain 0 dBi
INFO: Configuring TX LUT with 16 indexes
INFO: radio 0 enabled (type SX1257), center frequency 867500000, RSSI offset -166.000000, tx enabled 1, tx_notch_freq 129000
INFO: radio 1 enabled (type SX1257), center frequency 868500000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7
INFO: FSK channel> radio 1, IF 300000 Hz, 125000 Hz bw, 50000 bps datarate
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000000
INFO: server hostname or IP address is configured to "localhost"
INFO: upstream port is configured to "1680"
INFO: downstream port is configured to "1680"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: found local configuration file local_conf.json, parsing it
INFO: redefined parameters will overwrite global parameters
INFO: local_conf.json does not contain a JSON object named SX1301_conf
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000101
INFO: server hostname or IP address is configured to "172.17.0.1"
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
ERROR: [main] failed to start the concentrator

## Additional Information
I did look through some issues and this looks like a recurrent error but for US config. #gerfen had some similar issues.
In other posts it was suggested to use Ubuntu server 16.04, but Microsoft requires 18.04 for edge device. I am not sure how to proceed here.

All the resources were deployed with the template from the git repo.

Mandur, 2021-03-09:
Apologies for the late answers, I missed the issue. 
Could you please try to set RESET_PIN variable to 5 and let me know if that helps?

maxstefaniv, 2021-03-11:
Hello @Mandur, I did the change:
[image]
then restarted modules: `sudo systemctl restart iotedge` and rebooted the gateway.

Still getting the same Erorr but now for GPIO05:
```
EU region detected.
Accessing concentrator reset pin through GPIO5...
sh: echo: I/O error
./reset_lgw.sh: 30: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio5/direction: Directory nonexistent
./reset_lgw.sh: 33: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio5/value: Directory nonexistent
./reset_lgw.sh: 34: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio5/value: Directory nonexistent
./reset_lgw.sh: 37: ./reset_lgw.sh: cannot create /sys/class/gpio/gpio5/direction: Directory nonexistent
No SPI Dev version detected in environment variables on x86, defaulting to SPI Dev 2
*** Beacon Packet Forwarder for Lora Gateway ***
Version: 4.0.1
*** Lora concentrator HAL library version info ***
Version: 5.0.1;
***
INFO: Little endian host
INFO: found global configuration file global_conf.json, parsing it
INFO: global_conf.json does contain a JSON object named SX1301_conf, parsing SX1301 parameters
INFO: lorawan_public 1, clksrc 1
INFO: LBT is disabled
INFO: antenna_gain 0 dBi
INFO: Configuring TX LUT with 16 indexes
INFO: radio 0 enabled (type SX1257), center frequency 867500000, RSSI offset -166.000000, tx enabled 1, tx_notch_freq 129000
INFO: radio 1 enabled (type SX1257), center frequency 868500000, RSSI offset -166.000000, tx enabled 0, tx_notch_freq 0
INFO: Lora multi-SF channel 0>  radio 1, IF -400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 1>  radio 1, IF -200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 2>  radio 1, IF 0 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 3>  radio 0, IF -400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 4>  radio 0, IF -200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 5>  radio 0, IF 0 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 6>  radio 0, IF 200000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora multi-SF channel 7>  radio 0, IF 400000 Hz, 125 kHz bw, SF 7 to 12
INFO: Lora std channel> radio 1, IF -200000 Hz, 250000 Hz bw, SF 7
INFO: FSK channel> radio 1, IF 300000 Hz, 125000 Hz bw, 50000 bps datarate
INFO: global_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000000
INFO: server hostname or IP address is configured to "localhost"
INFO: upstream port is configured to "1680"
INFO: downstream port is configured to "1680"
INFO: downstream keep-alive interval is configured to 10 seconds
INFO: statistics display interval is configured to 30 seconds
INFO: upstream PUSH_DATA time-out is configured to 100 ms
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
INFO: found local configuration file local_conf.json, parsing it
INFO: redefined parameters will overwrite global parameters
INFO: local_conf.json does not contain a JSON object named SX1301_conf
INFO: local_conf.json does contain a JSON object named gateway_conf, parsing gateway parameters
INFO: gateway MAC address is configured to AA555A0000000101
INFO: server hostname or IP address is configured to "172.17.0.1"
INFO: packets received with a valid CRC will be forwarded
INFO: packets received with a CRC error will NOT be forwarded
INFO: packets received with no CRC will NOT be forwarded
ERROR: [main] failed to start the concentrator
```

ronniesa, 2021-03-11:
For that gateway that we also have in our lab you need this settings only on the module:

[image]

maxstefaniv, 2021-03-11:
Hello @ronniesa that is exactly what I have right now, as I undid the steps that did not help me.
[image]

ronniesa, 2021-03-11:
Some AEON have the spi disabled one way is to do this https://github.com/dcleri/spi-enable I don't recall if I we did on this particular model or not....

maxstefaniv, 2021-03-11:
@ronniesa got the files on the device and installed it. Didn't seem to help.
I will try to redo all steps from scratch, might help.

Mandur, 2021-03-11:
Looks like you have gpio enabled as ls /sys/class/gpio return some results... I just think the reset pin is not the normal one (5). Do you have maybe a readme or manual given to you by the manufacturer where you could find this information

maxstefaniv, 2021-03-11:
I have sent a request to support, hopefully they will get back to me soon. Thank you! Hopefully will solve it soon.

ronniesa, 2021-03-11:
One trick is to install the semtech packet forwarder directly on the OS to see it works once it is up and running you remove it and set the settings on the container and you are good to go.

maxstefaniv, 2021-03-16:
This solution works. [Issue 245 solution](https://github.com/Azure/iotedge-lorawan-starterkit/issues/245#issuecomment-660722255). GOing into Ubuntu Server 16.04.

SPI reset, as indicated by ronniesa in https://github.com/Azure/iotedge-lorawan-starterkit/issues/288#issuecomment-796642640 for Ubuntu Server 18.04 didn't work for me but for Ubuntu Server 16.04 it looks like id worked.

Mandur, 2021-03-23:
I will close the issue for now,feel free to reopen in case that changed

closed 2021-03-23 by Mandur