Azure/iotedge-lorawan-starterkit #289: Running LoRaWanNetworkSrvModule  in IoT Edge Simulator throws error on startup (listen tcp 172.17.0.1:1680: can't bind on the specified endpoint)

marxxxx, 2021-04-14:
## Expected Behavior
I want to build on the functionality provided by LoRaWanPktFwdModule and LoRaWanNetworkSrvModule and extend the deployment with my custom module.

In order to do so, i used the deployment template in the solution containing the above mentioned docker modules along with their configuration.

I would like to run this setup in IoT Edge Simulator in Visual Studio 2019 so i can test and debug my custom module along with the provided modules.

## Current Behavior
Currently on startup, the LoRaWanNetworkSrvModule throws this error:
`ERROR: for LoRaWanNetworkSrvModule  Cannot start service LoRaWanNetworkSrvModule: Ports are not available: listen tcp 172.17.0.1:1680: can't bind on the specified endpoint`

When i modify the HostIp setting in the createOptions of the container and use 127.0.0.1 the container can start.
However, LoRaWanPktFwdModule tries to send to the above mentioned IP Address, so i think this workaround would not work in my setup.

## Steps to Reproduce
1. Create new empty Azure IoT Edge Project in Visual Studio
2. Replace the content of the deployment manifest with the content below
3. Setup the Azure IoT Edge Simulator by providing the IoT Hub Connection String of an Edge Device
4. Right-click on the project in Visual Studio and select "Build and run IoT Edge Modules in Simulator"
5. Watch the error in the Output window

```
{
  "$schema-template": "2.0.0",
  "modulesContent": {
    "$edgeAgent": {
      "properties.desired": {
        "schemaVersion": "1.0",
        "runtime": {
          "type": "docker",
          "settings": {
            "loggingOptions": "",
            "minDockerVersion": "v1.25"
          }
        },
        "systemModules": {
          "edgeAgent": {
            "type": "docker",
            "settings": {
              "image": "mcr.microsoft.com/azureiotedge-agent:1.0.9.5",
              "createOptions": "{}"
            }
          },
          "edgeHub": {
            "type": "docker",
            "settings": {
              "image": "mcr.microsoft.com/azureiotedge-hub:1.0.9.5",
              "createOptions": "{ \"HostConfig\": {   \"PortBindings\": {\"8883/tcp\": [  {\"HostPort\": \"8883\" }  ], \"443/tcp\": [ { \"HostPort\": \"443\" } ], \"5671/tcp\": [ { \"HostPort\": \"5671\"  }] } }}"
            },
            "env": {
              "OptimizeForPerformance": {
                "value": "false"
              },
              "mqttSettings__enabled": {
                "value": "false"
              },
              "httpSettings__enabled": {
                "value": "false"
              },
              "TwinManagerVersion": {
                "value": "v2"
              }
            },
            "status": "running",
            "restartPolicy": "always"
          }
        },
        "modules": {
          "LoRaWanNetworkSrvModule": {
            "type": "docker",
            "settings": {
              "image": "loraedge/lorawannetworksrvmodule:1.0.5",
              "createOptions": "{\"ExposedPorts\": { \"1680/udp\": {}}, \"HostConfig\": {  \"PortBindings\": {\"1680/udp\": [  { \"HostPort\": \"1680\", \"HostIp\":\"172.17.0.1\" } ]}}}"
            },
            "version": "1.0",
            "env": {
              "LOG_LEVEL": {
                "value": "1"
              },
              "LOG_TO_HUB": {
                "value": "false"
              }
            },
            "status": "running",
            "restartPolicy": "always"
          },
          "LoRaWanPktFwdModule": {
            "type": "docker",
            "settings": {
              "image": "loraedge/lorawanpktfwdmodule:1.0.5",
              "createOptions": "  {\"HostConfig\": {\"NetworkMode\": \"host\", \"Privileged\": true },  \"NetworkingConfig\": {\"EndpointsConfig\": {\"host\": {} }}}"
            },
            "env": {
              "RESET_PIN": {
                "value": "1234"
              },
              "REGION": {
                "value": "EU"
              },
              "SPI_SPEED": {
                "value": "8"
              },
              "SPI_DEV": {
                "value": "2"
              }
            },
            "version": "1.0",
            "status": "running",
            "restartPolicy": "always"
          }
        }
      }
    },
    "$edgeHub": {
      "properties.desired": {
        "schemaVersion": "1.0",
        "routes": {
          "route": "FROM /* INTO $upstream"
        },
        "storeAndForwardConfiguration": {
          "timeToLiveSecs": 7200
        }
      }
    },
    "LoRaWanNetworkSrvModule": {
      "properties.desired": {
        "schemaVersion": "1.0",
        "FacadeServerUrl": "https://XXXX.azurewebsites.net/api/",
        "FacadeAuthCode": "XXXX"
      }
    }
  }
}
```

## Context (Environment)

### Device (Host) Operating System
Windows 10

### Architecture
x64

### Container Operating System
Linux containers

### LoRaWAN Module Version
1.0.5

#### Docker
```
Client: Docker Engine - Community
 Cloud integration: 1.0.9
 Version:           20.10.5
 API version:       1.41
 Go version:        go1.13.15
 Git commit:        55c4c88
 Built:             Tue Mar  2 20:14:53 2021
 OS/Arch:           windows/amd64
 Context:           default
 Experimental:      true

Server: Docker Engine - Community
 Engine:
  Version:          20.10.5
  API version:      1.41 (minimum version 1.12)
  Go version:       go1.13.15
  Git commit:       363e9a8
  Built:            Tue Mar  2 20:15:47 2021
  OS/Arch:          linux/amd64
  Experimental:     false
 containerd:
  Version:          1.4.3
  GitCommit:        269548fa27e0089a8b8278fc4fc781d7f65a939b
 runc:
  Version:          1.0.0-rc92
  GitCommit:        ff819c7e9184c13b7c2607fe6c30ae19403a7aff
 docker-init:
  Version:          0.19.0
  GitCommit:        de40ad0
```

## Logs
```
edgeHub image version is ignored in solution mode.
Stopping LoRaWanPktFwdModule ... 
Stopping edgeHubDev          ... 
Stopping LoRaWanPktFwdModule ... done
Stopping edgeHubDev          ... done
Removing LoRaWanPktFwdModule     ... 
Removing LoRaWanNetworkSrvModule ... 
Removing edgeHubDev              ... 
Removing LoRaWanPktFwdModule     ... done
Removing edgeHubDev              ... done
Removing LoRaWanNetworkSrvModule ... done
Network azure-iot-edge-dev is external, skipping
Pulling edgeHubDev ... 
Pulling edgeHubDev ... pulling from azureiotedge-hub
Pulling edgeHubDev ... digest: sha256:52798256b3973378e7...
Pulling edgeHubDev ... status: image is up to date for m...
Pulling edgeHubDev ... done
Creating edgeHubDev ... 
Creating edgeHubDev ... done
Creating LoRaWanNetworkSrvModule ... 
Creating LoRaWanPktFwdModule     ... 
Creating LoRaWanNetworkSrvModule ... error
ERROR: for LoRaWanNetworkSrvModule  Cannot start service LoRaWanNetworkSrvModule: Ports are not available: listen tcp 172.17.0.1:1680: can't bind on the specified endpoint
Creating LoRaWanPktFwdModule     ... done
ERROR: for LoRaWanNetworkSrvModule  Cannot start service LoRaWanNetworkSrvModule: Ports are not available: listen tcp 172.17.0.1:1680: can't bind on the specified endpoint
Encountered errors while bringing up the project.
```

Mandur, 2021-04-17:
Hello @marxxxx ,
It looks like port 1680 is already taken by something on your current machine, don't you have for example the nwk srv docker container already running?

You could try setting the values "serv_port_up": and "serv_port_down" in the [local_conf.json](https://github.com/Azure/iotedge-lorawan-starterkit/blob/dev/LoRaEngine/modules/LoRaWanPktFwdModule/local_conf.json) with the value on the port where the network server is listening within the gateway_conf section. Let us know if that works!

The NETWORK_SERVER seems indeed not to be working as expected, we will investigate in what happened there, thank you for submitting the issue.

marxxxx, 2021-04-21:
Thank you for your comment, @Mandur !

That the port was taken was also my first guess but checking with TCPView this is not the case:
[image]

And if this port was taken it would also not work if i change the HostIp value to "127.0.0.1" but it does.

So my guess is, that the Ip-Address 172.17.0.1 is available when deployed on a real device but can't be used on my development machine. Maybe some docker networking setting i am missing?

Can i edit the values in local_conf.json also when using in the existing container image loraedge/lorawanpktfwdmodule:1.0.5 ?
I suppose i would have to build the module from source with the updated configuration and publish the image myself?

So changing the address the package forwarder sends it's messages by configuring NETWORK_SERVER would definitely help a lot. Did you have chance to look into the issue? Maybe any hint when where might be a fix available?

Thank you very much for your help!

Mandur, 2021-04-22:
Ok, I am starting to better understand the issue. As you are running on Windows, may I ask why you are not starting the network server from outside the iot edge as suggested [here](https://github.com/Azure/iotedge-lorawan-starterkit/blob/dev/Docs/devguide.md#debugging-in-visual-studio-outside-of-iot-edge-and-docker)?  The path you are trying to follow have not been tested for quite some time and is not officially supported. Would that work for you to run the Network server in Visual Studio?

We are currently busy with some other work, but I can try to have a look next week on the issue above.

marxxxx, 2021-04-28:
@Mandur Thank you so far. We want to treat the current modules as "black box" as we are using them only for a prototyping/PoC phase. Our plan would be to extend the functionality by routing the traffic generated by LoRaWanNetworkSrvModule to our custom module via IoT Edge Hub, process it and then send to IoT Hub. This scenario would not be possible when we run Network Server outside the Edge Runtime to my understanding.