Azure/iotedge-lorawan-starterkit #291: AU915 Support

Radio-Bob, 2021-05-03:
**Is your feature request related to a problem? Please describe.**
Support of AU915 frequency range

**Describe the solution you'd like**
LoRa starter kit should support AU915 freq range

**Describe alternatives you've considered**
N/A

**Additional context**
Requested by Radio-Bob

Mandur, 2021-05-05:
Hello @Radio-Bob ,

Thank you for raising this issue and your interest in the kit. We actually did a quick poc on the AU915 some time ago. It is not officially supported/tested but we have a [branch](https://github.com/Azure/iotedge-lorawan-starterkit/tree/mik-dev-add-au) that should enable some part of it (ABP, upstream messages only). At the moment there is no plan to officially support it.
What is your use case? Would the current support be already enough for you or do you need advanced features like C2D messages,...?

ElliotWood, 2021-06-23:
Yes also interested here, C2D messages would be ideal ;)