Azure/iotedge-lorawan-starterkit #292: Periodical loss of connection on devices (+- 24H)

maxstefaniv, 2021-05-10:
## Expected Behavior
Devices will report telemetry non stop, reaching IoT Hub with no issues.

## Current Behavior
After Set Up devices Reported properly for around 24 hours after that wtarted displaying:
`could not send message to IoTHub/Edge with error: The operation timed out.`
Unless iotEdge is rebooted, all tevices will report the same message.

## Steps to Reproduce

1. Deploy the project as it is (gateway device AIOT-ILRA01) (Devices ELSYS Co2 and Netvox R311W)
2. Add additional decoder (should not be an issue as this module only decodes the message coming from the device as telemetry.)
3. Wait 24h

## Context (Environment)

### Device (Host) Operating System
Ubuntu 16.04

### Architecture
amd64

### LoRaWAN Module Version
```
LoRaWanNetworkSrvModule  running          Up 2 hours       loraedge/lorawannetworksrvmodule:1.0.5
LoRaWanPktFwdModule      running          Up 2 hours       loraedge/lorawanpktfwdmodule:1.0.5
edgeAgent                running          Up 2 hours       mcr.microsoft.com/azureiotedge-agent:1.0.9.5
edgeHub                  running          Up 2 hours       mcr.microsoft.com/azureiotedge-hub:1.0.9.5
```

## Logs
Attached support_bundle.zip
[support_bundle.zip](https://github.com/Azure/iotedge-lorawan-starterkit/files/6451463/support_bundle.zip)

ronniesa, 2021-05-11:
@maxstefaniv does this happens only after deploying a new decoder or every time after 24h?

maxstefaniv, 2021-05-11:
@Mandur Hello, yeah "ENABLE_GATEWAY to false" was already set otherwise I was getting an error, that was a fix I already scaned from existing issues.

@ronniesa It happens everytime after 24 hours.

Mandur, 2021-05-12:
@maxstefaniv When I look at your docker inspect logs, I cannot see the environment variable ENABLE_GATEWAY set to false. May I ask to quickly double check. You can run 
`docker exec LoRaWanNetworkSrvModule bash -c printenv |grep ENABLE_GATEWAY`
(I am just double checking as it solved the issue for me)

I have a device that is running to try to get a repro, but at the moment I am failing to reproduce the problem, I think it might be benefical to have a call to look at the problem more in depth

maxstefaniv, 2021-05-12:
@Mandur Sure I executed it. 
[image]

When are you available. I can add you to my Teams account and we can have a call.

Mandur, 2021-05-12:
Sure @maxstefaniv just add me, we can sync add hoc!

Mandur, 2021-05-12:
Following the discussion we are testing a new version with updated client librairies to see if that fix the issue

maxstefaniv, 2021-05-13:
Applied update LoRaWanNetworkSrvModule from 1.0.5 to 1.0.5.1 on test edge, and it seems to help. If works on Prod edge as well, will close issue.

closed 2021-05-16 by Mandur

Mandur, 2021-05-16:
As it seems our changes solved the issue, we merged this into dev. This fix will be be included in the next lorawan release.
Feel free to reopen if there are any more issues you face with it.

Thank you for raising this!

maxstefaniv, 2021-05-16:
Effectively I was monitoring the situation through this weekend on both devices and there were no errors. All devices reauthenticate with gateways and there are no drops. This solution works. Thank you very much @Mandur.

PR #294 (2021-05-16): Fix #293 #292
Fix #293
- An invalid max message caused the gateway to enter the process loop, due to wrong exception handling, the processing was stuck in an infinite loop. Added test case with the poisonous message.
Fix #292
- The former device SDK caused the devices to disconnect after 24 hours. Upgrading to the latest version seems to have risolved the issue