Azure/iotedge-lorawan-starterkit #310: Incorrect data type for the tmms property

ducphuc, 2021-06-27:
## Expected Behavior
The `tmms` property is defined by the Semtech packet forwarder protocol to be "GPS time of pkt RX, number of milliseconds since 06.Jan.1980".  The data type should be `ulong`.

## Current Behavior
Current values for the `tmms` property will overflow the `uint` data type used in the LoRaWAN Starter Kit.

## Steps to Reproduce

1. Enable GPS sync in a packet forwarder instance based on the Semtech packet forwarder implementation in GitHub
2. Send a _rxpk_ packet from the packet forwarder to the LoRaWan Network Server

## Context (Environment)

### Device (Host) Operating System
Windows 10 Enterprise Insider Dev (21390.2025)

### Architecture
amd64

### Container Operating System
EFLOW (Azure IoT Edge for Linux on Windows)

### LoRaWAN Module Version
v1.0.5

#### Docker
20.10.7

## Logs
2021-06-26 17:21:01.892 UDP: failed to dispatch messages: Error converting value 1308788479556 to type 'System.UInt32'. Path 'rxpk[0].tmms', line 1, position 94.

## Additional Information

Mandur, 2021-07-06:
Hello @ducphuc thank you for reporting this. indeed looks like that was wrongly interpreted from us from the specs, we will provide a fix.

Mandur, 2021-10-13:
We have decided to release a new major version with usage of the LoRa Basics station instead of the packet forwarder. This change will address the above error, you can follow the progress on this user story in dev and #385. Feel free to reopen this issue if this is not clear

closed 2021-10-13 by Mandur