Azure/iotedge-lorawan-starterkit #353: Support for PackageForwarder

roel4ez, 2021-09-22:
> description needed

gerfen, 2021-10-07:
I can confirm that the basic station code from the  lbs_experimental branch works on an AAEON AIOT-IP6801 gateway using US915 frequencies.  I have confirmed that a device can join the network. 

```
info: LoRaWan.NetworkServer.BasicStation.WebSocketServer.LnsController[0]
      Message received from Client: {"msgtype":"jreq","MHdr":0,"JoinEUI":"00-13-7A-10-00-00-00-00","DevEUI":"00-13-7A-10-00-00-42-CE","DevNonce":64545,"MIC":1261133519,"RefTime":0.000000,"DR":0,"Freq":902700000,"upinfo":{"rctx":0,"xtime":43347146603584924,"gpstime":0,"fts":-1,"rssi":-111,"snr":8.75,"rxtime":1633544180.285960}}
2021-10-06 11:16:19.699 00137A10000042CE: join request received
2021-10-06 11:16:19.699 00137A10000042CE: querying the registry for OTAA device
2021-10-06 11:16:19.951 00137A10000042CE: updating twin
2021-10-06 11:16:20.121 00137A10000042CE: twin updated
2021-10-06 11:16:20.121 00137A10000042CE: the provided RX2DataRate is not valid
2021-10-06 11:16:20.125 00137A10000042CE: device added to cache
2021-10-06 11:16:20.145 00137A10000042CE: join accepted
```
 I will test the same code on an AAEON AIOT-ILRA01 gateway.  I'll report back with my findings soon.

gerfen, 2021-10-07:
I was able to get the basic station code working on an AAEON AIOT-ILRA01 gateway.  There are a couple of differences to note:

1. The GPIO PIN is 5
2. The SPI_DEV is "/dev/spidev2.0"
3. The SPI _SPEED is 2 - just like the AIOT-IP6801 unit

Here is the complete config for the module:

```
  "LoRaWanPktFwdModule": {
      "settings": {
      "image": "mandrx/lbs:0.0.22-amd64",
      "createOptions": "{\"HostConfig\":{\"NetworkMode\":\"host\",\"Privileged\":true},\"NetworkingConfig\":{\"EndpointsConfig\":{\"host\":{}}}}"
                        },
    "type": "docker",
    "version": "1.0",
    "env": {
    "RESET_PIN": {
            "value": "5"
     },
    "REGION": {
            "value": "US"
     },
    "SPI_DEV": {
            "value": "/dev/spidev2.0"
     },
     "SPI_SPEED": {
            "value": "2"
     },
     "TC_URI": {
            "value": "<your local dev box IP Address>:5000" 
     }
  },
  "status": "running",
  "restartPolicy": "always"
},
```

I should also note that I did all of my testing on Ubuntu 20.04 with IoT Edge 1.2 installed for both the AIOT-IP6801 and AIOT-ILRA01 gateways.

closed 2021-10-14 by Mandur